iiab/iiab #2067: master broken: box.lan changes to 172.18.96.1 in the browser

jvonau, 2019-12-16:
### Expected behavior and actual behavior.
When I access box.lan on the my cell phone the url displayed in the browser changes to 172.18.96.1, should remain box.lan

### Steps to reproduce the problem.
install master and add #2065, run ./iiiab-network, the access box.lan over wifi

### Specifications like the version of the IIAB, operating system version, or hardware details.
think it's nginx related, this line: rewrite ^/$ $scheme://$server_addr/home/; in /etc/nginx/server.conf is forcing the use of the ip address

holta, 2019-12-16:
@georgejhunt can you take a look at this?

jvonau, 2019-12-16:
/home should really be defined by the variable {{iiab_home_url}}

holta, 2019-12-16:
@georgejhunt: @jvonau suggests you're probably the best person to solve this?

But if not please unassign yourself, Thanks!

georgejhunt, 2019-12-17:
Function > form for me. But it's true that I've done quite a bit of the
nginx migration.
It comes down to reading docs, and trial and error.
And since it is cosmetic, will be relatively low on my priority list.
You can leave it on my list, until someone appears who will make it a
higher priority.
Sometimes I'm bored, and looking for a small task that might give me a
sense of accomplishment.

holta, 2019-12-17:
Please review @jvonau's proposed solution at PR #1258.

Thanks all for bringing these many NGINX regressions under control.

We're not there yet (http://minutes.iiab.io) but hopefully St Nick will help ;)

closed 2019-12-21 by holta