iiab/iiab #2077: NGINX: http://box/mediawiki erroneously redirects to http://localhost/mediawiki/index.php/Main_Page

holta, 2019-12-21:
Test occurred on a BIG-sized http://download.iiab.io installed onto the Desktop version of Ubuntu 18.04 LTS using http://10.8.0.46/mediawiki

The same result occurred when testing http://192.168.0.191/mediawiki

Tangentially related NGINX issues like to:
#2055 "AWStats URL's erroneously redirect to http://box.lan/cgi-bin/awstats.pl?config=schoolserver"
#2056 "http://box/wiki (DokuWiki) erroneously redirects to http://127.0.0.1:8090/wiki/"
#2057 "http://box/gitea displays 404 and broken images"

tim-moody, 2019-12-21:
fixed in #2080

holta, 2019-12-28:
As DokuWiki is phased out, let's repurpose http://box/wiki for MediaWiki.

In addition to MediaWiki's legacy URL, that we should also preserve: http://box/mediawiki

holta, 2019-12-29:
> fixed in #2080

Above replaced by PR #2086, using NGINX to natively implement MediaWiki (http://box/wiki, formerly http://box/mediawiki) and WordPress (http://box/wordpress).

closed 2019-12-31 by holta