iiab/iiab #2078: NGINX somehow prevents rapid testing of IIAB apps/services

holta, 2019-12-21:
With Apache it was an absolute lifeline that I could actively test all apps during IIAB installation, using URL's like...

http://10.8.0.46/gitea
http://10.8.0.46/lokole
http://10.8.0.46/nodered
http://10.8.0.46/mediawiki
(etc)

But now with NGINX I can no longer, until an hour later after reboot after reboot or such, which makes efficient testing absolutely impossible :(

FYI the equivalent URL's with http://192.168.0.191/* also fail with NGINX 404's.

Any idea why this is happening?

tim-moody, 2019-12-21:
so is the problem that you get 404s or that a reboot (and a wait) are required for the links to work?
did you clear your cache?

tim-moody, 2019-12-21:
we need to document which services are nginx direct and which have shims

holta, 2019-12-21:
> so is the problem that you get 404s or that a reboot (and a wait) are required for the links to work?

The hourlong is the problem.  The above URL's work "an hour or more later" but this destroys the rapid feedback look needed for efficient/iterative testing.

holta, 2019-12-21:
@jvonau is the solution something as simple as running `systemctl restart nginx` (or similar) near the end of each playbook?

jvonau, 2019-12-21:
> @jvonau is the solution something as simple as running `systemctl restart nginx` (or similar) near the end of each playbook?

No, not that simple, the above 4 urls have apache as the backends (think you could add /nextcloud /wordpress /elgg to the list) without native support so the nginx 'shims' would need to be installed in the role in addition to adding systemctl restart nginx.

> we need to document which services are nginx direct and which have shims

 It's been started see https://github.com/iiab/iiab/blob/master/roles/nginx/files/README.md

holta, 2019-12-21:
> https://github.com/iiab/iiab/tree/master/roles/nginx/README.md

Great beginning.

PS that README needs to be visible from a level higher (https://github.com/iiab/iiab/tree/master/roles/nginx) as with all other playbooks.

tim-moody, 2019-12-21:
someone please create a definitive list of 'shims' so we can check them off as we migrate to nginx direct

holta, 2019-12-21:
> someone please create a definitive list of 'shims' so we can check them off as we migrate to nginx direct

Did @georgejhunt create these?

Can someone point to an example shim?

jvonau, 2019-12-22:
yes, shim = contains proxy_pass http://127.0.0.1:8090/

grep 8090 /etc/nginx/conf.d/*
/etc/nginx/conf.d/dokuwiki-nginx.conf:  proxy_pass http://127.0.0.1:8090//wiki;
/etc/nginx/conf.d/elgg-nginx.conf:  proxy_pass http://127.0.0.1:8090/elgg;
/etc/nginx/conf.d/lokole-nginx.conf:  proxy_pass http://127.0.0.1:8090/lokole;
/etc/nginx/conf.d/mediawiki-nginx.conf:  proxy_pass http://127.0.0.1:8090/mediawiki;
/etc/nginx/conf.d/mediawiki-nginx.conf:    proxy_pass http://127.0.0.1:8090;
/etc/nginx/conf.d/moodle-nginx.conf:    proxy_pass http://127.0.0.1:8090;
/etc/nginx/conf.d/moodle-nginx.conf:    proxy_pass http://127.0.0.1:8090;
/etc/nginx/conf.d/nextcloud-nginx.conf:  proxy_pass http://127.0.0.1:8090/nextcloud;
/etc/nginx/conf.d/nextcloud-nginx.conf:    proxy_pass http://127.0.0.1:8090;
/etc/nginx/conf.d/nodered-nginx.conf:  proxy_pass http://127.0.0.1:8090/nodered;
/etc/nginx/conf.d/wordpress-nginx.conf:  proxy_pass http://127.0.0.1:8090/wordpress;
/etc/nginx/conf.d/wordpress-nginx.conf:    proxy_pass http://127.0.0.1:8090;

The late addition of lokole is missing from the readme.

holta, 2019-12-26:
@tim-moody's PR #2080 ("change services to nginx and simplify php") seems to have been one attempt to solve this apparently ?

tim-moody, 2019-12-26:
working on another. very painful.

holta, 2019-12-30:
Fixing this means refining the (7 or so?) affected playbooks.  Context from @jvonau:

> I tested the install with `nginx_enabled: False` to test the underlying web apps worked under apache before installing the nginx shims and native support then switching to nginx with `nginx_enabled: True` and using ./runrole... allowing back to back comparisons for evaluation and further refinement. The files only_nginx.yml contains what would be in the role for nginx and disable.yml contains the code to enable the apache version of the site. The disable.yml uses a2ensite (should of used *_enable there for each app) which requires the role to completed first.

holta, 2019-12-30:
@jvonau do you recall the 3-or-so others that also affected, beyond this list here?

- http://box/gitea
- http://box/lokole
- http://box/nodered
- http://box/mediawiki

holta, 2019-12-31:
[Clarif above](https://github.com/iiab/iiab/issues/2078#issuecomment-568212029) by @jvonau:

> add /nextcloud /wordpress /elgg to the list

tim-moody, 2019-12-31:
mediawiki and wordpress are converted to native nginx in #2086

tim-moody, 2019-12-31:
> No, not that simple, the above 4 urls have apache as the backend (think you could add /nextcloud /wordpress /elgg to the list) without native support so the nginx 'shims' would need to be installed in the role in addition to adding systemctl restart nginx.

This is an example from mediawiki:

```
- name: Remove mediawiki-nginx.conf if not mediawiki_enabled (debuntu)
  file:
    path: "{{ nginx_config_dir }}/mediawiki-nginx.conf"
    state: absent
  when: not mediawiki_enabled and is_debuntu

- name: Restart nginx service to enable/disable http://box{{ mediawiki_url }}
  systemd:
    name: nginx
    daemon_reload: yes
    state: restarted
```

holta, 2020-01-02:
If NGINX was fully moved from [Stage 9-local-addons](https://github.com/iiab/iiab/wiki/IIAB-Contributors-Guide#ansible) to [Stage 3-base-server](https://github.com/iiab/iiab/wiki/IIAB-Contributors-Guide#ansible) (with Apache) then this would become possible.

* some shims might have to be moved around, into certain roles?

jvonau, 2020-01-13:
#2143 moves apache and niginx to stage 4

holta, 2020-01-16:
This NGINX regression is largely fixed now, thanks to @jvonau's PR #2131 and my follow-on PR #2139.

Fixing / mitigating #2145 ("http://box/elgg only partially loads when accessed from IIAB's WAN side [then fixes itself later on during IIAB install]") is probably still very possible, but not as high a priority.

Whereas #2148 ("http://box/munin is installed in iiab-install's Stage 8 but shows "403 Forbidden" until various points in Stage 9")) generally fixes itself within a minute or two, so acceptable as is, despite its very misleading "403 Forbidden" browser output.

closed 2020-01-16 by holta