iiab/iiab #2082: Up-to-date raspberry pi images

Mortvert, 2019-12-26:
Hello.

Is there any chance for release of up-to-date 7.0 and 7.1 images?
I can't install IIAB on pi zero W because node.js install keeps failing (Something about wrong platform).

Latest image is from last year and it.s 6.x image so it's rather out of date.

holta, 2019-12-26:
Hi @Mortvert 

Can you install http://download.iiab.io onto a microSD card using a Raspberry Pi 3 or Raspberry Pi 4?

This takes as little as an hour[*].

And then you can move the microSD card into your Raspberry Pi Zero W.

[*] however due to a temporary Raspbian bug that the Raspberry Pi Foundation should fix next week, please run these commands before installing Internet-in-a-Box:

```
apt update
apt -y dist-upgrade
rpi-update
reboot

holta, 2019-12-26:
Aside—details of Raspbian's upstream bug are here: https://github.com/iiab/iiab/issues/1993

Mortvert, 2019-12-26:
Sadly, I do not own a Raspberry Pi 3/4.

Disabling sugarizer during install seems to allow me to install IIAB just fine.
With sugarizer enabled during install I get the following error.
TASK [nodejs : Set up Node.js 10.x apt sources (debuntu)] **********************
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "curl -sL https://deb.nodesource.com/setup_10.x | bash -", "delta": "0:00:00.979019", "end": "2019-12-2619:27:39.850732", "msg": "non-zero return code", "rc": 1, "start": "2019-12-2619:27:38.871713", "stderr": "", "stderr_lines": [], "stdout": "\n## Installing the NodeSource Node.js 10.x repo...\n\n\n## You appear to be running on ARMv6 har`ware. Unfortunately this is not currently supported by the NodeSource Linux distributions. Please use the 'linux-armv6l' binary tarballs available directly from nodejs.org for Node.js 4 and later.", "stdout_lines": ["", "## Installing theNodeSource Node.js 10.x repo...", "", "", "## You appear to be running on ARMv6hardware. Unfortunately this is not currently supported by the NodeSource Linuxdistributions. Please use the 'linux-armv6l' binary tarballs available directlyfrom nodejs.org for Node.js 4 and later."]}

I was also asking about the images because usually it's flash and forget with raspberry pi, after initial setup.

Mortvert, 2019-12-26:
I should probably create a separate issue for the thing above.

tim-moody, 2019-12-26:
I have been intending to make a fresh image since, as you say, it's been
quite awhile, but was hoping to get a raspbian that didn't need patching.
btw ubuntu 19.10 works out of the box.

Mortvert, 2019-12-27:
> 
> I have been intending to make a fresh image since, as you say, it's been quite awhile, but was hoping to get a raspbian that didn't need patching. btw ubuntu 19.10 works out of the box.
> […](#)

Sadly, Ubuntu is only for Raspberry Pi 2/3/4 - they are ARMv7.
Raspberry Pi Zero W is ARMv6

holta, 2019-12-27:
> Sadly, Ubuntu is only for Raspberry Pi 2/3/4 - they are ARMv7.
Raspberry Pi Zero W is ARMv6

Another problem with Ubuntu is that IIAB's Captive Portal today works with Raspbian only.

holta, 2019-12-27:
Related:

#1799 Who can test IIAB 7.x on Raspberry Pi Zero W?

Mortvert, 2019-12-27:
Apparently, NodeJS 12 for ARMv6 is only available as unofficial build as per the following:
https://github.com/nodejs/build/issues/1677#issuecomment-486204349

Will this mean that there will be no 7.x for Pi0w?

holta, 2019-12-27:
> Apparently, NodeJS 12 for ARMv6 is only available as unofficial build as per the following:
> [nodejs/build#1677 (comment)](https://github.com/nodejs/build/issues/1677#issuecomment-486204349)

Thank you for uncovering that !

> Will this mean that there will be no 7.x for Pi0w?

No.  Most RPi Zero W users do not use Node.js, but in your case use the line `nodejs_version: 10.x` within your /etc/iiab/local_vars.yml

(Can you please confirm that works, after you've tried it?)

PS which Node.js apps is your IIAB depending on, among things like Sugarizer, nodered (Node-RED) and pbx (Asterix, FreePBX) ?

holta, 2019-12-27:
> Most RPi Zero W users do not use Node.js, but in your case use the line `nodejs_version: 10.x` within your /etc/iiab/local_vars.yml

Addendum: you might also need to install Node.js (version 10.x presumably) manually at the command-line, if `curl -sL https://deb.nodesource.com/setup_10.x | bash -` fails on ARMv6 / RPi Zero W as you've suggested above.

Are you aware of the runrole command?

This can help you install any IIAB app/service after IIAB and Node.js have been installed:

1. Edit /etc/iiab/local_vars.yml ensuring that IIAB app is the to be installed AND enabled, e.g. `sugarizer_install: True` and `sugarizer_enabled: True`
2. sudo su -
3. cd /opt/iiab/iiab
4. ./runrole <app>    # e.g. ./runrole sugarizer

jvonau, 2019-12-27:
> FAILED! => {"changed": true, "cmd": "curl -sL https://deb.nodesource.com/setup_10.x | bash -", 

Note the version number, 10.x, nodejs is already broken on ARMv6 using nodesource's ppa. This would not be an issue using nodejs from the distro's repo.

jvonau, 2019-12-27:
> btw ubuntu 19.10 works out of the box.

I would double check if that image boots on a pi zero W

Mortvert, 2019-12-27:
> 
> 
> > Most RPi Zero W users do not use Node.js, but in your case use the line `nodejs_version: 10.x` within your /etc/iiab/local_vars.yml
> 
> Addendum: you might also need to install Node.js (version 10.x presumably) manually at the command-line, if `curl -sL https://deb.nodesource.com/setup_10.x | bash -` fails on ARMv6 / RPi Zero W as you've suggested above.
> 
> Are you aware of the [runrole](https://github.com/iiab/iiab/blob/master/runrole) command?
> 
> This can help you install any IIAB app/service, after IIAB and Node.js have been installed:
> 
>     1. Edit /etc/iiab/local_vars.yml ensuring that IIAB app is set to be installed AND enabled, e.g. `sugarizer_install: True` and `sugarizer_enabled: True`
> 
>     2. Run: sudo su -
> 
>     3. Run: cd /opt/iiab/iiab
> 
>     4. Run: ./runrole [APP]    # e.g. ./runrole sugarizer

I have managed to get MIN install running on RPi 0W without any problems, save for the kiwix issue that's in the tracker.. Took several hours to install but it runs.
I'm now trying to install things like wordpress/calibre-web and others. I'll get back to you when it's done.

Mortvert, 2019-12-27:
Wordpress installs without issue.
Now to figure out how to remove it properly :D

holta, 2019-12-27:
@Mortvert if you succeed in installing Node.js 10.x on RPi Zero W...what exact command-line steps worked for you?

Mortvert, 2019-12-27:
Downloaded relevant package from https://nodejs.org/dist/latest-v10.x/
Used https://nodejs.org/dist/latest-v10.x/node-v10.18.0-linux-armv6l.tar.gz

Downloaded it via wget
tar -xzf node-v10.18.0-linux-armv6l.tar.gz
cd node-v10.18.0-linux-armv6l/
sudo cp -R * /usr/local/

At that point node -v works.

You can check which version of Node.js you need by executing "uname -m" apparently. Shows you which ARM version your raspberry is running.

And off I go, reinstalling raspbian again so I can get rid of wordpress :)

Mortvert, 2019-12-27:
[image]
Here you go, ignore my flailing at trying to remember what the node's package manager is called :smile:

holta, 2019-12-27:
Quite an accomplishment, that has the potential to serve many others: thank you @Mortvert !

fivdi, 2019-12-27:
I'm not sure if it's relevant to the conversation here but the unofficial ARMv6 Node.js v12 and v13 binaries for the Raspberry Pi Zero can be found at https://unofficial-builds.nodejs.org/download/release/

Mortvert, 2019-12-28:
Is there a IRC/Discord channel for this project? Can't seem to find one.

holta, 2019-12-28:
> Is there a IRC/Discord channel for this project? Can't seem to find one.

@Mortvert we have a chat channel that's unfortunately limited to 20 people, but write me (holt @ unleashkids.org) and I can add you to that if you want!

jvonau, 2021-07-21:
close?

holta, 2021-07-21:
> close?

@Mortvert any final thoughts/suggestions before this is closed?!

Certainly lots of progress has been made, even if not all pretty, e.g.:

- #2873
- https://github.com/iiab/iiab/wiki/IIAB-7.2-Release-Notes#known-issues