iiab/iiab #2092: http://box/nextcloud implemented natively with NGINX

holta, 2020-01-02:
Easier said than done?  Still let's try to make this happen in coming months, as @tim-moody has done for WordPress and MediaWiki etc here: PR #2086

holta, 2020-01-03:
@m-anish can you consider this small task, to get familiar with NGINX, so that Apache might be fully eliminated in favor of NGINX on your IIAB installations?

ASIDE: please shout about the greatest omissions/bugs within [NGINX's README](https://github.com/iiab/iiab/blob/master/roles/nginx/README.md) that George created 2019-10-16, so that I can make sure these are fixed prior to IIAB 7.1 release around 2020-01-30 !

holta, 2020-01-03:
@m-anish does the following help, as mentioned by Tim?

> How to Install Nextcloud with Nginx on Ubuntu 18.04 LTS
> https://www.howtoforge.com/tutorial/ubuntu-nginx-nextcloud/

m-anish, 2020-01-06:
shall look into it :) 

please assign to me.

holta, 2020-01-06:
@m-anish let's also make sure Nextcloud works with intensive loads, using the `apache_high_php_limits` variable (that we might want to rename `high_php_limits`) to work across both web servers: #2109

holta, 2020-01-06:
(And whatever gets implemented, let's update Sept 2019's [Nextcloud doc](https://github.com/iiab/iiab/blob/master/roles/nextcloud/README.md) so this remains understandable for all!)

holta, 2020-01-07:
FYI @m-anish and I would like to remove Nextcloud 15 support here, for OS's that lack PHP 7.1+:

https://github.com/iiab/iiab/blob/master/roles/nextcloud/defaults/main.yml#L22-L24
https://github.com/iiab/iiab/blob/master/roles/nextcloud/tasks/install.yml (see all 10 references to `php_new` which is the auto-detection of PHP 7.1+ I created mid-2019)

_**Effectively this means we're deciding to start deprecating Raspbian 9, Debian 9, Ubuntu 16.04 and similar OS's that pre-date June 2019...**_

m-anish, 2020-01-08:
#2119

holta, 2020-02-17:
Resolved by PR #2255.

(See also [Nextcloud video(s) not displaying](https://github.com/iiab/iiab/pull/2255#issuecomment-586789380) &mdash; I've mentioned this video/redirect problem within [IIAB 7.1's Known Issues](https://github.com/iiab/iiab/wiki/IIAB-7.1-Release-Notes#known-issues) just in case it's not immediately solved in coming days?!)

closed 2020-02-17 by holta