iiab/iiab #2097: Header Fonts not rendering in Firefox-on-Windows, Safari, Chrome-on-macOS

holta, 2020-01-04:
When clicking http://box/admin > Content Menus > Menu Properties > "Select Mobile Header Font" OR "Select Desktop Header Font" it should appear as follows...

[image]

Unfortunately these fonts are not rendering or visualizable within Firefox, Safari or Chrome on iOS &mdash; a typical result being:

[image]

avni, 2020-01-04:
This may not be easily fixable: https://developer.mozilla.org/en-US/docs/Learn/Forms/Advanced_styling_for_HTML_forms

tim-moody, 2020-01-08:
I confirm the problem on FF on Win 10.
I don't see how this can be fixed and I don't see any relevance of the above link which does not mention external fonts that I can see.

closed 2020-01-09 by holta