iiab/iiab #2103: Kolibri 0.13.0 no longer works with? /usr/bin/kolibri language setdefault "en"

holta, 2020-01-05:
@benjaoming has syntax apparently changed in Kolibri 0.13.0 ?

Do you have a recommendation or documentation here?

> TASK [kolibri : Set Kolibri default language (en)] *****************************
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "export KOLIBRI_HOME=\"/library/kolibri\" && \"/usr/bin/kolibri\" language setdefault \"en\"", "delta": "0:00:06.290086", "end": "2020-01-05 10:39:50.428319", "msg": "non-zero return code", "rc": 2, "start": "2020-01-05 10:39:44.138233", "stderr": "Usage: kolibri [OPTIONS] COMMAND [ARGS]...\nTry \"kolibri --help\" for help.\n\nError: No such command \"language\".", "stderr_lines": ["Usage: kolibri [OPTIONS] COMMAND [ARGS]...", "Try \"kolibri --help\" for help.", "", "Error: No such command \"language\"."], "stdout": "", "stdout_lines": []}

As you can see below, IIAB forcefully ignores any Kolibri's errors here (as @arky implemented for the purpose of reinstalls/upgrades) but now this error is being generated even on a fresh install:

https://github.com/iiab/iiab/blob/master/roles/kolibri/tasks/install.yml#L80-L85

What's a better way, if possible?

benjaoming, 2020-01-05:
This is true. I think the deprecation and changelog entry for removal of `kolibri language` was lost. I have added it to the release notes of 0.13.0 now.

The solution is to use the more feature rich `kolibri manage provisiondevice` -- you could set the language with `kolibri manage provisiondevice --language_id {{ kolibri_language }}`. But perhaps a better option is to omit this step entirely since provisioning a device is not limited to setting the language.

CC: @indirectlylit @rtibbles

rtibbles, 2020-01-05:
Note that the reason that this was completely removed was that the actual command itself has been essentially a no op since 0.6.x, as the setting would always be overridden by other language selection code.

closed 2020-01-05 by holta