iiab/iiab #2105: Update /opt/iiab/iiab/.travis.yml ...Python version? dist version? Ansible Lint?

holta, 2020-01-05:
1) https://github.com/iiab/iiab/blob/master/.travis.yml#L3 currently shows:

   ```
   python: "2.7"
   ```

2) https://github.com/iiab/iiab/blob/master/.travis.yml#L6 currently shows:

   ```
   dist: trusty
   ```

3) Can [Ansible Lint](https://github.com/ansible/ansible-lint) ([docs](https://docs.ansible.com/ansible-lint/)) be plugged into the above file and if so how?

holta, 2020-01-05:
Example output of `ansible-lint -p -R -v iiab-stages.yml` in the link below:

https://paste.ubuntu.com/p/9S9qWxNbB8/

(Thanks to @jvonau)

holta, 2021-02-24:
@jvonau suggests this should be "ported to bionic, use Python 3, and install the [Ansible] collections..."

holta, 2021-02-24:
This might become increasingly important if GitHub/Travis' testing environment's network connections fail more and more often?  As was happening earlier this morning, e.g. https://travis-ci.com/github/iiab/iiab/builds/218107893#L527 showed things like:

```
Err:77 http://archive.ubuntu.com/ubuntu trusty InRelease
  Could not connect to archive.ubuntu.com:80 (91.189.88.152), connection timed out [IP: 91.189.88.152 80]
```

jvonau, 2021-02-24:
https://config.travis-ci.com/explore Give this a shot.
>language: python
python: "3.8"
dist: focal
\# Install ansible
install:
\# Create ansible.cfg with correct roles_path and local_tmp
\# not sure is is needed now
\- "{ echo '[defaults]'; echo 'roles_path = ./roles/'; } >> ansible.cfg"
script:
\- /opt/iiab/iiab/scripts/ansible
\# now call the test
\- ansible-playbook tests/test.yml -i tests/inventory --syntax-check

holta, 2021-02-24:
> script:
> - opt/iiab/iiab/scripts/ansible

Very interesting idea.  Will this slow down Ansible installs during each CI syntax check?  (That ideally we'd like completed within ~2 minutes flat!)

holta, 2021-02-24:
FYI scripts/ansible fails when part of a Travis job as follows:

```
apt update; install dirmngr; PPA to /etc/apt/sources.list.d/iiab-ansible.list
Reading package lists... Done
E: Could not open lock file /var/lib/apt/lists/lock - open (13: Permission denied)
E: Unable to lock directory /var/lib/apt/lists/
W: Problem unlinking the file /var/cache/apt/pkgcache.bin - RemoveCaches (13: Permission denied)
W: Problem unlinking the file /var/cache/apt/srcpkgcache.bin - RemoveCaches (13: Permission denied)
The command "scripts/ansible" failed and exited with 100 during .
Your build has been stopped.
```

(As excerpted from https://travis-ci.com/github/holta/iiab/builds/218172697#L206)

jvonau, 2021-02-24:
Don't you think it would be best to test with the same ansible setup as what is used in production out in the wild? Some of the "slowness" of the ansible script is the ill thought out concept of using 'bonic' packages intended for python 3.7 on recent Ubuntu where python 3.8 is standard. Now [ansible's ppa](http://ppa.launchpad.net/ansible/ansible/ubuntu/dists/?C=M;O=D) supports 'focal' #2521 has been updated. 
>Could not open lock file /var/lib/apt/lists/lock - open (13: Permission denied)

Off the top of my head try `sudo /opt/iiab/iiab/scripts/ansible` instead, otherwise scripts/ansible needs to be picked apart and ported to use native calls. Let me know if if works or not.

jvonau, 2021-02-24:
[doc state both 'required' & 'deprecated'](https://config.travis-ci.com/), took a shot and left sudo out, above 'install:' place:
>sudo: true

holta, 2021-02-24:
> [doc state both 'required' & 'deprecated'](https://config.travis-ci.com/), took a shot and left sudo out, above 'install:' try adding:
> 
> > sudo: true

The above doc (and others!) say the `sudo:` tag is deprecated from Travis.

e.g. see PR #2689 which is confirmed working well (and fast!)