iiab/iiab #2107: IIAB users want (A) reliable maps (if fast maps are not possible) (B) countries/oceans/rivers/mountains

holta, 2020-01-06:
It's a different world that 2012 when IIAB began delivering offline maps.  Everyone wants very reliable maps if very fast maps are now possible.

And essentially all teachers and implementers now ask for:

1. Rapidly-installable Map Packs for their region w/o hassle.
2. _Search should work for countries & famous ocean/rivers/mountains_ if we want to be taken seriously by geography teachers.  Even if states/provinces/territories & seas/lakes do not show up, so be it &mdash; _we cannot please everyone!_
3. A reliable (if not fast!) User Experience is now expected by nearly everyone worldwide, no matter how rural, even if it will never match Google Map.  CONTEXT: teachers around the world now ALSO use Google Maps, on their personal smart phones when in the city. e.g. when they can afford mobile data etc, and they all now expect that.  CONSEQUENCE: our own OSM user experience / usability (UX Design) must now take this established culture into account &mdash; now that 8 years later, the whole planet now is aware of Google Maps and uses occasion (including its cached version offline!)  POSITIVE: our adoption can now accelerate dramatically, now that we no longer need extensive training sessions as we used to, to explain the concepts of zooming/searching/etc.

jvonau, 2021-07-21:
implemented

holta, 2021-07-21:
> implemented

Mostly!

Let's leave the "search question" open however &mdash; at least considering what improvements might be appropriate and/or feasible for the new decade.