iiab/iiab #2109: Can apache_high_php_limits: False/True work with NGINX?

holta, 2020-01-06:
1) Does IIAB's `apache_high_php_limits: False/True` work with NGINXtoday?

2) If not, what's needed to get it working?  As needed by schools requiring intensive usage of WordPress, Nextcloud, Moodle etc.

3) Should the variable be renamed to `high_php_limits` to serve both Apache and NGINX?

Ref: #1147 #2108

tim-moody, 2020-01-06:
1. no, but there are articles on optimizing php in nginx. also, except for wordpress, the php apps are still served by apache.
2. how do you know it's needed by nginx, that the performance profile is the same as apache? Is it an apache setting or a php setting?
3. don't think we know yet

holta, 2020-01-06:
> Is it an apache setting or a php setting?

It's set in `/etc/php/7.2/cli/php.ini` (on Ubuntu 18.04) or `/etc/php/7.3/apache2/php.ini` (on Raspbian) as explained here: https://github.com/iiab/iiab/blob/master/roles/httpd/tasks/enable.yml#L1-L14

Can someone make any kind of NGINX recommendation here (for larger schools & computer labs) even if worst case we have to revise it later based on field experience?

holta, 2020-01-08:
Nextcloud isn't yet natively served by NGINX, but FYI to help @m-anish & all of us navigate this...a fresh install of IIAB (on Ubuntu 19.10) shows:

> Security & setup warnings
> ...
> The PHP memory limit is below the recommended value of 512MB.

You can get to this page (http://10.8.0.46/nextcloud/index.php/settings/admin/overview) by clicking http://10.8.0.46/nextcloud > (circle containing 'A' in top-right) > Settings (in dropdown) > Overview (in Administration section, in left panel).

holta, 2020-02-18:
Variable apache_high_php_limits: False/True has been replaced by `nginx_high_php_limits: False/True`

Documentation needs to be updated in places like http://FAQ.IIAB.IO !

Ref: PR #2255

holta, 2020-02-18:
Documentation updated at http://FAQ.IIAB.IO

closed 2020-02-18 by holta