iiab/iiab #2110: Multiple IIAB menus or only home (default) menu ?

llovemixo, 2020-01-07:
### Expected behavior and actual behavior.
Version 7.0, Pi 4 (August 21, 2019) and Pi Zero W (fresh install of Buster and IIAB)

Admin Console,  Menu Save only accepts one name, the default of "home ".  Any other name fails with error: SAVE-MENU-DEF FAILED and reported unexpected error in Command SAVE-MENU-DEF  .

Is this the expected behavior ?  

The structure seems to imply multiple menu's might be possible and do become possible with a little terminal work.

In /library/www/html  is the directory home/ with files menu.json and index.html  .
If one copies the home/ to  child/, the Menu Save with name "child" will be saved  and can be run with http:/box.lan/child  .   The index file remains the same .  If desired, ,only the Title line needs to be changed from  "Home" to "Child" in the index.html  .  

One can construct the directory using terminal with owner/user = www-data, but only the menu.json file gets save when Admin Console>Menu.>MenuSave.is executed.

Why multiple menu's   ?   Different age groups may need different parts of a large menu and big menu's are awkward.  So a menu for kindergarten, elementary, etc. .

Also, a simple index.html replacement for that in home, can be imagined that would provide a simple menu calling :  http:/box.lan/ child  , etc. .

Nathan Riddle
xoapart@xoshare.org

llovemixo, 2020-01-07:
reopen

tim-moody, 2020-01-07:
@llovemixo the behavior you want is the expected behavior for the reasons you give and has worked for quite some time
http://medbox.iiab.me/home/ does this. It uses iiab menuing for the top level as well as 4 child menus.

> unexpected error in Command SAVE-MENU-DEF

means just what it says, something unexpectedly went wrong.

When this was first reported I tried to reproduce and couldn't:
Note that the target directory must exist.
As root
`mkdir  /library/www/html/child`
now you should be able to save to child (no slashes)
If this does not work I need to see menu.json

If you want to troubleshoot do the following as root after ssh into iiab:

```
cd /opt/admin/cmdsrv/
systemctl stop iiab-cmdsrv
./iiab-cmdsrv.py
```

this will put you in debug mode where errors are not trapped and you should see python errors

llovemixo, 2020-01-07:
Tim,
That is effectively what I was reporting :  the directories ("child" in my case) must be created manually. It works that way with August 7.0 on Pi 4 and all fresh  Buster  and IIAB Pi Zero W
Thanks for confirming that it is the expected behavior and warning that there is a 4 child limitation
Can you point to any documentation where this feature may be explored ?
Nathan Riddle

tim-moody, 2020-01-07:
I didn't mean to say there is a any limit as to how many of these child directories you may have.

The help doesn't explicitly mention the creation of a directory as I didn't want to confuse people. I have added a note to that effect now.

holta, 2020-01-07:
> The help doesn't explicitly mention the creation of a directory as I didn't want to confuse people. I have added a note to that effect now.

@llovemixo I think @tim-moody is talking about the help pages here:
https://github.com/iiab/iiab-admin-console/tree/master/roles/console/files/help

If so, it might take a few days for these help page clarifications to be published, i.e. when PR's (pull requests) like the following become official: https://github.com/iiab/iiab-admin-console/pull/278

tim-moody, 2020-01-07:
or could be sooner

holta, 2020-01-08:
@llovemixo are there further questions to help you out here?

(Or if not let's close this, Thanks!)

llovemixo, 2020-01-09:
The menu structure appears to support a menu of menu's.
A dummy application (en-child) was placed in modules directory; a menu item "child" was created and placed in "home" (default) menu.  When http://box.local, the menu item "child" is present.

Execution of the "child" has not been tested yet.  I can see 3 possible ways:  1) copy /library/www/html/child/ directory to my en-child ;  2) place index.html in en-child/ to point to /library/www/html/child/ ; or use execution URL field in menu item for "child".

Thanks Tim

Closing this question.

closed 2020-01-09 by llovemixo