iiab/iiab #2117: NGINX: Nextcloud broken icons & broken links, erroneously contain http://127.0.0.1:8090

holta, 2020-01-08:
On a freshly installed Nextcloud 7.0.2 as part of BIG-sized IIAB on Ubuntu 19.10, login to http://10.8.0.46/nextcloud and then click the "Notifications" icons in the top-right.

The notification dropdown then attempts (but fails) to display these broken icons/images:

http://127.0.0.1:8090/nextcloud/apps/firstrunwizard/img/apps/spreed.svg
http://127.0.0.1:8090/nextcloud/apps/firstrunwizard/img/apps/mail.svg
http://127.0.0.1:8090/nextcloud/apps/firstrunwizard/img/apps/contacts.svg
http://127.0.0.1:8090/nextcloud/apps/firstrunwizard/img/apps/calendar.svg

Leading to these broken links;

http://127.0.0.1:8090/nextcloud/index.php/settings/apps/social/spreed
http://127.0.0.1:8090/nextcloud/index.php/settings/apps/social/mail
http://127.0.0.1:8090/nextcloud/index.php/settings/apps/organization/contacts
http://127.0.0.1:8090/nextcloud/index.php/settings/apps/organization/calendar

Background: this BIG-sized IIAB (http://download.iiab.io) was freshly installed onto Ubuntu 19.10 (10.8.0.46) 5-6 hours ago.  Tagging @m-anish in case his work on #2092 might help us illustrate the root cause ?

jvonau, 2020-01-08:
http://127.0.0.1:8090 is not erroneous, it's because of the use of proxy_pass in the nginx config file that appears to be missing some additional code to do the correct rewrite/php handling.

holta, 2020-01-08:
Good News: @m-anish's PR #2119 proof-of-concept offers a possible fix, as was confirmed on RPi 4 (icons show up).

holta, 2020-02-18:
This is now part of IIAB 7.1 pre-releases, as a result of PR #2255.

Thanks again to @m-anish !

closed 2020-02-18 by holta