iiab/iiab #2123: Why does MIN-sized install on Raspbian contain /etc/php/7.0

holta, 2020-01-09:
Thanks @jvonau for finding this:

```
root@box:~# apt -a list php
Listing... Done
php/stable 2:7.3+69 all

root@box:~# locate php | grep 7\\.0
/etc/php/7.0
/etc/php/7.0/apache2
/etc/php/7.0/mods-available
/etc/php/7.0/apache2/conf.d
/etc/php/7.0/apache2/conf.d/20-stem.ini
/etc/php/7.0/mods-available/stem.ini

root@box:~# du -hsc /etc/php/*
20K     /etc/php/7.0
468K    /etc/php/7.3
488K    total
```

jvonau, 2020-01-09:
suspect is the contents of [php-stem.rpi.tar](https://github.com/iiab/iiab/blob/master/roles/httpd/tasks/php-stem.yml)

holta, 2020-01-09:
> suspect is the contents of [php-stem.rpi.tar](https://github.com/iiab/iiab/blob/master/roles/httpd/tasks/php-stem.yml)

If so that would be caused by...
https://github.com/iiab/iiab/blob/master/roles/httpd/tasks/main.yml which calls...
https://github.com/iiab/iiab/blob/master/roles/httpd/tasks/php-stem.yml

jvonau, 2020-01-09:
also creates /usr/lib/php/20151012/stem.so

holta, 2020-01-09:
@tim-moody your above php-stem patch to make WikiHow search still working?  (Or is it obsolete?)

holta, 2020-01-09:
June/July 2018 refs:

#829 "Is Spanish WikiHow's search fixable? (works on RACHEL site!)"
PR #925 "php-stem fixes search @ http://box/modules/es-wikihow"

holta, 2020-01-23:
Call proposed to comment out line 8 here for now:
https://github.com/iiab/iiab/blob/master/roles/httpd/tasks/main.yml#L8

As no modern OS's are using PHP 7.0...

(Even if stem.so might/maybe work in Raspbian new dir /usr/lib/php/20180731 which replaces the old /usr/lib/php/20151012 ?)

holta, 2020-01-24:
Thanks to @tim-moody who tried to get that old PHP 7.0 stuff (.so shared object etc) to work in Raspbian's latest PHP 7.3, even if no luck!

Good enough for now...mitigated by PR #2189 which comments out the 2018 workaround.

holta, 2020-01-24:
Re-opening this tkt as @tim-moody made a lot of progress here over the past 24h:

> decided to rebuild php-stem.so. first on vm. it works

Towards fixing Spanish WikiHow's (es-wikihow) search &mdash; on newer platforms shipping more recent versions of PHP than Raspbian Stretch's PHP 7.0...
- PHP 7.2 (Ubuntu 18.04)
- PHP 7.3 (Debian 10 Buster, Raspbian Buster)
- PHP 7.4 (Ubuntu 20.04 ?)

tim-moody, 2020-01-25:
I have built php-stem.so for 7.2 x64 and 7.3 for rpi and x64. tested successfully 7.3 on rpi and ubuntu 19.10 vm.

need to upload tar files to d.iiab.io
need to modify playbook to install and to add symlink to /etc/php/7.3/fpm/conf.d, which is no longer in tar file.

closed 2020-01-27 by holta