iiab/iiab #2124: Unify Apache & NGINX logs...to /var/log/nginx ?

holta, 2020-01-09:
Thanks @georgejhunt for researching and trying to solve this, as discussed during http://minutes.iiab.io today!

Side Questions:
- Is the Apache log file format sufficiently compatible with NGINX's log file format, to safely unify these?
- @tim-moody Asks: does this affect http://box/awstats or might we lose a bit of AWStats' analytics monthly or so?

jvonau, 2020-01-12:
This is actually a pretty serious bug, logrotate moves the apache log file but with nginx's logging going there the newly moved file (.1) continues to be used because ngnix held the log file open and is not being restarted.

holta, 2020-01-16:
> This is actually a pretty serious bug, logrotate moves the apache log file but with nginx's logging going there the newly moved file (.1) continues to be used because ngnix held the log file open and is not being restarted.

Would that be resolved by PR #2156 ?

Aside: PR #2155 does similar but just for AWStats.

holta, 2020-01-18:
@tim-moody can you explained your recommendation here, with the ticket's new subject line you added 2 days ago?

In context with things like #2121 and #2168 if possible?  Thanks!

holta, 2020-01-23:
Closing thanks @georgejhunt's PRs #2155 #2156.

closed 2020-01-23 by holta