iiab/iiab #2136: Nested IIAB Menu's simple example

llovemixo, 2020-01-12:
### Expected behavior and actual behavior.
Example use of Admin Console Menu to create nested menu's.
### Specifications like the version of the IIAB, operating system version, or hardware details.
Tested on IIAB version 7.0 / Raspbian Buster full desktop on Pl Zero W .

A menu item can be created to point to a menu as a whole.  This menu item can then be placed in other menu's to load this menu.

Create menu (kinder):
In /library/www/html   cp home/ kinder/
Create "index" file kinder.html (8 line file) to jump to url /kinder:
<sorry, code not permitted here>

Then, in Admin Console, clone the menu item "credits" , change entries as follows and save:
Menu Item code;  en-kinder
Optional Start URL :    /kinder
Fill other descriptions as desired.

This menu item can then be added to "home" menu.  
Put some sample menu items in kinder menu.

Connect to Internet-in-a-Box and start IIAB with http://box.local
Select Kinder menu item  -- Kinder menu gets loaded. 

Thanks Tim for your initial library Index.html from IIAB on XO.
ILovemixo

holta, 2020-01-12:
Background:

#2110 "Multiple IIAB menus or only home (default) menu ? [menu of menus ?]"

llovemixo, 2020-01-12:
It has been pointed out that it is unnecessary to create the kinder.html file.
The kinder directory has index.html, which can be called directly.
So in the menu item, for optional start url, use /kinder/index.html   instead of /kinder.html  .
Therefore, no special coding needed.  Need only IIAB cloning of  menu item as explained main text.

tim-moody, 2020-01-13:
"start_url": "/kinder" will also suffice as index.html will be found by the web server

holta, 2020-01-13:
> "start_url": "/kinder" will also suffice as index.html will be found by the web server

`start_url` is described here: https://github.com/iiab/iiab/wiki/IIAB-Menuing#menu-item-definition-fields

Context: menu-defs are the .json files found in /library/www/html/js-menu/menu-files/menu-defs (that define how each Content Pack or IIAB App appears on IIAB's home page etc).

QUITE SEPARATELY, for also just FYI https://github.com/iiab/iiab/blob/master/vars/default_vars.yml#L76-L78 mentions2 other variables that are somewhat relevant here:

```
# Homepage: set to /home or /wordpress or /wiki (for MediaWiki)
iiab_home_url: /home
# You might also want to set captiveportal_splash_page (below!)
```

i.e. both above variables can be set in [/etc/iiab/local_vars.yml](http://wiki.laptop.org/go/IIAB/FAQ#What_is_local_vars.yml_and_how_do_I_customize_it.3F) if nec.

closed 2020-01-14 by llovemixo