iiab/iiab #2139: KA Lite fails to install on RPi 4: piwheels.org's "setuptools requires Python '>=3.5' but the running Python is 2.7.16"

holta, 2020-01-13:
This is reminiscent of https://github.com/iiab/iiab/issues/392 in late 2017:  (Background: a BIG-sized install of IIAB (http://download.iiab.io) is being attempted on a 1GB RAM Raspberry Pi 4 and the OS is Raspbian Lite 2019-09-26 with all apt updates applied, and then rpi-update firmware/kernel (4.19.93) to work around #1993.)

> TASK [kalite : Install KA Lite static to /usr/local/kalite/venv, using pip (debuntu)] ***
fatal: [127.0.0.1]: FAILED! => {"changed": false, "cmd": "/usr/bin/virtualenv --no-site-packages -ppython2.7 /usr/local/kalite/venv", "msg": "stdout: New python executable in /usr/local/kalite/venv/bin/python2.7\nAlso creating executable in /usr/local/kalite/venv/bin/python\nInstalling setuptools, pkg_resources, pip, wheel...\n  Complete output from command /usr/local/kalite/venv/bin/python2.7 - setuptools pkg_resources pip wheel:\n  Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple\nLooking in links: /usr/lib/python3/dist-packages, /usr/share/python-wheels/\nCollecting setuptools\n  Downloading https://www.piwheels.org/simple/setuptools/setuptools-45.0.0-py2.py3-none-any.whl (583kB)\nsetuptools requires Python '>=3.5' but the running Python is 2.7.16\n----------------------------------------\n...Installing setuptools, pkg_resources, pip, wheel...done.\nRunning virtualenv with interpreter /usr/bin/python2.7\n\n:stderr: Traceback (most recent call last):\n  File \"/usr/lib/python3/dist-packages/virtualenv.py\", line 2379, in <module>\n    main()\n  File \"/usr/lib/python3/dist-packages/virtualenv.py\", line 724, in main\n    symlink=options.symlink)\n  File \"/usr/lib/python3/dist-packages/virtualenv.py\", line 996, in create_environment\n    download=download,\n  File \"/usr/lib/python3/dist-packages/virtualenv.py\", line 926, in install_wheel\n    call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)\n  File \"/usr/lib/python3/dist-packages/virtualenv.py\", line 817, in call_subprocess\n    % (cmd_desc, proc.returncode))\nOSError: Command /usr/local/kalite/venv/bin/python2.7 - setuptools pkg_resources pip wheel failed with error code 1\n"}

Does @jvonau have a possible workaround by commenting out all of /etc/pip.conf as follows?

> root@box:~# cat /etc/pip.conf
#[global]
#extra-index-url=https://www.piwheels.org/simple

jvonau, 2020-01-13:
>  New python executable in /usr/local/kalite/venv/bin/python2.7
Also creating executable in /usr/local/kalite/venv/bin/python
Installing setuptools, pkg_resources, pip, wheel... 
Complete output from command /usr/local/kalite/venv/bin/python2.7 - setuptools pkg_resources pip wheel: 
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Looking in links: /usr/lib/python3/dist-packages, /usr/share/python-wheels/\nCollecting setuptools Downloading https://www.piwheels.org/simple/setuptools/setuptools-45.0.0-py2.py3-none-any.whl (583kB)
setuptools requires Python '>=3.5' but the running Python is 2.7.16
...Installing setuptools, pkg_resources, pip, wheel...done.
Running virtualenv with interpreter /usr/bin/python2.7
stderr: Traceback (most recent call last):
File "/usr/lib/python3/dist-packages/virtualenv.py", line 2379, in
main() 
File "/usr/lib/python3/dist-packages/virtualenv.py", line 724, in main
symlink=options.symlink)
File "/usr/lib/python3/dist-packages/virtualenv.py", line 996, in create_environment
download=download 
File "/usr/lib/python3/dist-packages/virtualenv.py", line 926, in install_wheel 
call_subprocess(cmd, show_stdout=False, extra_env=env, stdin=SCRIPT)
File "/usr/lib/python3/dist-packages/virtualenv.py", line 817, in call_subprocess
% (cmd_desc, proc.returncode))
OSError: Command /usr/local/kalite/venv/bin/python2.7 - setuptools pkg_resources pip wheel failed with error code 1

jvonau, 2020-01-13:
https://github.com/piwheels/packages/issues/74

holta, 2020-01-13:
@georgejhunt has added full details of a 4GB RAM RPi 4 that suffered the same problem as my 1GB RAM RPi 4:

http://paste.debian.net/1125855

benjaoming, 2020-01-17:
Also investigating this upstream:

https://github.com/learningequality/ka-lite/pull/5626

holta, 2020-01-23:
@jvonau says good enough!

closed 2020-01-23 by holta