iiab/iiab #2148: http://box/munin shows "403 Forbidden" — until end of iiab-install's Stage 9?

holta, 2020-01-14:
Anybody know why http://box/munin shows "403 Forbidden" right after installing?

But about 5-or-6 Ansible playbooks later, at the very end of iiab-install's Stage 9 (or possibly immediately after that in the 'network' playbook) http://box/munin suddenly later starts working?

Any idea what if anything in IIAB's 'nginx' playbook gets http://box/munin on track?  (Or possibly it's something in the 'network' playbook?)

@tim-moody asks if NGINX's autoindex configuration might be involved, possibly as part of the full 'nginx' playbook running near the end of Stage 9?

holta, 2020-01-14:
FYI on a new install of IIAB (http://download.iiab.io) onto Debian 10.2, http://10.8.0.70/munin suddenly started to work 2min 8sec after Munin's install completed, during Captive Portal installation in Stage 9 here...

>2020-01-14 14:10:15,884 p=root u=14482 | TASK [9-local-addons : Run divert to generate diversion lists for nginx] *******
2020-01-14 14:10:16,502 p=root u=14482 | changed: [127.0.0.1]
2020-01-14 14:10:16,529 p=root u=14482 | TASK [9-local-addons : Run script to generate nginx servers from checkurls input list] ***
2020-01-14 14:10:16,935 p=root u=14482 | changed: [127.0.0.1]
2020-01-14 14:10:16,957 p=root u=14482 | TASK [9-local-addons : Enable nginx to service the sites in checkurls list] ****
2020-01-14 14:10:17,342 p=root u=14482 | changed: [127.0.0.1]

tim-moody, 2020-01-14:
Actually, now that I think about it, I believe it used to take some time for munin to generate its first report in order to have a page to display. We used to have to reboot.

holta, 2020-01-14:
> Actually, now that I think about it, I believe it used to take some time for munin to generate its first report in order to have a page to display. We used to have to reboot.

Here's another data point roughly confirming that: when I just now fresh-installed Internet-in-a-Box (http://download.iiab.io) onto Ubuntu 18.04 &mdash; http://box/munin showed "403 Forbidden" for only 34 seconds &mdash; until this step (different from earlier tests):

> 2020-01-14 16:34:28,076 p=root u=15105 | TASK [munin : Add 'munin' variable values to /etc/iiab/iiab.ini] ***************
...
> 2020-01-14 16:35:01,945 p=root u=15105 | TASK [yarn : Yarn | Update APT cache] ******************************************

So let's declare this as "wontfix" &mdash; as waiting a few minutes (and sometimes as little as ~30 seconds) is OK &mdash; even if the "403 Forbidden" browser message is certainly misleading.

deltacloud9, 2021-07-20:
Currently, running iiab in a VM on my home server. Install went smooth, rebooted and am now receiving this a 403 Forbidden on the box/home page.

I waited a good five minutes and rebooted the VM a few times. Issue seems to be persisting.

Screen is displaying:

403 Forbidden
nginx/1.14.2

holta, 2021-07-20:
@deltacloud9 can you post the **sprunge.us** URL, after you run results from running...?

```
sudo iiab-diagnostics
```

Thanks if poss!

deltacloud9, 2021-07-20:
> @deltacloud9 can you post the **sprunge.us** URL, that results from running...?
> 
> ```
> sudo iiab-diagnostics
> ```
> 
> Thanks if poss!

Thanks for checking this out!

http://sprunge.us/KuLk2s?en

holta, 2021-07-20:
> http://sprunge.us/KuLk2s?en

@jvonau: any idea why "Supply dhcpcd.conf" failed here on this Debian 10 VM?

> 1063 2021-07-19 21:45:44,337 p=3637 u=root n=ansible | TASK [network : Supply dhcpcd.conf] ******************************************************************************************
1064 2021-07-19 21:45:44,454 p=3637 u=root n=ansible | An exception occurred during task execution. To see the full traceback, use -vvv. The error was: ansible.errors.AnsibleUndefinedVariable: 'iiab_wireless_lan_iface' is undefined
1065 2021-07-19 21:45:44,455 p=3637 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "AnsibleUndefinedVariable: 'iiab_wireless_lan_iface' is undefined"}

jvonau, 2021-07-20:
>19 Raspberry Pi reference 2021-01-11
20 Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, bd9d70865f5e4093e7405e94cde11b70c538dc03, stage5
70 OS=raspbian
71 OS_VER=raspbian-10
89 arch = x86_64

I'd say because if the OS is identified as raspbian the assumption is there will be wifi hardware present on that **arm** based platform.

> 1076 FILE DOES NOT EXIST: /opt/iiab/iiab-admin-console/admin-install.log

Don't think "home" was installed yet...

deltacloud9, 2021-07-20:
@jvonau thanks for looking into this.

No worries, I think I will just test with Debian 11 later on.

Thanks again!

jvonau, 2021-07-20:
'Should work' OK on a x86_64 based laptop with wifi hardware available.

holta, 2021-07-20:
@deltacloud9 can you clarify exactly where you downloaded the "2021-01-11" OS from?

Is this "Raspberry Pi Desktop, Compatible with: PC and Mac" https://downloads.raspberrypi.org/rpd_x86/images/rpd_x86-2021-01-12/2021-01-11-raspios-buster-i386.iso ?

i.e. halfway down page https://www.raspberrypi.org/software/operating-systems/ ?

deltacloud9, 2021-07-20:
@holta I downloaded this from the rasberry pi website here: https://www.raspberrypi.org/software/operating-systems/

Yeah, Rasberry Pi Desktop

jvonau, 2021-07-20:
roles/network/templates/network/dhcpcd.conf.j2 would need a conditional around 'denyinterfaces {{ iiab_wireless_lan_iface }}'
 or conditionals around the stanza supplying dhcpcd.conf.j2

jvonau, 2021-07-20:
First time in 3-5 years that somebody ran raspbian on hardware without wifi, should of hit this with early rpi's that just have ethernet aswell but that is not on @holta or my usual test matrix.

deltacloud9, 2021-07-20:
@holta and @jvonau what is the recommended OS to use on a VM setup?

jvonau, 2021-07-20:
I'd lean towards Ubuntu-20.04 or later OR Debian-10 or later. Either one with a desktop you have a choice of different front-ends (gnome, mate, xfce, etc) most use Network-Manager as the network back-end which I support the best I can, others I can work on upon request and only if my interest in the back-end used looks do-able me.

holta, 2021-07-20:
> First time in 3-5 years that somebody ran raspbian on hardware without wifi, should of hit this with early rpi's that just have ethernet aswell but that is not on @holta or my usual test matrix.

Exactly.  Nevertheless, thanks @deltacloud9 for having made this happen.  Probably we should try to fix this.

> @holta and @jvonau what is the recommended OS to use on a VM setup?

@deltacloud9 if you can test ["fully frozen"](https://lists.debian.org/debian-devel-announce/2021/07/msg00002.html) Debian 11 Bullseye's latest daily, that would be especially helpful:

http://cdimage.debian.org/cdimage/daily-builds/daily/arch-latest/amd64/iso-cd/debian-testing-amd64-netinst.iso

It should work.  But there can be small glitches even during its very final days prior to release:

- #2749

jvonau, 2021-07-20:
> Exactly. Nevertheless, thanks @deltacloud9 for having made this happen. Probably we should try to fix this.
 No PR forthcoming, see the recommended change above.

holta, 2021-07-20:
> roles/network/templates/network/dhcpcd.conf.j2 would need a conditional around 'denyinterfaces {{ iiab_wireless_lan_iface }}'
> or conditionals around the stanza supplying dhcpcd.conf.j2

@deltacloud9 just to highlight Jerry's answer above.  Can you consider tweaking [/opt/iiab/iiab/roles/network/templates/network/dhcpcd.conf.j2](https://github.com/iiab/iiab/blob/master/roles/network/templates/network/dhcpcd.conf.j2) as he suggests?

If so &mdash; then test your change as follows:

```
cd /opt/iiab/iiab
sudo ./iiab-network
```

If `./iiab-network` completes, you're probably done!  (And can then finalize your IIAB installation by re-running `sudo iiab`)

holta, 2021-07-20:
> or conditionals around the stanza supplying dhcpcd.conf.j2

@jvonau should both of these be make conditional?

- roles/network/tasks/debian.yml:37:    src: network/dhcpcd.conf.j2
- roles/network/tasks/rpi_debian.yml:21:    src: network/dhcpcd.conf.j2

Or would it be safer/wiser to make a change in just 1 place, e.g. below?

- roles/network/templates/network/dhcpcd.conf.j2: denyinterfaces {{ iiab_wireless_lan_iface }}

deltacloud9, 2021-07-20:
> > roles/network/templates/network/dhcpcd.conf.j2 would need a conditional around 'denyinterfaces {{ iiab_wireless_lan_iface }}'
> > or conditionals around the stanza supplying dhcpcd.conf.j2
> 
> @deltacloud9 just to highlight Jerry's answer above. Can you consider tweaking [/opt/iiab/iiab/roles/network/templates/network/dhcpcd.conf.j2](https://github.com/iiab/iiab/blob/master/roles/network/templates/network/dhcpcd.conf.j2) as he suggests?
> 
> If so — then test your change as follows:
> 
> ```
> cd /opt/iiab/iiab
> sudo ./iiab-network
> ```
> 
> If `./iiab-network` completes, you're probably done! (And can then finalize your IIAB installation by re-running `sudo iiab`)

I deleted this VM to test my Debian 11 build. I can spin another one up if needed though.

jvonau, 2021-07-20:
> Or would it be safer/wiser to make a change in just 1 place, e.g. below?
> 
>     * [roles/network/templates/network/dhcpcd.conf.j2:41](https://github.com/iiab/iiab/blob/master/roles/network/templates/network/dhcpcd.conf.j2#L41): denyinterfaces {{ iiab_wireless_lan_iface }}

Would cover both with 'if defined', like how iiab_wired_lan_iface is handled below.

jvonau, 2021-07-21:
close

closed 2021-07-21 by holta