iiab/iiab #2152: Raise NGINX proxy timeout(s) from 60 sec--similar to Apache--to reduce "capacity" failures during Kiwix search etc

holta, 2020-01-15:
Building on PR #2151, towards mitigating the worst classroom failures of things like Kiwix search as mentioned in PR #2063 and kiwix/kiwix-tools#345.

Summary possible action plan, that I've outlined from @jvonau's suggestions:
https://github.com/iiab/iiab/blob/master/roles/httpd/templates/010-iiab.conf.j2#L46-L56

holta, 2020-01-23:
Much discussed during our 2020-01-23 call.  @jvonau will propose a PR.

holta, 2020-02-01:
Let's move the discussion from #2063 to here (#2152), to see if these timeout tweaks help solve this/these quite serious ongoing Kiwix problem(s), that basically every school seems to be experiencing!

Ref: kiwix/kiwix-tools#345

tim-moody, 2020-02-03:
## Overview

In an attempt to diagnose the various Bad Gateway or Gateway Time-out errors I have done a great deal of experimentation with a load generator. For a given level of simultaneous kiwix requests it generates a mixed load of 50% random and 50% search from a fixed list of terms. Kiwix in general and especially search is cpu bound, so cpu very much matters as kiwix pins any available cores at near 100% when working on a search.

I tried various setting from 5 simultaneous requests to 100 on a nuc and on a rpi 4 with 2G ram. I tested with kiwix at 3, 4, and 15 threads and found no material difference. nginx worker processes are only about 0.7% cpu, so they manage to sneak in even if kiwix has all cores available. But if kalite or some other process needs more than that it could make a difference and 3 kiwix threads may be a better choice.

I tried changing the nginx proxy timeout settings, but found no substantial difference between setting them as high as 600 seconds and as low as 5, which should cause many timeouts but doesn't. I conclude that the default is fine.

Statistics are below, but I found that on the rpi 4 with 2G ram 40 simultaneous requests can operate mostly with no timeouts. Above that they begin to appear and are almost 15% by 100 requests. With the load generator running at 40 requests I also did manual access to kiwix for both search and page access and got no timeouts.

This is all difficult to reconcile with the initial report that timeouts were happening with one or two people on the system. We don't know what else was going on. Also, this report was for a RPi 3 which has less than 1/4 the computing power of RPi 4. So if that translates into a maximum error free request rate of 10, it starts to look more plausible.

https://images.ctfassets.net/tvfg2m04ppj4/yxGvWNAyseuYEl4KzqT2u/37862ad163077129b1f2e5865fc39d8a/Raspberry_Pi_4_Benchmarks.jpg

## Statistics

In the following the initial n/m is the current request vs total elapsed time for all requests or sometimes just the latter,
err/req: 0/500 means zero errors in 500 total requests,
followed by the resulting title (e.g.Search: pueblo),
followed by response time for this request, average and maximum response times.

### Against RPi 4

Req|Results|Comments
---|-------|--------
5|run: 1/343s err/req: 0/500 Search: pueblo (resp: 4s, avg: 3s, max: 15s)|Early resp of 13s and 15s killed avg
10|run: 1/347s err/req: 0/502 Melgar de Fernamental (resp: 6s, avg: 7s, max: 15s)|Early max is 10s. kiwix now primed?
15|run: 1/340s err/req: 0/501 Search: pueblo (resp: 5s, avg: 10s, max: 28s)	
20|run: 1/385s err/req: 0/503 Search: mexico (resp: 6s, avg: 15s, max: 93s)|Why such a high max?
30|run: 1/366s err/req: 0/502 Copa Mundial de Fútbol de 1970 (resp: 14s, avg: 21s, max: 45s)	
40|run: 1/356s err/req: 1/500 Search: futbol (resp: 26s, avg: 28s, max: 96s)|1 Bad Gateway. Restarted kiwix prior to run.
50|run: 1/382s err/req: 10/500 Rosie MacLennan (resp: 29s, avg: 37s, max: 137s)|10 Bad Gateway. At about 179 a random took 104s. (jumping up from 69s previous max.)
100|run: 1/354s err/req: 63/501 Search: amlo (resp: 125s, avg: 59s, max: 237s)|63 Bad GW, but earlier runs had 2 to 3 times that. Restarted kiwix prior to run. Bad Gateway on first request

### Against a NUC

Req|Results|Comments
---|-------|--------
5|sec: 202; err/req: 0/500 Search: pueblo (resp: 2s, avg: 2s, max: 5s)	
10|sec: 204; err/req: 0/502 Search: pueblo (resp: 3s, avg: 4s, max: 12s)	
15|sec: 190; err/req: 0/505 Search: mexico (resp: 4s, avg: 5s, max: 27s)	
20|sec: 189; err/req: 0/502 Search: mexico (resp: 4s, avg: 7s, max: 21s)	
25|sec: 189; err/req: 0/502 Arquitectura islámica (resp: 13s, avg: 9s, max: 76s)	
30|run: 1/186s err/req: 0/500 Search: pueblo (resp: 4s, avg: 11s, max: 28s)		
50|run: 1/230s err/req: 2/501 Emerging and Growth-Leading Economies (resp: 137s, avg: 18s, max: 137s)|Both bad gateways were at end (because of long timeout?) Max resp was 82 before them.
100|sec: 238; err/req: 7/502 Gobernantes de Guatemala (resp: 133s, avg: 36s, max: 206s)|Saw one network EMPTY RESPONSE

tim-moody, 2020-02-04:
# Correction

I retested with all proxy timeouts set to 5 seconds and 30 simultaneous requests, and now almost all requests timeout, so I conclude I was wrong about the timeout settings having no effect at low settings. However the defaults work for the useful range of up to 40 requests.

Accordingly, we could change kiwix-nginx.conf as it may help and probably does no harm.

holta, 2020-02-14:
FYI @tim-moody posted a new/enhanced [kiwix-load-test.js](https://github.com/iiab/iiab-factory/blob/master/testing/kiwix-load-test.js), updating the prior version from 2019-12-19.

PR https://github.com/iiab/iiab-factory/pull/93

holta, 2020-03-30:
@tim-moody are we in good shape here, for IIAB 7.1's release in about a month?

(Should this be moved to the [IIAB 7.2 Milestone](https://github.com/iiab/iiab/milestone/7) if so?)

tim-moody, 2020-03-30:
fixed in #2235

closed 2020-03-30 by holta