iiab/iiab #2154: Stub /etc/nginx/conf.d/nodered-nginx.conf to port Node-RED to NGINX?

holta, 2020-01-16:
@m-anish as suggested by @jvonau if either of you have hopefully validate this?

```
location /nodered/ {
#  proxy_pass http://127.0.0.1:8090/nodered;
  proxy_pass http://127.0.0.1:1880/nodered/;
}

location /nodered/debug/ws/ {
  proxy_pass http://127.0.0.1:1880/nodered/debug/ws;
}

location /nodered/comms/ {
  proxy_pass http://127.0.0.1:1880/nodered/comms;
}
```

holta, 2020-01-16:
In any case, let's build off the minimal [nodered/templates/nodered-nginx.conf.j2](https://github.com/iiab/iiab/blob/master/roles/nodered/templates/nodered-nginx.conf.j2) here:

```
location /nodered/ {
  proxy_pass http://127.0.0.1:{{ apache_port }}/nodered/;
}
```

holta, 2020-01-23:
@m-anish please review this...@jvonau has done basic tests...can you test for corner cases etc?

jvonau, 2021-07-20:
close

closed 2021-07-20 by holta