iiab/iiab #2157: Cham: How do we make it alert user incase we run out of storage.

m-anish, 2020-01-16:
I would like inputs on how we want to handle this before spending any energy on developing a solution. 

What do we want to protect against?
* That a single unattended live stream might go on for so long that it may eat up all the space on a server?
* That multiple stream archives accumulated over a period of time might make the server run out of storage space. 

Which of these (or both) problems do we foresee, and how do we want to fix them w.r.t. user experience. 

Inputs welcome...

p.s. the default streaming datarates on the server are low, however a 'source' quality stream is also recorded - this is the quality at which a streaming computer streams a rtmp stream to the server. This may be uncapped in bitrate.

holta, 2020-01-16:
#1743

holta, 2020-01-17:
The most valuable thing we can do is put an "intentionally annoying red message" at the top of http://box.lan alerting all users when there's less than 1GB of disk available...

Sites with UGC (user-generated content) and similar (Nextcloud, WordPress, etc) could also benefit with a softer (yellow?) cautionary DEW (distant early warning) banner e.g. when there's less than 10GB of disk available?

Either way, strategies like this (can) empower field communities to learn from their own disk-hoarding mistakes... *so long as the delete-old-videos button is never far away!*

m-anish, 2020-01-17:
I guess then better to fix this problem globally first (homepage) and specifically later (individual services). 

Are we generally in agreement on that?

mitra42, 2020-01-27:
Another option might be by default to use a different disk specifically for this. We use a strategy of looking for "archiveorg" top levels on mounted volumes. Maybe some variant of that would be useful.

holta, 2020-01-30:
Cham streaming is not the only "disk discipline" question affecting popular IIAB servers, but just so don't we don't lose track...here's the larger/original ticket's design challenge:

https://github.com/eka-foundation/home/issues/31 "Discoverable, decentralized, cached live video streaming"