iiab/iiab #2162: What is current status of Menu item call to Maps osm ?

llovemixo, 2020-01-17:
### Expected behavior and actual behavior.

### Steps to reproduce the problem.

### Specifications like the version of the IIAB, operating system version, or hardware details.

holta, 2020-01-17:
@georgejhunt can you help answer Nathan Riddle?

holta, 2020-01-17:
@llovemixo have you read over https://github.com/iiab/iiab/wiki/IIAB-Maps just to make sure basic questions are squared away?

llovemixo, 2020-01-18:
It reduces to  issue 1496, your comment of 7 July 2019.  My use fell in time of your modifications. 
Asking if that issue actually resolved.
Appears that the efforts on that issue stepped on Tim's vision  of menu items , issue 1724, June 14 comment.
The menu items for osm maps all fail since all calls are redirected to osm-vector-maps/ rather than the map packs (en-osm*) within it, which is specified in the menu item's fields and menu definition .json .
My last install was at end of Dec.

holta, 2020-01-18:
@llovemixo #1496 is largely resolved, even if we're not 100% there yet in terms of UX (usability) for new implementers especially!

Keep on us and @georgejhunt @tim-moody @jvonau @m-anish @avni & I will help you along the way!

ASIDE: IIAB 7.2 in Q2 2020 should see some major advances in terms of map (OpenStreetMap & associated satellite photos) usability, for kids/teachers and also implementers.  However I don't expect significant changes for February 2020's [IIAB 7.1](https://github.com/iiab/iiab/milestone/6) at this point.  But things can always change, pending your input and other field communities' priorities, as we understand more of your specific "fieldback" needs ~

closed 2020-01-19 by llovemixo