iiab/iiab #2164: NGINX: MediaWiki has broken & quirky links

holta, 2020-01-17:
1) After you enter Admin/changeme2020 to try to login to http://box/wiki it erroneously redirects to broken link http://localhost/wiki/Main_Page
   - If you then manually force your browser to http://box/wiki/Main_Page (for any valid value of 'box') most MediaWiki links then work...
2) The 'Random page' link in the left margin erroneously redirects from http://box/wiki/Special:Random to broken link http://localhost/wiki/Main_Page as in (1) above.
3) These links appear quirky...containing "/mwlink/" instead of "/wiki/" in the middle of the URL...is this really what we want?
   - Top...
      - Edit http://box/mwlink/index.php?title=Main_Page&action=edit
      - View history http://box/mwlink/index.php?title=Main_Page&action=history
   - Lower Left...
      - Printable version http://box/mwlink/index.php?title=Main_Page&printable=yes
      - Permanent link http://box/mwlink/index.php?title=Main_Page&oldid=1
      - Page information http://box/mwlink/index.php?title=Main_Page&action=info
   - Lower Right...
      - Mark this page as patrolled http://box/mwlink/index.php?title=Main_Page&action=markpatrolled&rcid=1

See:
https://github.com/iiab/iiab/blob/master/roles/mediawiki/templates/mediawiki-nginx.conf.j2

Refs:
PR #2080 "change services to nginx and simplify php"
PR #2086 "Native nginx services"
PR #2142 "Fix http://box/mediawiki"
PR #2159 "MediaWiki [...] disallows password 'changeme' so for now let's use 'changeme2020'"

holta, 2020-01-23:
Discussed on our weekly call (http://minutes.iiab.io)

@tim-moody will look into this...

tim-moody, 2020-01-23:
probably the issue is in /library/mediawiki-1.34.0/LocalSettings.php:

## The protocol and server name to use in fully-qualified URLs
$wgServer = "http://localhost";

tim-moody, 2020-01-23:
https://www.mediawiki.org/wiki/Manual:$wgServer 

> Please make sure that $wgServer is not set to http://localhost or //localhost! If it is, a number of things will break, e.g. links inside your wiki will not work, stylesheets will not display properly and many more. Set it to the actual base URL instead!

tim-moody, 2020-01-23:
$wgServer = "//" . $_SERVER["HTTP_HOST"];

holta, 2020-01-23:
> 1. After you enter Admin/changeme2020 to try to login to http://box/wiki it erroneously redirects to broken link http://localhost/wiki/Main_Page
>    
>    * If you then manually force your browser to http://box/wiki/Main_Page (for any valid value of 'box') most MediaWiki links then work, however...
> 2. The 'Random page' link in the left margin erroneously redirects from http://box/wiki/Special:Random to broken link http://localhost/wiki/Main_Page as in (1) above.

@tim-moody are both of the above solved by PR #2188?

tim-moody, 2020-01-23:
neither 1 nor 2 happen. however random always goes to main, though there aren't many pages

tim-moody, 2020-01-23:
actually now random went to a Test page I created, so maybe works. at least doesn't go  to localhost

holta, 2020-01-28:
@mitra42 you raised some concerns privately about IIAB's MediaWiki implementation...after your working on that for a day or so...if possible, can you prioritize those questions and make them precise on this ticket, or new ticket if that's a better place?

Thanks Much, as I was going to close this ticket thanks to @tim-moody's hard work here in recent weeks, but perhaps there's more needed here &mdash; and ideally you might have some NGINX suggestions for us here?!

mitra42, 2020-01-28:
@holta - I saw this after posting a new issue at #2212

holta, 2020-01-28:
See @tim-moody PR #2213 "Mitra's tweaks to mediawiki install" which further refines MediaWiki, with several of @mitra42's recommendations from #2212 "MediaWiki - tweaks ?"

tim-moody, 2020-01-30:
closing. reopen if necessary.

closed 2020-01-30 by tim-moody