iiab/iiab #2168: Captive Portal unable to create /var/log/nginx/portal.log (Permission Denied)

holta, 2020-01-18:
@georgejhunt 

1) I've seen this on 2 different RPi 4's now.  Is `/opt/iiab/captiveportal/users.sqlite` missing for the same underlying reason possibly?

```
root@box:~# systemctl status uwsgi-captiveportal.service
● uwsgi-captiveportal.service - uWSGI Service
   Loaded: loaded (/etc/systemd/system/uwsgi-captiveportal.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2020-01-18 00:32:55 EST; 11min ago
 Main PID: 6398 (uwsgi)
   Status: "uWSGI is ready"
    Tasks: 3 (limit: 4035)
   Memory: 11.0M
   CGroup: /system.slice/uwsgi-captiveportal.service
           ├─6398 /usr/bin/uwsgi --ini /opt/iiab/captiveportal/captiveportal.ini
           └─6399 /usr/bin/uwsgi --ini /opt/iiab/captiveportal/captiveportal.ini

Jan 18 00:32:55 box.lan uwsgi[6398]:   File "/usr/lib/python3.7/logging/__init__.py", line 1121, in _open
Jan 18 00:32:55 box.lan uwsgi[6398]:     return open(self.baseFilename, self.mode, encoding=self.encoding)
Jan 18 00:32:55 box.lan uwsgi[6398]: PermissionError: [Errno 13] Permission denied: '/var/log/nginx/portal.log'
Jan 18 00:32:55 box.lan uwsgi[6398]: unable to load app 0 (mountpoint='') (callable not found or import error)
Jan 18 00:32:55 box.lan uwsgi[6398]: *** no app loaded. going in full dynamic mode ***
Jan 18 00:32:55 box.lan uwsgi[6398]: *** uWSGI is running in multiple interpreter mode ***
Jan 18 00:32:55 box.lan uwsgi[6398]: spawned uWSGI master process (pid: 6398)
Jan 18 00:32:55 box.lan systemd[1]: Started uWSGI Service.
Jan 18 00:32:55 box.lan uwsgi[6398]: spawned uWSGI worker 1 (pid: 6399, cores: 1)
Jan 18 00:32:55 box.lan uwsgi[6398]: Python auto-reloader enabled
```

2) `./runrole captiveportal` doesn't work to install /opt/iiab/captiveportal/users.sqlite even when `captiveportal_installed: True` is removed from /etc/iiab/iiab_state.yml and `systemctl daemon-reload` is run, etc.

   BUT `./iiab-install --reinstall` does seem work to install & enable Captive Portal...on a 2nd pass anyway?  Any ideas why?  Originally reported @ https://github.com/iiab/iiab/pull/2158#issuecomment-575735066

3) Even though these 2 log files had their ownership set to www-data:www-data & permissions set '0770' within [2-common/tasks/fl.yml#L41-L50](https://github.com/iiab/iiab/blob/master/roles/2-common/tasks/fl.yml#L41-L50) it has since changed, presumably during the installation of Apache & NGINX themselves...to:

   ```
   root@box:~# ls -ld /var/log/apache2/ /var/log/nginx/
   drwxr-x--- 2 www-data www-data 4096 Jan 17 17:16 /var/log/apache2/
   drwxr-xr-x 2 root     adm      4096 Jan 17 17:28 /var/log/nginx/
   ```

jvonau, 2020-01-18:
Chat excerpt: 
> Jerry: why does a non-nginx (standalone) program log to a nginx owned directory? Think a better place for CP to store its logs would be /opt/iiab/captiveportal being the easiest or /var/log/captiveportal requiring creation of the directory.
Tim: never said should go to nginx. only that ownership can be changed to solve permissions
 would prefer var/logs

related https://github.com/iiab/iiab/pull/2121
This logging worked under apache as CP was called by apache as a cgi-bin. Today CP is started via a systemd file as a standalone program.

Chat excerpt: 
> Jerry: The issue I see with CP logging to nginx is logrotate knows nothing about CP being active and should be restarted/reloaded after the logs are moved by logrotate, have a look at other stock files in /etc/logrotate.d/. I see that httpd-prerotate looks to be the place to add CP restart but correcting the log location and adding a correct logrotate file for CP would be preferred.

holta, 2020-01-23:
Solved by PRs #2171 #2185

closed 2020-01-23 by holta