iiab/iiab #2170: @jvonau questions re: uWSGI use by (1) NGINX (2) Captive Portal (3) Admin Console

holta, 2020-01-19:
@jvonau can explain infinitely better than I &mdash; so these questions are *just place markers for now* so we don't lose his insights &mdash; and then investigate patiently in coming weeks:

- [nginx/tasks/install.yml#L30](https://github.com/iiab/iiab/blob/master/roles/nginx/tasks/install.yml#L30) installs [template roles/nginx/templates/uwsgi.service](https://github.com/iiab/iiab/blob/master/roles/nginx/templates/uwsgi.service) thereby overriding stock unit file `/etc/systemd/system/uwsgi.service` &mdash; altering how all other uwsgi apps might be implemented &mdash; e.g. Admin Console?
   - Might this cause intermittent Admin Console problems, e.g. after 2nd run of certain install scripts?
   - Should different unit file names be used, e.g. something like iiab-admin-console.service similar to Captive Portal's [uwsgi-captiveportal.service](https://github.com/iiab/iiab/blob/master/roles/captiveportal/templates/uwsgi-captiveportal.service) ?
- Is it possibly important that uWSGI only be started/run once &mdash; with apps spawned as workers under the main thread?
   - @jvonau suggests that "anything in `/etc/uwsgi/apps-enabled/` should be spawned when [uwsgi.service] is restarted under the stock layout by upstream"
   - Are there any other/similar designs we should be considering?

(Again: these are just place markers so we don't lose Jerry's questions, and together verify our designs in coming weeks!)

Possibly Related:

#2168 "Captive Portal unable to create (1) /var/log/nginx/portal.log - Permission Denied (2) /opt/iiab/captiveportal/users.sqlite"

jvonau, 2020-01-19:
cat /etc/uwsgi/apps-enabled/README
>Some files found in this directory are processed by uWSGI init.d script as
uWSGI configuration files.
On system boot for each configuration file new uWSGI daemon instance is started
with additional option. Name of this option is based on configuration file
extension. Path to configuration files is passed as option value.
See more detailed information at:
  /usr/share/doc/uwsgi/README.Debian.gz
  /etc/default/uwsgi

cat /etc/default/uwsgi  with removed #  and << variable to point out code in use
 >Defaults for uWSGI initscript
 sourced by /etc/init.d/uwsgi
 If you don't want uWSGI to be started as a system service,
 disable it using the command 'update-rc.d uwsgi disable'
 At startup VERBOSE value is setted in 'no'. So when user invokes
 uWSGI init.d script, no output is showed.
 It could be unexpected behaviour, because it is common practice for
 init.d script to ignore VERBOSE value.
 Here VERBOSE is overriden to conform such the practice.
VERBOSE=yes                             << variable
Should init.d script print configuration file names while marking progress of
it's execution?

>If 'no', then init.d script prints one-character symbols instead file names.
Printing confnames is quite informative, but could mess terminal output or
be somewhat dangerous (as filename could contain arbitary characters).
ASCII control characters in file names are replaced with '?' in init.d script
output, nevertheless you were warned.
PRINT_CONFNAMES_IN_INITD_SCRIPT_OUTPUT=no          << variable
init.d script starts instance of uWSGI daemon for each found user-created
configuration file.

>Options from inherited configuration file are passed to each instance by
default. They could be overrided (or extended) by user configuration file.
INHERITED_CONFIG=/usr/share/uwsgi/conf/default.ini            << variable 

Remember systemd would run /etc/init.d/uwsgi directly if the override /etc/systemd/system/uwsgi.service file is not present. This was the behavior prior to captiveportal and was working well.

holta, 2020-01-19:
Is it possible #2149 "Fix for Admin Console login popup loop: install NGINX after Apache & MySQL" could have anything to do with these uWSGI issues?

jvonau, 2020-01-19:
Think part of the confusion is the nginx role holds templates for admin-console.ini that was moved, the real installation is occurring in admin-console's [install](https://github.com/iiab/iiab-admin-console/blob/master/roles/console/tasks/main.yml) and the systemd call at the end.

jvonau, 2020-01-19:
with override in place
>  pi@box:/opt/iiab/iiab $ systemctl status uwsgi
● uwsgi.service - uWSGI Service
   Loaded: loaded (/etc/systemd/system/uwsgi.service; disabled; vendor preset: enabled)
   Active: active (running) since Thu 2020-01-02 19:15:45 GMT; 2 weeks 2 days ago
 Main PID: 6321 (uwsgi)
   Status: "uWSGI is ready"
    Tasks: 2 (limit: 4915)
   Memory: 8.5M
   CGroup: /system.slice/uwsgi.service
           ├─6321 /usr/bin/uwsgi --ini /etc/uwsgi/apps-enabled/admin-console.ini
           └─6322 /usr/bin/uwsgi --ini /etc/uwsgi/apps-enabled/admin-console.ini

sudo systemctl stop uwsgi
sudo mv /etc/systemd/system/uwsgi.service /etc/systemd/system/uwsgi.service.disabled
sudo systemctl daemon-reload
sudo systemctl start uwsgi

>sudo systemctl status uwsgi
● uwsgi.service - LSB: Start/stop uWSGI server instance(s)
   Loaded: loaded (/etc/init.d/uwsgi; generated)
   Active: active (running) since Sun 2020-01-19 04:17:24 GMT; 2min 22s ago
     Docs: man:systemd-sysv-generator(8)
  Process: 28892 ExecStart=/etc/init.d/uwsgi start (code=exited, status=0/SUCCESS)
    Tasks: 3 (limit: 4915)
   Memory: 8.6M
   CGroup: /system.slice/uwsgi.service
           ├─28993 /usr/bin/uwsgi --ini /usr/share/uwsgi/conf/default.ini --ini /etc/uwsgi/apps-enabled/admin-console.ini --daemonize /var/log/uwsgi/app/admin-console.log
           ├─29002 /usr/bin/uwsgi --ini /usr/share/uwsgi/conf/default.ini --ini /etc/uwsgi/apps-enabled/admin-console.ini --daemonize /var/log/uwsgi/app/admin-console.log
           └─29003 /usr/bin/uwsgi --ini /usr/share/uwsgi/conf/default.ini --ini /etc/uwsgi/apps-enabled/admin-console.ini --daemonize /var/log/uwsgi/app/admin-console.log
Jan 19 04:17:23 box.lan systemd[1]: Starting LSB: Start/stop uWSGI server instance(s)...
Jan 19 04:17:24 box.lan uwsgi[28892]: Starting app server(s): uwsgi -> . done.
Jan 19 04:17:24 box.lan systemd[1]: Started LSB: Start/stop uWSGI server instance(s).

jvonau, 2020-01-19:
> pi@box:/opt/iiab/iiab $ sudo systemctl status uwsgi-captiveportal.service 
● uwsgi-captiveportal.service - uWSGI Service
   Loaded: loaded (/etc/systemd/system/uwsgi-captiveportal.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2020-01-13 18:22:32 GMT; 5 days ago
 Main PID: 13744 (uwsgi)
   Status: "uWSGI is ready"
    Tasks: 3 (limit: 4915)
   Memory: 11.2M
   CGroup: /system.slice/uwsgi-captiveportal.service
           ├─13744 /usr/bin/uwsgi --ini /opt/iiab/captiveportal/captiveportal.ini
           └─13745 /usr/bin/uwsgi --ini /opt/iiab/captiveportal/captiveportal.ini

sudo systemctl stop uwsgi-captiveportal.service
sudo mv /etc/systemd/system/uwsgi-captiveportal.service /etc/systemd/system/uwsgi-captiveportal.service.disabled
sudo cp /opt/iiab/captiveportal/captiveportal.ini /etc/uwsgi/apps-enabled/
sudo systemctl daemon-reload
sudo systemctl restart uwsgi
> sudo systemctl status uwsgi
● uwsgi.service - LSB: Start/stop uWSGI server instance(s)
   Loaded: loaded (/etc/init.d/uwsgi; generated)
   Active: active (running) since Sun 2020-01-19 04:30:51 GMT; 30s ago
     Docs: man:systemd-sysv-generator(8)
  Process: 31652 ExecStart=/etc/init.d/uwsgi start (code=exited, status=0/SUCCESS)
    Tasks: 8 (limit: 4915)
   Memory: 21.3M
   CGroup: /system.slice/uwsgi.service
           ├─31753 /usr/bin/uwsgi --ini /usr/share/uwsgi/conf/default.ini --ini /etc/uwsgi/apps-enabled/admin-console.ini --daemonize /var/log/uwsgi/app/admin-console.log
           ├─31840 /usr/bin/uwsgi --ini /usr/share/uwsgi/conf/default.ini --ini /etc/uwsgi/apps-enabled/captiveportal.ini --daemonize /var/log/uwsgi/app/captiveportal.log
           ├─31846 /usr/bin/uwsgi --ini /usr/share/uwsgi/conf/default.ini --ini /etc/uwsgi/apps-enabled/admin-console.ini --daemonize /var/log/uwsgi/app/admin-console.log
           ├─31847 /usr/bin/uwsgi --ini /usr/share/uwsgi/conf/default.ini --ini /etc/uwsgi/apps-enabled/admin-console.ini --daemonize /var/log/uwsgi/app/admin-console.log
           ├─31851 /usr/bin/uwsgi --ini /usr/share/uwsgi/conf/default.ini --ini /etc/uwsgi/apps-enabled/captiveportal.ini --daemonize /var/log/uwsgi/app/captiveportal.log
           └─31852 /usr/bin/uwsgi --ini /usr/share/uwsgi/conf/default.ini --ini /etc/uwsgi/apps-enabled/captiveportal.ini --daemonize /var/log/uwsgi/app/captiveportal.log
Jan 19 04:30:50 box.lan systemd[1]: Starting LSB: Start/stop uWSGI server instance(s)...
Jan 19 04:30:51 box.lan uwsgi[31652]: Starting app server(s): uwsgi -> . . done.
Jan 19 04:30:51 box.lan systemd[1]: Started LSB: Start/stop uWSGI server instance(s).

jvonau, 2020-01-19:
now adjust the logging to go to  /var/log/uwsgi/app/captiveportal.log in [capture-wsgi.py.j2](https://github.com/iiab/iiab/blob/master/roles/captiveportal/templates/capture-wsgi.py.j2)

jvonau, 2020-01-19:
I'll go do a PR...

jvonau, 2020-01-19:
>sudo cat /var/log/uwsgi/app/captiveportal.log
Sun Jan 19 04:30:51 2020 - *** Starting uWSGI 2.0.18-debian (32bit) on [Sun Jan 19 04:30:51 2020] ***
Sun Jan 19 04:30:51 2020 - compiled with version: 8.2.0 on 10 February 2019 02:42:46
Sun Jan 19 04:30:51 2020 - os: Linux-4.19.79-v7l+ #1273 SMP Fri Oct 11 18:23:45 BST 2019
Sun Jan 19 04:30:51 2020 - nodename: box.lan
Sun Jan 19 04:30:51 2020 - machine: armv7l
Sun Jan 19 04:30:51 2020 - clock source: unix
Sun Jan 19 04:30:51 2020 - pcre jit disabled
Sun Jan 19 04:30:51 2020 - detected number of CPU cores: 4
Sun Jan 19 04:30:51 2020 - current working directory: /
Sun Jan 19 04:30:51 2020 - writing pidfile to /run/uwsgi/app/captiveportal/pid
Sun Jan 19 04:30:51 2020 - detected binary path: /usr/bin/uwsgi-core
Sun Jan 19 04:30:51 2020 - chdir() to /opt/iiab/captiveportal
Sun Jan 19 04:30:51 2020 - your processes number limit is 29184
Sun Jan 19 04:30:51 2020 - your memory page size is 4096 bytes
Sun Jan 19 04:30:51 2020 - detected max file descriptor number: 1024
Sun Jan 19 04:30:51 2020 - lock engine: pthread robust mutexes
Sun Jan 19 04:30:51 2020 - thunder lock: disabled (you can enable it with --thunder-lock)
Sun Jan 19 04:30:51 2020 - uwsgi socket 0 bound to UNIX address /run/uwsgi/app/captiveportal/socket fd 3
Sun Jan 19 04:30:51 2020 - uwsgi socket 1 bound to TCP address :9090 fd 5
Sun Jan 19 04:30:51 2020 - setgid() to 33
Sun Jan 19 04:30:51 2020 - set additional group 42 (shadow)
Sun Jan 19 04:30:51 2020 - setuid() to 33
Sun Jan 19 04:30:51 2020 - Python version: 3.7.3 (default, Apr  3 2019, 05:39:12)  [GCC 8.2.0]
Sun Jan 19 04:30:51 2020 - Python main interpreter initialized at 0xbf71d8
Sun Jan 19 04:30:51 2020 - python threads support enabled
Sun Jan 19 04:30:51 2020 - your server socket listen backlog is limited to 100 connections
Sun Jan 19 04:30:51 2020 - your mercy for graceful operations on workers is 60 seconds
Sun Jan 19 04:30:51 2020 - mapped 193200 bytes (188 KB) for 2 cores
Sun Jan 19 04:30:51 2020 - *** Operational MODE: preforking ***
Sun Jan 19 04:30:51 2020 - WSGI app 0 (mountpoint='') ready in 0 seconds on interpreter 0xbf71d8 pid: 31840 (default app)
Sun Jan 19 04:30:51 2020 - *** uWSGI is running in multiple interpreter mode ***
Sun Jan 19 04:30:51 2020 - spawned uWSGI master process (pid: 31840)
Sun Jan 19 04:30:51 2020 - spawned uWSGI worker 1 (pid: 31851, cores: 1)
Sun Jan 19 04:30:51 2020 - spawned uWSGI worker 2 (pid: 31852, cores: 1)
Sun Jan 19 04:30:51 2020 - Python auto-reloader enabled

>pi@box:/opt/iiab/iiab $ sudo cat /var/log/uwsgi/app/admin-console.log
Sun Jan 19 04:30:51 2020 - *** Starting uWSGI 2.0.18-debian (32bit) on [Sun Jan 19 04:30:51 2020] ***
Sun Jan 19 04:30:51 2020 - compiled with version: 8.2.0 on 10 February 2019 02:42:46
Sun Jan 19 04:30:51 2020 - os: Linux-4.19.79-v7l+ #1273 SMP Fri Oct 11 18:23:45 BST 2019
Sun Jan 19 04:30:51 2020 - nodename: box.lan
Sun Jan 19 04:30:51 2020 - machine: armv7l
Sun Jan 19 04:30:51 2020 - clock source: unix
Sun Jan 19 04:30:51 2020 - pcre jit disabled
Sun Jan 19 04:30:51 2020 - detected number of CPU cores: 4
Sun Jan 19 04:30:51 2020 - current working directory: /
Sun Jan 19 04:30:51 2020 - writing pidfile to /run/uwsgi/app/admin-console/pid
Sun Jan 19 04:30:51 2020 - detected binary path: /usr/bin/uwsgi-core
Sun Jan 19 04:30:51 2020 - chdir() to /opt/admin/console
Sun Jan 19 04:30:51 2020 - your processes number limit is 29184
Sun Jan 19 04:30:51 2020 - your memory page size is 4096 bytes
Sun Jan 19 04:30:51 2020 - detected max file descriptor number: 1024
Sun Jan 19 04:30:51 2020 - lock engine: pthread robust mutexes
Sun Jan 19 04:30:51 2020 - thunder lock: disabled (you can enable it with --thunder-lock)
Sun Jan 19 04:30:51 2020 - uwsgi socket 0 bound to UNIX address /run/uwsgi/app/admin-console/socket fd 3
Sun Jan 19 04:30:51 2020 - uwsgi socket 1 bound to UNIX address /tmp/admin-console.sock fd 5
Sun Jan 19 04:30:51 2020 - setgid() to 33
Sun Jan 19 04:30:51 2020 - set additional group 42 (shadow)
Sun Jan 19 04:30:51 2020 - setuid() to 33
Sun Jan 19 04:30:51 2020 - Python version: 3.7.3 (default, Apr  3 2019, 05:39:12)  [GCC 8.2.0]
Sun Jan 19 04:30:51 2020 - *** Python threads support is disabled. You can enable it with --enable-threads ***
Sun Jan 19 04:30:51 2020 - Python main interpreter initialized at 0x1123190
Sun Jan 19 04:30:51 2020 - your server socket listen backlog is limited to 100 connections
Sun Jan 19 04:30:51 2020 - your mercy for graceful operations on workers is 60 seconds
Sun Jan 19 04:30:51 2020 - mapped 193200 bytes (188 KB) for 2 cores
Sun Jan 19 04:30:51 2020 - *** Operational MODE: preforking ***
Sun Jan 19 04:30:51 2020 - WSGI app 0 (mountpoint='') ready in 0 seconds on interpreter 0x1123190 pid: 31753 (default app)
Sun Jan 19 04:30:51 2020 - *** uWSGI is running in multiple interpreter mode ***
Sun Jan 19 04:30:51 2020 - spawned uWSGI master process (pid: 31753)
Sun Jan 19 04:30:51 2020 - spawned uWSGI worker 1 (pid: 31846, cores: 1)
Sun Jan 19 04:30:51 2020 - spawned uWSGI worker 2 (pid: 31847, cores: 1)

holta, 2020-01-22:
Appears solved by PRs #2171 #2185.  Please re-open if any ongoing issues arise!

closed 2020-01-22 by holta