iiab/iiab #2175: Sudo curl d.iiab.io/install-test.txt | sudo bash  fails @ TASK [kolibri :

menneal, 2020-01-20:
Same results after two re-install attempts @ 10;00 AM PST 1/20/2020
Sudo curl d.iiab.io/install-test.txt | sudo bash

TASK [kolibri : apt install latest Kolibri .deb from http://ppa.launchpad.net/learningequality/kolibri/ubuntu/pool/main/k/kolibri-source/kolibri_0.12.9-0ubuntu2_all.deb (populates /library/kolibri, migrates database)] ***
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "Failure downloading http://ppa.launchpad.net/learningequality/kolibri/ubuntu/pool/main/k/kolibri-source/kolibri_0.12.9-0ubuntu2_all.deb, HTTP Error 404: Not Found"}

PLAY RECAP *********************************************************************
127.0.0.1                  : ok=274  changed=62   unreachable=0    failed=1    skipped=111  rescued=0    ignored=0

holta, 2020-01-20:
Hi @menneal

> sudo curl d.iiab.io/install-test.txt | sudo bash

1) Why are you running this older install-test.txt ?

   You should be running...

   `curl d.iiab.io/install.txt | sudo bash`

   NOTE: on Ubuntu 18.04 it's currently necessary to run this from the `root` account.

2) Can you run `iiab-diagnostics` and reply with the resulting pastebin URL?

   Run it as root if possible, e.g. `sudo iiab-diagnostics` ?  Thanks!

menneal, 2020-01-20:
My search for the latest version led me here: http://download.iiab.io/7.1/iiab-test
I'll  run curl d.iiab.io/install.txt | sudo bash this afternoon and report back.
Thanks for your quick reply.

holta, 2020-01-20:
For good measure I just upgraded our testing branch, so if you want to help with testing, that once again contains the latest testing branch, Thanks for the reminder!  Specifically, I just updated the IIAB testing branch underlying... 'curl http://d.iiab.io/install-test.txt | sudo bash'

So under the covers it now uses the latest testing branch from PR #2171 ...
`git clone https://github.com/jvonau/iiab -b uwsgi`

Whereas since November 28 it had been using stale testing branch...
`git clone https://github.com/jvonau/iiab -b gh-nginx-3`

<br>

ASIDE: PR #2171 should be merged today hopefully, making both branches equivalent later today if things go well!  In Any Case: most folk should stick to IIAB's regular 1-line installer here:
`curl d.iiab.io/install.txt | sudo bash`

menneal, 2020-01-20:
I'm more than willing to be a tester of any revision. I really wanted to look into the captive portal issue (if it still exists) I read about somewhere in my explorations.
Best regards,
Al

holta, 2020-01-21:
Thanks @menneal !

We should have better news later this week.

My apologies our testing branch (corresponding to PR #2171) is not yet ready to merge according to my Captive Portal testing today.

holta, 2020-01-21:
@menneal I'm closing this immediate issue now that `curl d.iiab.io/install-test.txt | sudo bash` works again.

Thanks for bringing that to our attention & keep in touch over the coming week when Captive Portal progress should be substantial (on Raspberry Pi Anyway!)

PS very early DRAFT [Release Notes for Internet-in-a-Box (IIAB) 7.1](https://github.com/iiab/iiab/wiki/IIAB-7.1-Release-Notes) in case that helps.

menneal, 2020-01-21:
Send me a link to the next version you want me to test. I'm all set to run
these tests from a 'clean' Raspbian install.
I have Zero's 3's and 4's.

holta, 2020-01-21:
Thanks You @menneal !  @jvonau added a fix overnight so I'd be interested if the testing branch's Captive Portal works for you today.  Install it as follows:

```
curl d.iiab.io/install-test.txt | sudo bash
```

1) If Captive Portal does work for you, which exact client device(s) HW/OS works did you test specifically?

2) After install's complete, do run `iiab-diagnostics` and post that to a public pastebin if that's ok, sending us the resulting URL if so!

menneal, 2020-01-21:
Installing d.iiab.io/install-test.txt | sudo bash on rpi3 64gig 3 hour @
7:51 PST.

menneal, 2020-01-21:
1/21/2020

sudo curl d.iiab.io/install-test.txt | sudo bash

TASK [js-menu : Post process the downloaded menu defs]
*************************

fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd":
"/opt/admin/cmdsrv/scripts/install_menu_defs.py", "delta":
"0:00:00.846602", "end": "2020-01-21 09:11:20.889295", "msg": "non-zero
return code", "rc": 1, "start": "2020-01-21 09:11:20.042693", "stderr": "",
"stderr_lines": [], "stdout": "Unable to contact Server", "stdout_lines":
["Unable to contact Server"]}

PLAY RECAP
*********************************************************************

127.0.0.1 : ok=64 changed=56 unreachable=0 failed=1 skipped=11 rescued=0
ignored=0

pi@raspberrypi:~ $

PUBLISHING TO URL... /usr/bin/pastebinit:42: DeprecationWarning: dist() and
linux_distribution() functions are deprecated in Python 3.5

release = platform.linux_distribution()[0].lower()

/usr/bin/pastebinit:413: DeprecationWarning: pasteURLopener style of
invoking requests is deprecated. Use newer urlopen functions/methods

url_opener = pasteURLopener()

http://paste.debian.net/1127011/

jvonau, 2020-01-21:
There was an oversight when editing install-test.txt, the admin-console branch was also pointing to a stale unused branch. As a workaround could you `sudo rm -rf /opt/iiab/iiab/-admin-console` then `sudo iiab` to continue with the install. The install will skip over parts that were already completed, sorry for the inconvenience caused.

menneal, 2020-01-21:
Will do

menneal, 2020-01-21:
1/21/2020 10:45 PST

sudo rm -rf /opt/iiab/iiab/-admin-console

sudo iiab

TASK [js-menu : Otherwise sync local menu defs with repo]
**********************

fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd":
"/opt/admin/cmdsrv/scripts/sync_menu_defs.py", "delta": "0:00:00.872878",
"end": "2020-01-21 10:40:45.225258", "msg": "non-zero return code", "rc":
1, "start": "2020-01-21 10:40:44.352380", "stderr": "", "stderr_lines": [],
"stdout": "Unable to contact Server", "stdout_lines": ["Unable to contact
Server"]}

PLAY RECAP
*********************************************************************

127.0.0.1 : ok=61 changed=10 unreachable=0 failed=1 skipped=15 rescued=0
ignored=0

COMPLETE! Your diagnostics file (76K, 1560 lines) is:

/etc/iiab/diag/7.1_raspbian-10_2020-01-21_10:46:42_PST_menneal

Publish it to a web pastebin? [Y/n] y

PUBLISHING TO URL... /usr/bin/pastebinit:42: DeprecationWarning: dist() and
linux_distribution() functions are deprecated in Python 3.5

release = platform.linux_distribution()[0].lower()

/usr/bin/pastebinit:413: DeprecationWarning: pasteURLopener style of
invoking requests is deprecated. Use newer urlopen functions/methods

url_opener = pasteURLopener()

jvonau, 2020-01-21:
`sudo rm -rf /opt/iiab/iiab/-admin-console` is wrong should be `sudo rm -rf /opt/iiab/iiab-admin-console
` note the removal of the /

menneal, 2020-01-21:
Running again. Results ~ 3:00PM PST

menneal, 2020-01-21:
Install went well and iiab up and running... no errors.
Access to iiab-admin with install created PW works.
Captive portal..not so much on linux mint firefox or win10 chrome.
Is it OK to re-claim use of the RPI chrome browser or will this break
something?
Best regards,
Al

holta, 2020-01-21:
> Install went well and iiab up and running... no errors. Access to iiab-admin with install created PW works. Captive portal..not so much on linux mint firefox or win10 chrome.

Thanks for the report!  We'll work it.  Likely PR #2171 needs a bit more tuning before merging &mdash; as `cd /opt/iiab/iiab ; ./runrole captiveportal` should work in general (to rapidly install or uninstall Captive Portal, based on its 2 vars `captiveportal_install` and `captiveportal_enabled` in /etc/iiab/local_vars.yml).

But this is not quite yet working on my end.  (FWIW my current test machine is a bit memory-constrained, it being a 1GB RAM RPi 4.)

> Is it OK to re-claim use of the RPI chrome browser or will this break something?

What do you mean be re-claim?  Using  the Chromium browser (apt pkg `chromium-browser`) that's built into "Raspbian Buster with desktop" should be absolutely fine...are you seeing any issues there?

menneal, 2020-01-21:
I'll capture the warning message for you next install. I decided to goforit since I was going to do more installs and so far all is still fine.

closed 2020-01-24 by holta