iiab/iiab #2184: IIAB serves files to XO laptop

llovemixo, 2020-01-22:
IIAB 7.0 on Pi Zero W with Buster with full desktop.

Below is a screen shot of a file served by IIAB to Browse on an XO-1 laptop in Browse.

[image]

A menu item motoli was created as in previous post.
The file /library/www/html/motoli/index.html was then replaced with a short section of the startup section of RACHEL static 2016. (about 15 lines).
The Browse was wirelessly connected to IIAB and file opened in Browse with http://box.local/motoli

This a proof of concept that IIAB can deliver files to an XO laptop. Most of the OER2Go (Rachel) modules should be deliverable, but delivering zim files would be more complex .  Advance graphic modules such Feed the Monster are not expected to work.

Such "menu items" could be collected in a single menu or mix with normal "menu items"
This example delivers only a menu of one item.  For several items, one would have to add code to extract menu items from /library/www/html/motoli/menu.jason.
An alternative would be to modifiy the original index.html to provide this alternate css and style.

This Motoli example (semi-commercial product) is not from Rachel, but only required simple changes to fit the Rachel  pattern.

tim-moody, 2020-01-22:
> but delivering zim files would be more complex

Can you elaborate?

In terms of motili setup, the narrative talks about /library/www/html/motoli/index.html, but the url shows box/modules/motili, which is the standard place for 'html', e.g. oer2go style content.

llovemixo, 2020-01-22:
Zim is "complex" since it requires "unzipping" zim file, which involves more than 15 lines of code.

The motoli app is in modules directory as en-motoli

My replacement index.html contains the css and style folders from Rachel and the index file (yours from IIAB on XO) that starts Rachel, which jumps to modules/en-motoli/index.html.

All modules in module/ seem to contain both htmlf and html index files.

Creating more "menus" would require manually changing the destination index.html in the equivalent  of the /motoli/index.html file.  This is, after all, a crude proof of concept.

The IIAB menu item is a clone of the credits menu item with the menu code of en-motoli as a place holder .  A place holder, since it is not used since the jump is done by the hard value in my index.html

The best of all worlds would be a clone of the standard  index.html  as second standard index.html with css and styles that are compatible with Browse on XO .  Then, zim files would be available in addition to Rachel modules (?).

tim-moody, 2020-01-23:
for en-motoli you would be better cloning en-oerafrica-reading which is an 'html' type menu def, which means it assume the content is in /modules, unlike en-credits, which is a 'webroot' type.

the htmlf files, if present, come from Rachel's menuing system and are not used.

motoli is not in oer2go or it would probably already have a menu def

zims in common languages have menu defs. otherwise if downloaded with the admin console a menu def is generated and added to the main menu.

llovemixo, 2020-01-23:
The information is very helpful.
I will be exploring two paths to give XO laptops here partial access to IIAB content.

For immediate needs, the entire menu of RACHEL 2016 Static will be used as for Motoli and RACHEL menu items will be pointed to the modules/ directory.  For this to be more complete, Wikipedia for Schools and Khan Academy from 2016 static will have to be copied to modules/ directory.

Long term, from examination of the "home/" menu's index.html, it appears that CSS 's and STYLE's  of the "home/" menu can be modified for a "xohome" menu (http://box.local/xohome).  Not capable of doing this without learning more about modifying these.

llovemixo, 2020-01-27:
Using RACHEL path works.  Create /xo menu; clone "credits" with URL start "/XO"; place RACHEL menu in /XO; Replace original IIAB index.html in /XO with simple index.html to jump to RACHEL menu.

In the RACHEL menu paths to "modules", replace with "/../../../modules".

Still, vision is to create menus in IIAB .  View these menus and  select item with XO compatible
CSS / Style.

closed 2020-01-27 by llovemixo