iiab/iiab #2186: osm-vector-maps breaks when no internet available and alreay installed.

jvonau, 2020-01-23:
### Expected behavior and actual behavior.
not to contact internet when already installed and operating offline.
### Steps to reproduce the problem.
disconnect from internet and run iiab-configure or ICO post install.

TASK [OSM-VECTOR-MAPS] ********************************************************************************

TASK [osm-vector-maps : Ensure directory /library/www/osm-vector-maps/maplist/assets exists] **********
ok: [127.0.0.1]

TASK [osm-vector-maps : Download map catalog http://download.iiab.io/content/OSM/vector-tiles/maplist/hidden/assets/regions.json to /library/www/osm-vector-maps/maplist/assets/] ***
fatal: [127.0.0.1]: FAILED! => {"changed": false, "dest": "/library/www/osm-vector-maps/maplist/assets/", "elapsed": 0, "gid": 33, "group": "www-data", "mode": "0755", "msg": "Request failed: <urlopen error [Errno 101] Network is unreachable>", "owner": "www-data", "size": 4096, "state": "directory", "uid": 33, "url": "http://download.iiab.io/content/OSM/vector-tiles/maplist/hidden/assets/regions.json"}

PLAY RECAP ********************************************************************************************
127.0.0.1                  : ok=182  changed=59   unreachable=0    failed=1    skipped=109  rescued=0    ignored=1   

Needs the install/enabled-or-disabled split applied to the role to ensure off-line operation.

tim-moody, 2020-01-23:
There are a number services that try to refresh external data during configuration. I suppose should be conditional on {{ internet_available }}, but may have unintended consequences.

holta, 2020-01-23:
@avni & I might re-test this tmrw...and implement @jvonau's recommended fix by splitting main.yml into {install.yml online, enable-or-disable.yml offline}

closed 2020-01-28 by holta