iiab/iiab #2197: OSM Error: "refusing to convert from file to symlink for /library/www/html/common/assets/regions.json"

holta, 2020-01-24:
@georgejhunt OSM install routines have proven fragile below.  Let's discuss with @tim-moody how to make this more resilient?

Thanks to @jvonau who figured out (below) what seems to be happen when './iiab-install --reinstall' (and './runrole osm-vector-maps') both fail every time on BIG-sized master when layered on top of 182-rpi4-2g-MIN-0122-PR2171 = 10.8.0.66 that had no osm-vector-maps installed originally:

> TASK [osm-vector-maps : Symlink catalog /library/www/html/common/assets/regions.json -> /library/www/osm-vector-maps/maplist/assets/regions.json] ***
fatal: [127.0.0.1]: FAILED! => {"changed": false, "gid": 0, "group": "root", "mode": "0644", "msg": "refusing to convert from file to symlink for /library/www/html/common/assets/regions.json", "owner": "root", "path": "/library/www/html/common/assets/regions.json", "size": 21, "state": "file", "uid": 0}

Chat Log:

AH: OSM Error Context: I re-cloned from master and put a brand new BIG-sized /etc/iiab/local_vars.yml in place, before I ran "./iiab-install --reinstall"

JV: did you install a real map prior to --reisntall ?

AH: No.  (I did not install any maps or content.)

JV: https://github.com/iiab/iiab-admin-console/blob/master/roles/js-menu/tasks/main.yml#L132  starts some osm routines about regions.json

replacing the symlink is noted.

AH: Should I re-run './iiab-install' or './runrole osm-vector-maps' ?

JV: figure out which playbook is running the show first..
iiab or iiab-admin-console

AH: Not sure what that means in terms of next steps...

JV: you need to do the install/enabled split first
keep the symlink creation in install.yml as it can changed later as you see now..

AH: You mean https://github.com/iiab/iiab/blob/master/roles/osm-vector-maps/tasks/main.yml ?

JV: yup.. just another service

AH: Why didn't this happen before, do you know?

JV: and should be treated as such..
I can guess at incomplete testing, like offline behavior I found. (#2186)

doing a admin-console install over prior install "Clone js-menu-files" doesn't run but "Otherwise sync local menu defs" does it's sync thing... taking a while.

TASK [js-menu : Remove region.json dummy file if osm_vector_maps_enabled] *
changed: [127.0.0.1]

TASK [js-menu : Create a link to osm catalog in /common/assets] *
changed: [127.0.0.1]

TASK [js-menu : Add any MAPS that are installed to home/menu.js] **
changed: [127.0.0.1]

./runrole osm-vector-maps runs correctly... 

AH: Runs correctly where?  I'm not following.

JV: I can't reproduce your issue... 

AH: What machine are you on?

JV: mine...

AH: Ok.  On RPi 4 10.8.0.66 './runrole osm-vectors-maps' fails each time.

JV: doing a admin-console install over prior install "Clone js-menu-files" doesn't run but "Otherwise sync local menu defs" does it's sync thing... taking a while.

AH: I did not re-run Admin Console's ' ./install', if that's what you're asking?

JV: maybe that is the fix?
that was an observation, should see a new commit in the menu-def repo..

AH: Weird if things changed from 2 days ago, but I can try Admin Console's ./install first if that might help?

AH: Seems quite strange to install Admin Console before running './iiab-install --reinstall' but sure I can try?

JV: same file 2 masters... 

AH: iiab-admin-console is already up-to-date when I try 'git pull' there...

JV: fix the logic between the 2..

AH: So should I really try to ./install it again?  Am not seeing how this is connected...but if it's important I'll do it?

JV: try it and note the status of the last 3 stanzas

AH: ./install running on 10.8.0.66

Running ./install fails as follows...

> TASK [js-menu : Add any MAPS that are installed to home/menu.js] ***************
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "/usr/bin/iiab-update-map", "delta": "0:00:00.009312", "end": "2020-01-24 14:00:40.428324", "msg": "non-zero return code", "rc": 127, "start": "2020-01-24 14:00:40.419012", "stderr": "/bin/sh: 1: /usr/bin/iiab-update-map: not found", "stderr_lines": ["/bin/sh: 1: /usr/bin/iiab-update-map: not found"], "stdout": "", "stdout_lines": []}

Recap: there has been no change in code... /opt/iiab/iiab-admin-console has the latest master, the same as it did when I installed it 48h ago.

That would explain it.
As this was the local_vars.yml we trimmed down [during RPi 4's orig installation 48 ago] to include only 2 things (Captive Portal and Kiwix).

JV: what are the other 2?
that would be why... 

AH: So the following evidently leads to any problems...

osm_vector_maps_install: False
osm_vector_maps_enabled: False

So a sentence is needed in local_vars_{min,medium,big}.yml to remind folks not to touch those 2 lines for now [?]

JV: why is maps running if you set the install to False?

AH: I converted from sub-MIN (48h ago, "2 apps only" i.e. Captive Portal & Kiwix) to BIG-sized (12h ago) which caused the failure for some reason...

JV: the other 2 stanza's have what?

AH: What 2 stanzas?

JV: above the iiab-admin failure
both are conditional on *_enabled

AH: I'm lost.  What iiab-admin failure / where?

JV: above TASK [js-menu : Add any MAPS that are installed to home/menu.js] *

> TASK [js-menu : Copy Osm Vector Maps dummy files] ******************************
skipping: [127.0.0.1] => (item=regions.json)
skipping: [127.0.0.1] => (item=vector-map-idx.json)
>
> TASK [js-menu : Remove region.json dummy file if osm_vector_maps_enabled] ******
changed: [127.0.0.1]
>
> TASK [js-menu : Create a link to osm catalog in /common/assets] ****************
changed: [127.0.0.1]
>
> TASK [js-menu : Add any MAPS that are installed to home/menu.js] ***************
fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": "/usr/bin/iiab-update-map", "delta": "0:00:00.009312", "end": "2020-01-24 14:00:40.428324", "msg": "non-zero return code", "rc": 127, "start": "2020-01-24 14:00:40.419012", "stderr": "/bin/sh: 1: /usr/bin/iiab-update-map: not found", "stderr_lines": ["/bin/sh: 1: /usr/bin/iiab-update-map: not found"], "stdout": "", "stdout_lines": []}

JV: the 2 stanzas are both are conditional on osm*_enabled, hence 2 masters of behavior... 
now there is a symlink runrole should not fail

AH: What's your recommendation for George, in summary?

JV: the first install would of hit ASK [js-menu : Copy Osm Vector Maps dummy files] placing real files in place of symlinks due to osm_*enabled: False

setting up the failure on the second pass you saw..

so runrole osm-vector-maps should now not fail at the symlink test and ./install should complete

the osm*_enabled changes should occur in one place supply the dummy files in iiab maybe?

georgejhunt, 2020-01-25:
this is due to the admin console generating a dummy file: see     -
https://github.com/iiab/iiab-admin-console/blob/74ed5c476ec6380f8052493f41bc7f8ec1cfe4aa/roles/js-menu/tasks/main.yml#L146
-- when osm-vector-maps are not enabled.

I think the easiest would be to make sure the dummy regions.json is absent
before creating the link to the one that is up to date.

I think this is the sort of bug that shows up with testing.

tim-moody, 2020-01-25:
One question that needs to be answered is whether OSM is like kiwix and should always be installed. These dummy files are baked into the menuing and the startup of adm cons.

tim-moody, 2020-01-25:
If OSM stored regions.json in /etc/iiab, then the dummy could go there and a link just be created.

georgejhunt, 2020-01-25:
I'm in favor of installing osm-vector-maps role always, similar to
kiwix. And then writing directly to /etc/iiab/regions.json. That's
what the next version of MAPS is already is doing.

tim-moody, 2020-01-27:
what do #2199 and #2201 have to do with this? #2186 is likely a duplicate.

holta, 2020-01-27:
"/usr/bin/iiab-update-map: not found" is the error message (pasted into original bug report / chat transcript above) that resulted when @jvonau asked me to re-run `cd /opt/iiab/iiab-admin-console` then `./install`

I'd assume from @jvonau's PR #2199 that #2186 is now solved, and that this #2197 is a bit different, but perhaps there's more to the story he can clarify.

holta, 2020-01-28:
Context for others...
https://github.com/iiab/iiab-admin-console/blob/master/roles/js-menu/tasks/main.yml#L132-L165

(A few days ago @jvonau earlier mentioned that's relevant...to working out OSM bugs like #2186, #2197 etc!)

holta, 2020-01-30:
@tim-moody looked into this and made progress here that he explained on today's IIAB call.  Kind of a transition problem, that he can mitigate using `/etc/iiab` in the short term, to avoid this risk.  Also problem will go away in future with @georgejhunt's new OSM Vector Maps framework expected to be a part of [IIAB 7.2](https://github.com/iiab/iiab/milestone/7) around mid-2020.

holta, 2020-04-08:
@georgejhunt @tim-moody can we close this now?

(Or is this a 'Known Issue' for IIAB 7.1's release expected around 2020-04-30?)

tim-moody, 2020-04-08:
think so

holta, 2020-04-08:
Thanks @tim-moody 

(Anybody feel free to re-open this is this or similar flares up!)

tim-moody, 2020-04-10:
Just got this error again. Did an rpi 4 install with 
osm_vector_maps_install: False
osm_vector_maps_enabled: False
commented both lines out and did runrole (assuming that the values in vars/default_vars.yml would prevail)

jvonau, 2020-04-10:
> Just got this error again. Did an rpi 4 install with
> osm_vector_maps_install: False
> osm_vector_maps_enabled: False

cat vars/default_vars.yml | grep osm
>osm_vector_maps_install: True
osm_vector_maps_enabled: True

> commented both lines out and did runrole (assuming that the values in vars/default_vars.yml would prevail)

That ran the role, osm is now turned on, I don't think that was the intention if you had set the install value to False. The role should not of ran at all with entries missing from local_vars, that should of been caught like when installed exists but install is False.

> failed: [127.0.0.1] (item=osm_vector_maps) => {"ansible_loop_var": "item", "assertion": "osm_vector_maps_install or osm_vector_maps_installed is undefined", "changed": false, "evaluated_to": false, "item": "osm_vector_maps", "msg": "DISALLOWED: 'osm_vector_maps_install: False' (e.g. in /etc/iiab/local_vars.yml) WHEN 'osm_vector_maps_installed' is defined (e.g. in /etc/iiab/iiab_state.yml) -- IIAB DOES NOT SUPPORT UNINSTALLS -- please verify those 2 files especially, and other places variables are defined?"}

@holta there is an error in the validation, commented out/missing vars in local_vars.yml should error out, especially when iiab_state has *_installed set.

tim-moody, 2020-04-10:
Default vars has 

> osm_vector_maps_install: True
> osm_vector_maps_enabled: True

When IIAB was first installed local_vars had

> osm_vector_maps_install: False
> osm_vector_maps_enabled: False

Subsequently the lines in local vars were commented out and runrole executed. This is not an uninstall. It is the opposite, a new install. That was my intention. Local vars is an override and should only have to contain values different from default vars. Otherwise there is no reason for default vars.

tim-moody, 2020-04-10:
And even if local vars had redundantly had the same values as default vars, this error would still have occurred because the existing code tries to put a symlink where there is already a file and fails.

holta, 2020-04-10:
The goal of variable validation is not to sanity-check any specific file.

The goal of is to sanity-check the values of variables &mdash; regardless what files (or runtime processes) defined those variables.

Reminder- IIAB's initial Ansible variable values can arise from about 7 different files listed in: https://github.com/iiab/iiab/wiki/IIAB-Variables#order-of-execution-and-precedence

jvonau, 2020-04-10:
> Default vars has
> 
> > osm_vector_maps_install: True
> > osm_vector_maps_enabled: True
> 
> When IIAB was first installed local_vars had
> 
> > osm_vector_maps_install: False
> > osm_vector_maps_enabled: False
> 
Then install.yml in the role should not of run and if it didn't run there would be no symlink created.

> Subsequently the lines in local vars were commented out and runrole executed. This is not an uninstall. It is the opposite, a new install.

Why not just leave local_vars as setup with _install set to True then? In the end what is your goal? To have OSM installed or skipped? 

> That was my intention. Local vars is an override and should only have to contain values different from default vars. Otherwise there is no reason for default vars.

Other than carry vars that are not defined in local_vars, I would agree with.

jvonau, 2020-04-10:
> And even if local vars had redundantly had the same values as default vars, this error would still have occurred because the existing code tries to put a symlink where there is already a file and fails.

Could you be more explicit around where this error is, where is this occurring in iiab osm's role or in js-menu?

> The goal of [variable validation](https://github.com/iiab/iiab/blob/master/roles/0-init/tasks/validate_vars.yml) is not to sanity-check any specific file.
> 

DISALLOWED: 'osm_vector_maps_install: False' (e.g. in /etc/iiab/local_vars.yml) WHEN 'osm_vector_maps_installed' is defined (e.g. in /etc/iiab/iiab_state.yml) -- IIAB DOES NOT SUPPORT UNINSTALLS

> The goal is to sanity-check the values of variables — regardless what files (or runtime processes) defined those variables.
> 
> Reminder- IIAB's initial Ansible variable values can arise from about 7 different files listed in: https://github.com/iiab/iiab/wiki/IIAB-Variables#order-of-execution-and-precedence

Sure then the default state should be set to *_install: False in default_vars otherwise absent vars in local_vars are not taken into consideration against iiab_state.
The action with default_vars set to _install False and _install commented out in local_vars 
> failed: [127.0.0.1] (item=osm_vector_maps) => {"ansible_loop_var": "item", "assertion": "osm_vector_maps_install or osm_vector_maps_installed is undefined", "changed": false, "evaluated_to": false, [...]

holta, 2020-04-10:
> DISALLOWED: 'osm_vector_maps_install: False' (e.g. in /etc/iiab/local_vars.yml) WHEN 'osm_vector_maps_installed' is defined (e.g. in /etc/iiab/iiab_state.yml) -- IIAB DOES NOT SUPPORT UNINSTALLS

e.g. (exempli gratia) = “for example"

Specifically, any other the ~7 files listed in https://github.com/iiab/iiab/wiki/IIAB-Variables#order-of-execution-and-precedence (or any other possibly runtime process) can be used to set variables. 
 Which is why the error message explains "please verify those 2 files especially, and other places variables are defined".

In other words, [variable validation](https://github.com/iiab/iiab/blob/master/roles/0-init/tasks/validate_vars.yml) does not care what file or runtime process set variables.  Rather, its objective is to verify that variables are set to sane/plausible values.

jvonau, 2020-04-11:
The one way I could recreate this issue is with *_enabled False and runrole --reinstall **after** the admin-console is installed and the dummy files are created there using force: yes.
>2020-04-11 00:00:51,526 p=17345 u=root n=ansible | TASK [js-menu : Copy Osm Vector Maps dummy files] ******************************
2020-04-11 00:00:52,704 p=17345 u=root n=ansible | changed: [127.0.0.1] => (item=regions.json)
2020-04-11 00:00:53,793 p=17345 u=root n=ansible | changed: [127.0.0.1] => (item=vector-map-idx.json)

One way would just add force: yes to the symlink creation in TASK osm-vector-maps or as the PR suggests delete the symlink, but now the question becomes which set of dummy files should be used? Do we really want to overwrite the files provided by js-menu in what is primarily a bootstrap stage? Think I would like to err to the later supplied files.

jvonau, 2020-04-11:
Also reverts from js-menu, is that what is wanted?
>- name: Create a link to osm catalog in /common/assets
  file:
    src: "{{ vector_map_path }}/maplist/assets/regions.json"
    dest: "{{ doc_root }}/common/assets/regions.json"
    state: link
  when: osm_vector_maps_enabled

holta, 2020-04-25:
Now that OSM's #2350 and https://github.com/iiab/iiab-admin-console/pull/303 are both merged, please if you can help test OSM Map Packs in preparation for [IIAB 7.1](https://github.com/iiab/iiab/wiki/IIAB-7.1-Release-Notes) coming soon!

Then I will work with George & All in coming days to bring our public-facing IIAB Maps doc up to date:
https://github.com/iiab/iiab/wiki/IIAB-Maps

holta, 2020-09-03:
@georgejhunt please re-open if this-or-similar is still an ongoing issue, Thanks!

PR #2334 (2020-04-11): remove any possible file before creating regions symlink
### fixes #2197
closed 2020-09-03 by holta