iiab/iiab #2204: Internet Archive http://box:4244 isn't working on fresh install of IIAB on Ubuntu 18.04

holta, 2020-01-27:
@mitra42 would you possibly know the reason?  And/or what I should investigate to find the cause?

Internet Archive's port (4244) does not appear in `netstat -ltup | grep 4244` whereas most other [IIAB ports](https://github.com/iiab/iiab/wiki/IIAB-Networking#list-of-ports--services) do appear there.

FYI this machine is running the latest IIAB 7.1 pre-release (installed using http://download.iiab.io) on Ubuntu Desktop 18.04 in preparation for Afghan, Mideast & African refugees.

Aside: machine nickname is 191-u1804-desk-BIG+calib-0127 = 10.8.0.14

jvonau, 2020-01-27:
> Jan 27 11:04:28 box.lan internetarchive[27069]: /opt/iiab/internetarchive/node_modules/@internetarchive/dweb-transports/Errors.js:38
Jan 27 11:04:28 box.lan internetarchive[27069]: }
Jan 27 11:04:28 box.lan internetarchive[27069]: ^
Jan 27 11:04:28 box.lan internetarchive[27069]: SyntaxError: Unexpected token '}'
Jan 27 11:04:28 box.lan internetarchive[27069]:     at Module._compile (internal/modules/cjs/loader.js:891:18)
Jan 27 11:04:28 box.lan internetarchive[27069]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:991:10)
Jan 27 11:04:28 box.lan internetarchive[27069]:     at Module.load (internal/modules/cjs/loader.js:811:32)
Jan 27 11:04:28 box.lan internetarchive[27069]:     at Function.Module._load (internal/modules/cjs/loader.js:723:14)
Jan 27 11:04:28 box.lan internetarchive[27069]:     at Module.require (internal/modules/cjs/loader.js:848:19)
Jan 27 11:04:28 box.lan internetarchive[27069]:     at require (internal/modules/cjs/helpers.js:74:18)
Jan 27 11:04:28 box.lan internetarchive[27069]:     at Object.<anonymous> (/opt/iiab/internetarchive/node_modules/@internetarchive/dweb-transports/Transports.js:3:28)
Jan 27 11:04:28 box.lan internetarchive[27069]:     at Module._compile (internal/modules/cjs/loader.js:955:30)
Jan 27 11:04:28 box.lan internetarchive[27069]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:991:10)
Jan 27 11:04:28 box.lan internetarchive[27069]:     at Module.load (internal/modules/cjs/loader.js:811:32)

jvonau, 2020-01-27:
thinking the filesystem has been filled up is the root cause.
> iiab-admin@box:/opt/iiab/iiab$ df
Filesystem     1K-blocks     Used Available Use% Mounted on
udev             1991160        0   1991160   0% /dev
tmpfs             403096    41668    361428  11% /run
/dev/sda1       20509264 20492880         0 100% /
tmpfs            2015464        0   2015464   0% /dev/shm
tmpfs               5120        4      5116   1% /run/lock
tmpfs            2015464        0   2015464   0% /sys/fs/cgroup
/dev/loop0         15104    15104         0 100% /snap/gnome-characters/399
/dev/loop2          3840     3840         0 100% /snap/gnome-system-monitor/123
/dev/loop1         46080    46080         0 100% /snap/gtk-common-themes/1440
/dev/loop3        164096   164096         0 100% /snap/gnome-3-28-1804/116
/dev/loop4          4352     4352         0 100% /snap/gnome-calculator/544
/dev/loop5        160512   160512         0 100% /snap/gnome-3-28-1804/110
/dev/loop6         45312    45312         0 100% /snap/gtk-common-themes/1353
/dev/loop7         56064    56064         0 100% /snap/core18/1650
/dev/loop8          4224     4224         0 100% /snap/gnome-calculator/406
/dev/loop9         55936    55936         0 100% /snap/core18/1288
/dev/loop10        91264    91264         0 100% /snap/core/8268
/dev/loop11         3840     3840         0 100% /snap/gnome-system-monitor/127
/dev/loop12         1024     1024         0 100% /snap/gnome-logs/81
/dev/loop13        15104    15104         0 100% /snap/gnome-characters/375
/dev/loop14        90624    90624         0 100% /snap/core/7270
/dev/loop15         1024     1024         0 100% /snap/gnome-logs/61
tmpfs             403092        0    403092   0% /run/user/1000
tmpfs             403092       16    403076   1% /run/user/121

holta, 2020-01-27:
Refs:

#2157 "Cham: How do we make it alert user in case we run out of storage"
#2205 "OSM Vector Maps (like other content?) can fill smaller disks, without much warning"

holta, 2020-01-27:
1) Sorry @mitra42 & @jvonau about the full disk which I've now taken care of, and was NOT related to this failure of http://box:4244 !

2) Source of  the confusion (my apologies) is that I used this same VM to test something else (large OpenStreetMap Map Packs, on a small 20GB disk) and it turns out every time I deleted the files being unzipped into /library/working/maps they instantly began recreating themselves on reboot...until I cancelled that unzip job[*] Very Rapidly after the ~3rd reboot...using http://box/admin -> Utilities -> Display Job Status.

[*] which unfortunately kept restarting, freezing the disk & OS within about a minute after each boot.  Suggests that Admin Console should avoid auto-restarting jobs when the disk is nearly full, to avoid exacerbating OS freezing/lockout problems like this.

mitra42, 2020-01-27:
No problem :-) glad you solved it.

holta, 2020-01-27:
> No problem :-) glad you solved it.

Not Solvedm Sadly (-:

Do you know of any issues running http://box:4244 or http://box.lan:4244 or http://box.local:4244 or http://localhost:4244 on Ubuntu 18.04?  None of which work so far, yikes!

mitra42, 2020-01-27:
I don't know why there would be problems.  I don't have Ubuntu 18.04 but in general the code isn't fragile about its OS - have it working currently on a Nuc (not sure which OS); MacOSX; Raspbian stand-alone; Raspbian+IIAB; Docker (node:12-alpine on 3 architectures; node:12); 

`journalctl -u internetarchive` I think should get the log with any failure. 

It would also be worth doing a `./runrole --reinstall internetarchive` as I have on occasions pushed non-working versions, I just pushed a new one 0.2.73 to make sure all dependencies are current

holta, 2020-01-27:
```
root@box:~# netstat -ltup | grep 4244

root@box:~# journalctl -u internetarchive | wc
  89475  787338 9369575

root@box:~# journalctl -u internetarchive | head -20
-- Logs begin at Tue 2019-12-10 23:18:55 EST, end at Mon 2020-01-27 18:37:12 EST. --
Jan 27 04:36:15 box.lan systemd[1]: Started Internet Archive Universal Library service.
Jan 27 04:36:15 box.lan internetarchive[27752]: /opt/iiab/internetarchive/node_modules/@internetarchive/dweb-transports/Errors.js:38
Jan 27 04:36:15 box.lan internetarchive[27752]: }
Jan 27 04:36:15 box.lan internetarchive[27752]: ^
Jan 27 04:36:15 box.lan internetarchive[27752]: SyntaxError: Unexpected token '}'
Jan 27 04:36:15 box.lan internetarchive[27752]:     at Module._compile (internal/modules/cjs/loader.js:891:18)
Jan 27 04:36:15 box.lan internetarchive[27752]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:991:10)
Jan 27 04:36:15 box.lan internetarchive[27752]:     at Module.load (internal/modules/cjs/loader.js:811:32)
Jan 27 04:36:15 box.lan internetarchive[27752]:     at Function.Module._load (internal/modules/cjs/loader.js:723:14)
Jan 27 04:36:15 box.lan internetarchive[27752]:     at Module.require (internal/modules/cjs/loader.js:848:19)
Jan 27 04:36:15 box.lan internetarchive[27752]:     at require (internal/modules/cjs/helpers.js:74:18)
Jan 27 04:36:15 box.lan internetarchive[27752]:     at Object.<anonymous> (/opt/iiab/internetarchive/node_modules/@internetarchive/dweb-transports/Transports.js:3:28)
Jan 27 04:36:15 box.lan internetarchive[27752]:     at Module._compile (internal/modules/cjs/loader.js:955:30)
Jan 27 04:36:15 box.lan internetarchive[27752]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:991:10)
Jan 27 04:36:15 box.lan internetarchive[27752]:     at Module.load (internal/modules/cjs/loader.js:811:32)
Jan 27 04:36:15 box.lan systemd[1]: internetarchive.service: Main process exited, code=exited, status=1/FAILURE
Jan 27 04:36:15 box.lan systemd[1]: internetarchive.service: Failed with result 'exit-code'.
Jan 27 04:36:26 box.lan systemd[1]: internetarchive.service: Service hold-off time over, scheduling restart.
Jan 27 04:36:26 box.lan systemd[1]: internetarchive.service: Scheduled restart job, restart counter is at 1.

root@box:~# journalctl -u internetarchive | tail -20
Jan 27 18:37:21 box.lan systemd[1]: internetarchive.service: Service hold-off time over, scheduling restart.
Jan 27 18:37:21 box.lan systemd[1]: internetarchive.service: Scheduled restart job, restart counter is at 2422.
Jan 27 18:37:21 box.lan systemd[1]: Stopped Internet Archive Universal Library service.
Jan 27 18:37:21 box.lan systemd[1]: Started Internet Archive Universal Library service.
Jan 27 18:37:21 box.lan internetarchive[19341]: /opt/iiab/internetarchive/node_modules/@internetarchive/dweb-transports/Errors.js:38
Jan 27 18:37:21 box.lan internetarchive[19341]: }
Jan 27 18:37:21 box.lan internetarchive[19341]: ^
Jan 27 18:37:21 box.lan internetarchive[19341]: SyntaxError: Unexpected token '}'
Jan 27 18:37:21 box.lan internetarchive[19341]:     at Module._compile (internal/modules/cjs/loader.js:891:18)
Jan 27 18:37:21 box.lan internetarchive[19341]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:991:10)
Jan 27 18:37:21 box.lan internetarchive[19341]:     at Module.load (internal/modules/cjs/loader.js:811:32)
Jan 27 18:37:21 box.lan internetarchive[19341]:     at Function.Module._load (internal/modules/cjs/loader.js:723:14)
Jan 27 18:37:21 box.lan internetarchive[19341]:     at Module.require (internal/modules/cjs/loader.js:848:19)
Jan 27 18:37:21 box.lan internetarchive[19341]:     at require (internal/modules/cjs/helpers.js:74:18)
Jan 27 18:37:21 box.lan internetarchive[19341]:     at Object.<anonymous> (/opt/iiab/internetarchive/node_modules/@internetarchive/dweb-transports/Transports.js:3:28)
Jan 27 18:37:21 box.lan internetarchive[19341]:     at Module._compile (internal/modules/cjs/loader.js:955:30)
Jan 27 18:37:21 box.lan internetarchive[19341]:     at Object.Module._extensions..js (internal/modules/cjs/loader.js:991:10)
Jan 27 18:37:21 box.lan internetarchive[19341]:     at Module.load (internal/modules/cjs/loader.js:811:32)
Jan 27 18:37:21 box.lan systemd[1]: internetarchive.service: Main process exited, code=exited, status=1/FAILURE
Jan 27 18:37:21 box.lan systemd[1]: internetarchive.service: Failed with result 'exit-code'.
```

mitra42, 2020-01-27:
Sorry - bad timing, I pushed something last night and didn't catch that syntax error till this orning.

holta, 2020-01-27:
> It would also be worth doing a `./runrole --reinstall internetarchive` as I have on occasions pushed non-working versions, I just pushed a new one 0.2.73 to make sure all dependencies are current

That fixed it, Thanks!

closed 2020-01-27 by holta