iiab/iiab #2205: OSM Vector Maps (like other content?) can fill smaller disks, without much warning

holta, 2020-01-27:
It used to be that "Central America" (https://archive.org/download/en-osm-omt_central_america_2017-07-03_v0.23/) was installable on a 20GB disk running BIG-sized IIAB.

Close call, but no longer &mdash; even though http://box/admin -> Install Content -> Get Map Region is a bit misleading &mdash; when it suggest this install would fit...this is what is shows right before you click "Install Selected Region" :

> Library Space Available : 4.98 GB
> Estimated Space Required:      3.77 GB

Today the 20GB disk fills up (unlike in the past, likely as a result of other IIAB Apps/Services growing in size) and in the end http://box/admin -> Utilities -> "Display Job Status" does not make that obvious:

[image]

Perhaps this is too hard to fix?  If so we can declare it "wontfix" for now.

In any case, this relates to @m-anish's question at #2157 &mdash; on what IIAB and apps/services like Cham should do in general &mdash; when disk appears close to full?

holta, 2020-01-27:
Disk details should not matter, but just FYI disk is confirmed filled up by the following:

> root@box:~# ls -l /library/downloads/maps/en-osm-omt_central_america_2017-07-03_v0.23.zip
-rw-r--r-- 1 root root 3175695080 Jun  9  2019 /library/downloads/maps/en-osm-omt_central_america_2017-07-03_v0.23.zip

> root@box:~# ls -l /library/working/maps/
total 4
drwxrwxr-x 4 root root 4096 Jun  7  2019 en-osm-omt_central_america_2017-07-03_v0.23

> root@box:~# du -hsc /library/working/maps/
3.1G    /library/working/maps/

> root@box:~# df
Filesystem     1K-blocks     Used Available Use% Mounted on
...
/dev/sda1       20509264 20492864         0 100% /
...

tim-moody, 2020-01-27:
Inaccurately estimated disk size, especially with maps, is a known problem because:

* The catalog has to have the correct size. 
* Zip files take an additional, unknown amount of space, which only matters during the install process and only if it fills up the disk before we can get rid of it. I used to double the space estimate for zips, but not sure I do it for maps. This is why the unzip step in the screen shot broke.
* We don't really know how much of a download has completed.

So if you want to cut it close you should probably do maps first to leave room for the zip.

The plan was to revisit all this when the next version of maps is complete.