iiab/iiab #2207: Smaller Map Pack needed for both (1) testing & (2) rapid understanding of what IIAB offers [OSM Vector Maps]

holta, 2020-01-27:
Currently testing is onerously & unnecessarily slow as a result our not having a small OSM Map Pack available.

Example: currently the smallest clickable/available Map Pack is ~3.77GB "Central America" which takes a long time to download, and cannot be installed on 20GB VM disks where BIG-sized IIAB has already been installed.  As it needs almost double the size to unzip even this smallest zip file: https://archive.org/download/en-osm-omt_central_america_2017-07-03_v0.23/

This is a known issue of course!  (But prioritizing testing is important _both_ to enable rapid understanding/adoption of our work _and_ to refine a quality product!)  So let's fix it in coming days/weeks/months as soon as @georgejhunt finds it realistic to do that.

georgejhunt, 2020-01-27:
I put san_jose back in and labeled it "World to Zoom 10".

holta, 2020-01-30:
Done, Thanks @georgejhunt

holta, 2020-01-30:
@georgejhunt I installed this on http://10.8.0.86/maps = http://10.8.0.86/osm-vector-maps/en-osm-omt_san_jose_2017-07-03_v0.23/ and streets do not show up after you zoom in past the satellite photos.

No matter whether I zoom into San Jose or Boston or otherwise.

As a testing tool it's absolutely fine with satellite photos alone of course &mdash; but are you aware of this issue?

holta, 2020-02-13:
@georgejhunt you're aware of this glitch right?

i.e. all streets not showing up when you zoom in, no matter where in the world you zoom in (even if you zoom into San Jose!)

Regardless whether classify this as "wontfix" or whatev...let us know what's likely happening here...if you know!  As originates from: https://archive.org/download/en-osm-omt_san_jose_2017-07-03_v0.23/en-osm-omt_san_jose_2017-07-03_v0.23.zip

holta, 2020-02-13:
@georgejhunt is aware of the issue, but we don't fully understand it yet why this San Jose map pack "displayed streets ~9 months ago, but it no longer does."  As can be seen here:

http://10.8.0.58/maps

Not a big deal here as "World to Zoom 10 1.69 GB" is really just a demo map pack &mdash; though George if you figure out what the issue is in coming months let us know if poss~

georgejhunt, 2020-02-13:
> map pack — though George if you figure out what the issue is in coming

I changed regions.json at unleashkids.org to point to  version v.2 of
san_jose map pack rather than v.23 and did a clean install on rpi4 of lite
9/26/19. The san_jose pack works correctly now. Thanks for noticing.

closed 2020-02-14 by holta