iiab/iiab #2208: Proposed OSM Vector Maps Data Structures

tim-moody, 2020-01-27:
###	External, global catalog
* Describes download sources
* Contains a list of all map sets available for download
* Key is map set id

### Locally installed catalog
* Same format, but serves as list of installed map sets
* Has attributes for map sets even that may no longer be in global catalog. 
* Key is map set id

###	Version index
* Maps generic name to version instance  
* Key is perma_ref.
* Question: is there more than one perma_ref per region or is perma_ref actually region

###	Map set meta data
* It would be nice to have meta data for a map set version stored in that set’s directory. This would allow regeneration of Locally installed catalog and Version index. 
* Same format as one item in Locally installed catalog.

tim-moody, 2020-01-27:
## Examples from Current Data

### External, global catalog
"en-osm-omt_spanish_speaking_regions_2017-07-03_v0.23" : {
      "region" : "spanish_speaking_regions",
      "center_lat": "-8.0",
      "center_lon": "-70.0",
      "date": "2017-07-03",
      "east": "-18.0",
      "north": "33.00",
      "osm_size": "7888969728",
      "perma_ref": "en-osm-omt_spanish_speaking_regions",
      "sat_is_regional": "False",
      "sat_size": "976416768",
      "sat_url": "https://archive.org/download/satellite_z0-z9_v3.mbtiles/satellite_z0-z9_v3.mbtiles",
      "seq": 25,
      "size": "8699784000",
      "south": "-84.0",
      "title": "Spanish-speaking Regions",
      "url": "https://archive.org/download/en-osm-omt_spanish_speaking_regions_2017-07-03_v0.23/en-osm-omt_spanish_speaking_regions_2017-07-03_v0.23.zip",
      "version": "v0.23",
      "west": "-179.00",
      "zoom": "3"
    }
### Version Index

{
  "en-osm-omt_world": {
    "file_name": "en-osm-omt_world_2017-07-03_v0.23",
    "menu_item": "en-osm-omt_world",
    "size": "55752568832",
    "date": "2017-07-03",
    "region": "world",
    "language": "en"
  }

holta, 2020-01-27:
Added to OSM Ideas Checklist @ #877 so we keep tabs on this!

holta, 2020-01-27:
Reminder: please add this proposal to central doc https://github.com/iiab/iiab/wiki/IIAB-Maps if it becomes "official", making it findable by Mappy People one & all!

holta, 2020-01-28:
Context for others...
https://github.com/iiab/iiab-admin-console/blob/master/roles/js-menu/tasks/main.yml#L132-L165

(A few days ago @jvonau earlier mentioned that's relevant...to working out OSM bugs like #2186, #2197 etc!)

tim-moody, 2020-01-28:
> Context for others...

This has nothing to do with existing bug, but rather with redesign.

jvonau, 2020-04-16:
Please clarify which should be the default regions.json used, the file that iiab installs retrieved from the web or the later version installed in admin-console. The differences between the two are in "regions": { } where the web supplied version has additional regions defined while the later admin-console does not. #2197 #2234 #2236

tim-moody, 2020-04-16:
admin

jvonau, 2021-07-21:
think this is in place now

holta, 2021-07-21:
@tim-moody should this be closed?

tim-moody, 2021-07-22:
I think this is now stale as some is implemented and some not planned to be. So I'm OK to close.

closed 2021-07-22 by holta