iiab/iiab #2209: Enabling cham

mitra42, 2020-01-27:
### Expected behavior and actual behavior.
I'm trying to enable cham to test it out ...
But its not in default_vars.yml

### Steps to reproduce the problem.
On Rasbian
`curl d.iiab.io/install.txt | sudo bash`
edit `/opt/iiab/iiab/varsdefault_vars.yml` when prompted
look for `cham` (or `Cham`)

holta, 2020-01-27:
@m-anish can you respond to @mitra42 above?

And if it will take a few days to prepare a Cham playbook for IIAB, can you model that on https://github.com/iiab/iiab/blob/master/roles/munin/tasks/main.yml please?

holta, 2020-01-27:
> edit /opt/iiab/iiab/varsdefault_vars.yml when prompted

The correct file to edit is [/etc/iiab/local_vars.yml](http://wiki.laptop.org/go/IIAB/FAQ#What_is_local_vars.yml_and_how_do_I_customize_it.3F) but yes that's the idea!

( [/opt/iiab/iiab/vars/default_vars.yml](https://github.com/iiab/iiab/blob/master/vars/default_vars.yml) itself should never be edited :-)

mitra42, 2020-01-29:
Am I correct in my reading of https://github.com/iiab/iiab/pull/1743 that we can't really test it until that PR gets merged ?

holta, 2020-01-29:
> Am I correct in my reading of #1743 that we can't really test it until that PR gets merged ?

@m-anish can you clarify?

m-anish, 2020-01-30:
@mitra42 @holta sorry, missed this, shall look into it and reply asap.

holta, 2020-01-30:
> @mitra42 @holta sorry, missed this, shall look into it and reply asap.

Thanks.  Even if the related #2157 disk discipline challenge is obviously not solvable this month ;)