iiab/iiab #2210: './runrole pbx' fails due to predefined var 'pbx_installed' in pbx/defaults/main.yml

holta, 2020-01-28:
`./runrole pbx` fails validation in 0-init for several quite subtle reasons...

Thanks @jvonau for uncovering this complex error, exacerbated by [./iiab-install](https://github.com/iiab/iiab/blob/master/iiab-install)'s Ansible syntax which fails to instantiate/get variable values within every roles/*/defaults/main.yml (whereas [./runrole](https://github.com/iiab/iiab/blob/master/runrole) DOES instantiate/get these variable values for the single role it's trying to run, and hence validates vars correctly, as defined across the many files where they might originate!)

It'll take some time/thought/engineering to resolve this multi-layered problem.  Thanks @jvonau working on that in coming days...

@m-anish: it may be the case (likely?) that we'll have to rename your variable `pbx_installed` &mdash; given that all such `<ROLE NAME>_installed` variable names are now reserved for /etc/iiab/iiab_state.yml (sorry if that's what happens in the end!?)

Anyway at last we now know the root cause, which boils down to [Line 17 of pbx/defaults/main.yml](https://github.com/iiab/iiab/blob/master/roles/pbx/defaults/main.yml#L17) here:

> pbx_installed: False

Which thankfully uncovered several much deeper Ansible issues...making clear that our new 0-init [variable validation](https://github.com/iiab/iiab/blob/master/roles/0-init/tasks/validate_vars.yml) is not quite complete/sufficient when invoked from [./iiab-install](https://github.com/iiab/iiab/blob/master/iiab-stages.yml) (but it works perfectly when invoked from [./runrole](https://github.com/iiab/iiab/blob/master/run-one-role.yml) !)

holta, 2020-01-28:
Related:
#2211 "Var Validation Explained: Error msgs that tell you what's happening & what to do"

holta, 2020-01-28:
Some of this Error Messaging should be further improved to mention:

```
cd /opt/iiab/iiab
./runrole --reinstall <ROLE NAME>
```

(And where to get more help about the `./runrole` command, as it and `./iiab-install` are both evolving, with #2210 etc.)

tim-moody, 2020-01-28:
I don't think the install and enabled vars should ever be in <role>/defaults except during development. We should think of vars/default.yml as global scope and able to be overridden. <role>/defaults are more like private.

jvonau, 2020-02-02:
the other error noted is the role's test for nodeje is outdated.

m-anish, 2020-02-02:
Its confusing to me from this discussion what the fix to this is. If you guys already know the problem and the changes dont change any underlying functionality, then go right ahead and make them :)

jvonau, 2020-09-07:
addressed in #2489

closed 2020-09-07 by holta