iiab/iiab #2212: Mediawiki - tweaks ?

mitra42, 2020-01-28:
I just installed `mediawiki` - using the standard IIAB installer and then tried (and eventually succeeded) to import the [palmleaf.org] wiki over it, and integrate it with the `internetarchive`.

I've got it working, so I don't *need* any changes, but some of the challenges I encountered might be worth fixing.  

 I'm not much of a mediawiki expert, and the docs on the mediawiki site are pretty bad, and I couldn't find anywhere to converse with mediawiki-ers (the talk pages on the mediawiki site dont generate responses), so maybe someone else has a channel to check with someone with more experience. 

#### Permissions: 
The directories (e.g. `images`) are `rwxr-xr-x root.www-data` but the maintenance scripts, which are (I believe) not intended to be run NOT as root, need to be able to write to those directories. In particular for example `maintenance/importImages.php` creates a bunch of subdirectories under `images` which fails.  I'd suggest one of two alternatives:
* in `roles/mediawiki/tasks/install.yml` that it should be `g+rw` not `g+r` OR
* OR in the standalone mediawiki I built I had the directories as `775 www-data.www-data` and then added `pi` to the group `www-data` so that these scripts could be run by a logged in user. 
I'm not sure which would be more mediawiki standard, personally I prefer the 2nd solution as it allows for example `composer` to be run as `pi` to install things. 

#### Extension assets: 
Extensions contain more than the files, so the line in `templates/mediawiki-nginx.conf.j2` 
`location ~ ^/mwlink/(skins|extensions)/.+\.(css|js|gif|jpg|jpeg|png|svg)$ {`
should be at least 
`location ~ ^/mwlink/(skins|extensions)/.+\.(css|js|gif|jpg|jpeg|png|svg|ttf|woff|woff2)$ {` and possibly needs other types. 

#### mwlink or w
There seems to be a semi-standard convention of things living under `/w`, enough that I found links to `/w/api.php` in particular in various places. On IIAB its all under `/mwlink` I suspect other things will be broken by this change, running `grep -lR '/w/' /library/mediawiki-1.34.0 ` shows this assumption appears in many places. 

I had to add 
```
# ArchiveLeaf assumes extension is at /w
location ^~ /w/api.php {
  rewrite ^/w/(.*)$ /mwlink/$1;
}
```
to `mediawiki-nginx.conf.j2` to fix one of our extensions, but I suspect other things will break and unless its going to break something else (outside mediawiki), I'd expand this to match `/w/`.

holta, 2020-01-28:
Thanks much @mitra42 for writing up your experiences.

I'll ask @tim-moody & others to see if they can investigate improvements similar to what you suggest in coming weeks here.

Feel free also to submit your own PR here, e.g. if you see concrete/immediate improvements possible?

tim-moody, 2020-01-28:
Thanks for comments. This was a first pass to get what was very easy with apache to work with nginx.

I thought that mwlink was softcoded, but I see that it is not. I think I tried /w/ and had problems, but maybe not. 

The only issue might be security by opening up permissions on image, etc.

mitra42, 2020-01-28:
I didn't write it up as a PR because while my suggestions are concrete, I'm not enough of a mediawiki expert to know whether they are right.  It sounds like @tim-moody is onto changing it ...

tim-moody, 2020-01-30:
closing. reopen if necessary.

closed 2020-01-30 by tim-moody