iiab/iiab #2219: Search engine API proposal

traverseda, 2020-01-31:
I'm working on a search engine, the API was originally discussed in #2218 , here it is in it's own issue.

>lets collaborate on an API so that an App can feed the data to your search engine,

Absolutely. The first step, in my opinion, would be figuring out a data structure of articles. Here's a first draft. Let's take a look at some of the fields. Everything is optional except `id`, of course you need to include some of these if you want it to be actually searchable.

 * id: A unique id for this article, a lot of the time this can just be a URL but when dealing with WARC files that store multiple versions of the same URL we will need to take a slightly more nuanced approach. Adding two documents with the same ID replaces the older one.

 * title: The header tag that returns with each entry.

 * real_url: the actual url to the content, if it differs from what we display in our results.

 * url: the url to show in the search results. If the content originally came from wikipedia it should show wikipedia.org/somearticle

 * summary: The html document to display when we show a summary. We strip the text out of the html and store that.

 * summary_nostore: Like summary, but don't save the resulting document. Must be used with `text_url` if you want to show any kind of summary.

 * text_url: A url returning an HTML representation of a page. We use the same text-extraction as we do for the summary. Of course you can also return the text representation of the page, and that save us a step.

 * data_url: A url that returns a json-encoded "mask" that can cover any of these fields for display. Allows for more dynamism in the data display, as it's loading data at display-time.

 * display_type: If the article is a video, image, or audio file, we can specify a custom display_type which tells the search UI to render the result differently.

 * search_boost: Request that this result be more/less relevant than it normally would be. I'll try to find some appropriate ranges for this. Indexes of source code should probably be less relevant that indexes of wikipedia/stackoverflow, as an example. Indexes of user-created content should probably be the most important.

---

You can of course include custom fields, and along with `display_type` show them in search results. In the future meilisearch might support faceting, and those custom fields should be usable for faceting some time in the future.

Does that make sense for an initial data type specification? Presumably there would be a url endpoint somewhere where you can call that accepts those data types, and another that lets you delete an article by ID. Also a command line utility that lets you GC dangling articles, articles that don't exist any more but which weren't cleaned up properly.

mitra42, 2020-01-31:
There may be different requirements. 

I’m starting from …. metadata: { description, author, title, mediatype, etc …. } others might be starting from just { url, text } 

First thing I notice is a mix in this straw-person proposal of options, and fields. For example `summary` `summary_nostore` I’d suggest that the API should  separate the fields/data from what you want the search engine to do with them (index &/or display). 

So I’d suggest a record is ….
{ id, title, url, description (default to blank), display_url (default to url), data } 

Where data could be 
HTML || TEXT || OBJECT 
Where OBJECT is just a JSON object with fields that should be indexed. 

And any of these could be presented directly or via a URL. 

Open questions for me … 
Do any other metadata want conveying to the search engine for display e.g. 
mediatype (audio|image|texts|software|movies) - these terms need defining - this is the set we use at internet archive
date - if so which - created or updated 

Anything else - like search_boost should probably be in a opts field to allow for future expansion.

mitra42, 2020-01-31:
In terms of end points - yes URL and possibly needs to be both-way , a way for the search engine to initiate an initial dump of records, and a way for the app to incrementally add them. 
OR a way for the search-engine to tell the app to feed it all the data again.

traverseda, 2020-01-31:
So meilisearch is *essentially* a json document store, but where each field is also added to a search index. Everything except `id` would be optional. You can also add other fields to your json call, and they'll get stored in the database.

You could absolutely add `description, author, title, mediatype` to the json document you're posting, and they'd be stored/indexed. You only include the actual fields you want in your request. Essentially everything is `OBJECT`.

It's a bit messy, since there really isn't any typing and this is all just convention, and you can post anything to the endpoint and it will get stored, those are just the fields we use.

There isn't necessarily a difference between the data in something like `text_to_index` and `text_to_display`, so I'd rather not inline them twice. I could add something like `store_summary: True`, but if I have two separate fields I can essentially just dump whatever json-document the client provides into the database/search-engine.

Summary is a kind of confusing name for those fields, so I'll change it to `content`.

>mediatype (audio|image|texts|software|movies) - these terms need defining

My intent would be to allow apps to package their own template to render search results, so apps can provide their own `display_type`. We will of course need to ship a few by default, and when they get defined we'll need to document them.

traverseda, 2020-01-31:
Hmm, looks like meilisearch is less of an arbitrary json-document store than I though. We may need to provide a schema.

Still, whether or not a field is stored or just indexed is determined on a per-field basis, so we still need two separate fields for our content, one for if it should store the a copy of the document and one for if it should just index it.

mitra42, 2020-01-31:
I'm presuming you mean the template for when a page of results is displayed, before the user clicks through to the URL. 

I think having that as a template isn't going to work well - its going to be seriously confusing to have results from internetarchive look totally differnt from those from kiwix when presented in a list of results. 

Understood about json store, so a) if you have a template then "text_to_index" v "text_to_display" is handled by the template. 
or b) if no template, I'd suggest a configuration field that decided which fields to show for each app. (in fact that could be a per-app-template managed by the search engine) ?

mitra42, 2020-01-31:
(that comment crossed-in-the-post with yours)

mitra42, 2020-01-31:
You are going to have to do some kind of processing anyway, for example to decide what to do with the two different urls.

traverseda, 2020-01-31:
What's the difference between the description field and the data field? I'm presuming that we're using highlighting to show the relevant parts of the content. Something like this

[image]

Custom templates would have to be limited to places where it really makes sense, and we'd have to provide a set of sane defaults for things that are common. I assume that both a kiwix result and an internet-archive result would use the same `display_type`.

closed 2020-05-31 by traverseda

holta, 2020-05-31:
Please see:

"Review search engine" #2429