iiab/iiab #2220: Dnsmasq problems on ubuntu install via wifi

georgejhunt, 2020-01-31:
http://paste.ubuntu.com/p/RKfKc7MYhK/
I'm sure more background will surface

holta, 2020-01-31:
This definitely affects the desktop version of Ubuntu 18.04 (which is very popular out there in the wild, love it or hate it!) and _possibly does not_ affect the server version of Ubuntu 18.04.

That's my assumption anyway.  @jvonau suggests the issue relates to NetworkManager (which is a part of the desktop version but not server version of Ubuntu releases, if I recall correctly!)

holta, 2020-01-31:
Further Context:

Different generations disagree (such is life, an eternal historical fact sadly?) but in any case this is one such example...where the younger generation *almost always* strongly prefers graphical OS's (e.g. Ubuntu 18.04.3 with desktop) to build up their understanding of Linux & IIAB.

I've spoken to many such people who tell me (in no uncertain terms!) that they need and want a built-in browser on the same computer, so as to get to know IIAB by provisioning content over WiFi, without any possible networking hassles or risks (I typically advise such people to use Ethernet instead of WiFi to build up their IIAB, but they cannot in so many such situations, as Ethernet connections to the Internet are typically impossible to come where they live...)

georgejhunt, 2020-01-31:
It looks to me that Jerry is testing for wan iface being the same as lan iface, and disabling the definition of iiab_wireless_interface
I'd like to chane line 139 👍 
when: discovered_wireless_iface is defined and discovered_wireless_iface != "none" and discovered_wireless_iface != iiab_wan_iface
to
when: discovered_wireless_iface is defined and discovered_wireless_iface != "none" 
to get rid of the lack of definition.
But I'm not in contack with Victor, and it should be done with communication

jvonau, 2020-01-31:
I don't think you want to enable hostapd to bring up br0 while the wifi network interface is being managed by NetworkManager to supply the default gateway. Look at the set_fact no_net_restart in main.yml that is in place to ensure the wifi is not interrupted, files are slid into place but no attempt is made to start br0 at this point, would be active on reboot. Yes hostapd.conf has a blank entry but the service is disabled from starting on boot in [hostapd.yml] so the default gateway is still available. Your answer is at the bottom of restart.yml in one of the # out parts.

jvonau, 2020-01-31:
Now the question becomes should we even configure br0 at all. Disabling auto LAN used to be possible by placing iiab_lan_enabled: False in local_vars.yml as an override. Due to a regression where gui_desired_network_role is always defined in defaults/main.yml taking precedence the value of user_lan_iface and iiab_lan_enabled are ignored but otherwise would result in appliance mode being applied.

georgejhunt, 2020-01-31:
I don't really know what is wanted. During install the wifi is wan. How do we do the transition to having the device offline, and both eth0 and wifi under the bridge? Is that what is wanted?
 
For ubuntu 1804, seems like a script that turns everything around and restarts would be simpler. I don't think this case is main line

holta, 2020-01-31:
> I don't really know what is wanted. During install the wifi is wan. How do we do the transition to having the device offline, and both eth0 and wifi under the bridge? Is that what is wanted?
>
> For [...]

Good questions.  For now IIAB's 1-line installer (http://download.iiab.io) offers these `/usr/bin/iiab-hotspot-off` instructions to complete a WiFi-based install of IIAB...
https://github.com/iiab/iiab-factory/blob/master/iiab#L16-L18

And also hereunder...these instructions need to evolve for sure...but let's start with something like the status quo here...towards keeping this sane for less-technical implementers of which there are a growing many ;)
https://github.com/iiab/iiab-factory/blob/master/iiab#L305-L309

jvonau, 2020-01-31:
Getting through the install is one issue, suppress dnsmasq restart with no_net_restart but br0 will be present but no slaves, dnsmasq should start but might fail but with no clients to serve not a great loss. The other is after reboot behavior, with iiab_lan_enabled: working correctly False would force appliance mode with no br0

To later enable the hotspot AP the user must first disconnect from the upstream connection then run ./iiab-network to ensure the br0, hostapd, and dnsmasq files are present and handles the restart the network stack.

The use of wired slaves under br0 needs further discussion.

georgejhunt, 2020-01-31:
To get Victor up and running (my take on Jerry's response):
1. Hand edit /etc/dnsmasq.conf  setting interface to wlxd0374559ad8c
(because right now it is set to non-existent br0)
1.5 Systemctl start dnsmasq -- if it does not work, then we will need
to fix other things until dnsmasq can be started.
2. Edit local_vars.yml to include "no_net_restart: True" ( so that
dnsmasq does not try to restart, and refuse to start a non-existent
br0)
3. Then do a ./iiab-install --reinstall -- hopefully it will run to completion.

Then maybe iiab-hotspot on will turn wifi around?
Jerry, what do you think?

georgejhunt, 2020-01-31:
or maybe just set usr_lan_iface: wlxd0374559ad8c which might override
the automatic code setting it to br0.
I do think we need to give Victor a solution which can be restarted,
which the above does not do.

holta, 2020-01-31:
Victor's not at all a Linux expert so the dream would be for him to run something like `cd /opt/iiab/iiab && git pull && ./iiab-network && /usr/sbin/iiab` to wrap up his 1-line install of IIAB &mdash; but of course that luxury might be many, many days off.

So he'll just have to sit tight for that kind of thing to be mainlined into IIAB.

Shorter Term: let's feel free to attempt a patch on his live/current machine if it that has a reasonable probability of working &mdash; and most important serves to inform the design IIAB that should be aiming to help people/groups like his.

Nothing ventured nothing gained!

georgejhunt, 2020-01-31:
> /usr/sbin/iiab to wrap up his 1-line install of IIAB — but of course that
> machine if it that has a reasonable probability of working — and most

So Jerry, is there a combination of local_vars.yml which will force ubuntu
to appliance mode with wifi as wan?

jvonau, 2020-01-31:
Yes but it's broken by setting 'gui_desired_network_role' in default/main.yml # that out, that also broke usr_lan_iface btw.

next add  iiab_lan_enabled: False to local_vars.yml to force appliance mode
run ./iiab-network (why on earth would you suggest forced full reinstall?)
This will keep the upstream wifi working with no br0 preset and hostapd not configured.

To enable hotspot right click disconnect from wifi in the gui, now there is no gateway tying up the device, toggle iiab_lan_enabled: True 
run ./iiab-network

holta, 2020-01-31:
This looks like really great progress thanks to @georgejhunt & @jvonau.

Just spoke to Victor who's excited to get back on track installing & learning IIAB in coming days, for the flood of refugees he's supporting...

holta, 2020-01-31:
Victor's Current Plan, FYI:

1) His getting to know IIAB (provisioning content etc, without any attached client machines) using this IIAB Ubuntu 18.04.3 Desktop PC is the priority for the coming week.

2) Then Victor will find an external Wi-Fi router to attach to this IIAB PC (using an Ethernet cable) for refugees and their children, allowing them Wi-Fi access the highest quality IIAB learning content that he is able to assemble.

3) The IIAB PC's own "internal" Wi-Fi should remain connected to the Internet for occasional content updates etc in coming days/weeks/months.

4) Victor will toggle the Internet (the upstream/internal WiFi connection) on and off regularly, using Ubuntu's widget (in the top-right of Ubuntu 18.04.3's desktop) when he needs content updates and/or IIAB updates.

5) This IIAB PC's external Wi-Fi router should NOT give Internet access to refugees & their children at this time — and for the forseeable future — for many different financial & pedagogically-focused reasons.

6) Conceivably this policy decision could change in the more distant future, whereupon the teacher/coordinator would be trained to use any form of "Internet Master Switch" — to regularly enable & disable Internet access to all client machines — depending on the lesson plan for that particular hour, or possibly even minute-by-minute.  i.e. the Internet could be toggled ON/OFF during different moments, e.g. during particular lessons requiring Internet access to materials (or Internet conversations) not available within IIAB.

7) **If channel / radio frequency conflicts are likely between the IIAB PC's 2 different Wi-Fi radios ("upstream = internal Wi-Fi" vs. "downstream = external Wi-Fi"), our suggestions on how to mitigate the worst such known problems would be most appreciated!**

tim-moody, 2020-01-31:
For content provisioning only and only on rpi you can use Admin Console to set wan wifi password and switch wifi to wan. You can also turn on bluetooth so that you retain a client connection.

holta, 2020-01-31:
Just for the interim, @georgejhunt is going to try getting dnsmasq to start manually, and then hopefully automatically on each reboot (hopefully OpenVPN will auto-start too!)  So George will give this a shot on Victor's IIAB PC in coming hours.

Worth a shot.  And if that works out, we'll ask Victor not to run `./iiab-network` for now, out of an abundance of caution &mdash; at least until he finds an external WiFi router, hopefully in February sometime soon?

Aside: @georgejhunt will not try to create br0 at this point.  For now Victor just wants to get his feet wet e.g. using Admin Console and familiarizing himself with getting purposeful content/presentation/apps.

Then hopefully stuff like `./iiab-network` or any other necessary network improvements/changes/etc can happen sometime later in February we hope.

holta, 2020-01-31:
Aside: the `reserved_wifi` flag documented below might be useful later in Feb?

> (2) If your Internet-in-a-Box (IIAB) contains multiple Wi-Fi interfaces, put the following into /etc/iiab/local_vars.yml to ask IIAB not to create a LAN-side access point on the wlan0 interface:
>
> `reserved_wifi: wlan0`

Per https://github.com/iiab/iiab/wiki/IIAB-Networking#common-customizations

georgejhunt, 2020-01-31:
Log of GH actions:
1. set interface in /etc/dnsmasq.d/iiab.conf to wlxd0374559ad8.
2. dnsmasq started successully
3. set roles/netaork/defaults/gui_desired_network_role to Appliance
4. ran iiab-admin-console ./install and kiwix install by hand
5. tested http:// xxxx/home and xxxx/admin -- both succeeded
6. ran "./iiab-network" which completed and did not disconnect vpn
7. new iiab-diagnostics at http://paste.ubuntu.com/p/K7cCCNwBHJ/
8. Then I rebooted the client and lost my connection ("victor" did not show up on xscenet.net)
9. the diagnostics indicate that network_applied was Appliance -- so apparently that was not enough to let it reboot properly)

holta, 2020-01-31:
Valliant.  Thanks so much @georgejhunt for the careful effort.  And @jvonau for having informed the situation.

I'll ask Victor to assess whether the PC's booting at all (and try to assess what state it's in if so) when I reach him in the coming ~24h.

Whereas the prior effort (on Monday, on a different PC of his) failed to reboot at all, as if Ubuntu 18.04.3 was somehow accidentally damaged by IIAB's installation process.  Let's hope that's not the case, but if we have a pattern here we'll investigate in coming days.

holta, 2020-01-31:
News: Victor's IIAB PC is in great shape now, and he's using it (Admin Console) to install content etc!

What I misunderstood earlier is that this is an i5 desktop PC whose Wi-Fi really isn't internal.  In fact it's a TP-LINK TL-WN823N USB dongle, that will be repurposed later.  Anyway for whatever reason: Victor has to manually turn on this Wi-Fi dongle's connection to the Internet after every reboot, using Ubuntu 18.04.3's graphical widget in the top-right of his screen.  So remote reboots are unfortunately not practical, but he and we can all live with that!

Thanks again to @georgejhunt & @jvonau for helping him bootstrap to where he needs to be &mdash; good enough for now anyway &mdash; as he explores + installs + customizes content &mdash; and also a few apps that should work fine on this relatively beefy server...where he'd line to upgrade the RAM from 4GB to 8GB just to be safe!

_Asides: Victor will try to buy a WiFi router or get one donated in early February hopefully.  The languages of choice so far are English, Greek, French, Farsi (and other Afghanistan languages?), Arabic, and possibly a few others.  He's got a lot to learn about IIAB but is thrilled this is making good progress now._

georgejhunt, 2020-01-31:
Three hours later, the ubuntu machine with the handle 'Victor' appears at
xscenet.net.
The history command does not indicate any command line actions since my
last reboot.
Trying home page services: The following fail:
moodle, maps
Then Victor went offline again.
The moodle service is really slow.
Thanks for the update. I realize that the slowness may have been associated
with something going on interactively -- There was a local sesion which I
did not factor in.

holta, 2020-01-31:
> The following fail: moodle, maps

Thanks for the warning!  I'll look into those live with Victor later this wkd, with an actual screenshare as nec.  He's learning IIAB fast, but this is all very new to him, and a screenshare session (TeamViewer of whatever) can greatly accelerate community action...

> Then Victor went offline again.

Yes: (Victor the person) is turning it off (Victor the PC) when he's not using it to save green(earth) + green($).  Or we can ask him to leave it fully online day/night for a few more days if that's important, as he'll often be download content regardless...

> The moodle service is really slow. Thanks for the update. I realize that the slowness may have been associated with something going on interactively -- There was a local sesion which I did not factor in.

Indeed Victor himself is now using it to [download content & evaluate apps](https://github.com/iiab/iiab/issues/2220#issuecomment-580919445) like Kolibri etc.

jvonau, 2020-02-01:
> Log of GH actions:
> 
>     1. set interface in /etc/dnsmasq.d/iiab.conf to wlxd0374559ad8.
> 
>     2. dnsmasq started successully
>

Now dnsmasq is running as dhcp server on the upstream WAN, great way to get banned from the upstream ISP.

>     3. set roles/netaork/defaults/gui_desired_network_role to Appliance
> 
>     4. ran iiab-admin-console ./install and kiwix install by hand
> 

Good, that would of deleted /etc/dnsmasq.d/iiab.conf hiding the mis-edit. 

> 5. tested http:// xxxx/home and xxxx/admin -- both succeeded
>
>     6. ran "./iiab-network" which completed and did not disconnect vpn
>
>     7. new iiab-diagnostics at [...]

As noted, desktops tend to hold off the connection until a user has logged into the desktop.

holta, 2020-02-01:
> > 2. dnsmasq started [successfully]
> 
> Now dnsmasq is running as dhcp server on the upstream WAN, great way to get banned from the upstream ISP.

Thanks much @jvonau for pointing out this risk.  And thanks again to @georgejhunt for the telemedicine putting the patient back on his feet.

If there are rough instructions/guidelines, I'll fix this potential ISP risk the next time I'm working live in person with Victor.

(He'd doesn't live the command-line, but he uses it and he'd love each step documented & reproducible so he can become more independent, making other IIAB's as necessary for the 20,000 refugees in the island.)

holta, 2020-02-02:
Here's a possible fix, thanks to @jvonau:

PR #2226 "wifi install dnsmasq restart issue"

Many thanks to @jvonau, @georgejhunt and everybody.

PR #2259 (2020-02-17): Merge @jvonau's PR #2226 fix for WiFi installs
While we haven't tested Ubuntu 18.04 where the original problem occured, this _should_ fix #2220 in general.

Thanks to @jvonau's PR #2226 (basically the same as this PR, which includes it) which has now been well tested over 2 different WiFi installs onto RPi 4).

Thanks also to @georgejhunt who did the "telemedicine" fixing Victor's original machine in the field, via OpenVPN.

And to @tim-moody who helped with the discussion this past Thursday.
closed 2020-02-17 by holta