iiab/iiab #2221: lokole fails to download on IIAB 7.1 with Raspbian-Buster-Lite Build on PI3B+

shanti-bhardwa, 2020-01-31:
### Expected behavior and actual behavior.
get this error when running one line script which has been copied to
sudo sh iiab-7-1-install.sh
with curl d.iiab.io/install.txt -o iiab-7-1-install.sh
iab-7-1-install.sh created from this command

the results of iiab-diagnostics are here:
http://paste.debian.net/1128554/

The errors is
here is the text from log file*****************************************************************************************************************
2020-01-30 11:57:32,757 p=3596 u=root n=ansible | fatal: [127.0.0.1]: FAILED! => {"changed": true, "cmd": ". /home/lokole/state/webapp_secrets.sh\n
/library/lokole/venv/bin/manage.py createadmin --name='admin' --password='changeme'\n"
, "delta": "0:00:05.099453", "end": "2020-01-30 11:57:32.579072", "msg": "non-zero return code", "rc": 1, "start": "2020-01-30 11:57:27.479619", "stderr": "Traceback (most recent call last):\n
File \"/library/lokole/venv/bin/manage.py\", line 15, in <module>\n

from opwen_email_client.webapp import app\n
File \"/library/lokole/venv/lib/python3.7/site-packages/opwen_email_client/webapp/__init__.py\", line 14, in <module>\n
from opwen_email_client.webapp import views # noqa: F401,E402 # isort:skip\n
File \"/library/lokole/venv/lib/python3.7/site-packages/opwen_email_client/webapp/views.py\", line 29, in <module>\n
from opwen_email_client.webapp.forms.email import NewEmailForm\n
File \"/library/lokole/venv/lib/python3.7/site-packages/opwen_email_client/webapp/forms/email.py\", line 14, in <module>\n
from PIL import Image\n
File \"/library/lokole/venv/lib/python3.7/site-packages/PIL/Image.py\", line 90, in <module>\n
from . import _imaging as core\n

ImportError: libopenjp2.so.7: cannot open shared object file: No such file or directory", "stderr_lines": ["Traceback (most recent call last):", " File \"/library/lokole/venv/bin/manage.py\", line 15, in <module>", " from opwen_email_client.webapp import app", " File \"/library/lokole/venv/lib/python3.7/site-packages/opwen_email_client/webapp/__init__.py\", line 14, in <module>", " from opwen_email_client.webapp import views # noqa: F401,E402 # isort:skip", " File \"/library/lokole/venv/lib/python3.7/site-packages/opwen_email_client/webapp/views.py\", line 29, in <module>", " from opwen_email_client.webapp.forms.email import NewEmailForm", " File \"/library/lokole/venv/lib/python3.7/site-packages/opwen_email_client/webapp/forms/email.py\", line 14, in <module>", " from PIL import Image", " File \"/library/lokole/venv/lib/python3.7/site-packages/PIL/Image.py\", line 90, in <module>", " from . import _imaging as core", "ImportError: libopenjp2.so.7: cannot open shared object file: No such file or directory"], "stdout": "", "stdout_lines": []}
2020-01-30 11:57:32,771 p=3596 u=root n=ansible | PLAY RECAP

### Steps to reproduce the problem.

### Specifications like the version of the IIAB, operating system version, or hardware details.

holta, 2020-01-31:
1) @shanti-bhardwa, is this issue reproducible every time, when you run the following?

   ```
   sudo su -
   cd /opt/iiab/iiab
   ./runrole lokole
   ```

2) On different RPi's ideally?  e.g. on my own RPi 4 (2GB RAM), Lokole installed successfully yesterday (as part of [BIG-sized IIAB](http://wiki.laptop.org/go/IIAB/local_vars_big.yml)) on Raspbian Lite.

   As such are we able to isolate what machines this _is_ affecting, and what machines it is _not_ affecting?

3) Finally, are you running bleeding-edge kernel 4.19.97 for a reason?  As shown in the output of your iiab-diagnostics posting at http://paste.debian.net/1128554/

   (Obviously you're taking a risk there, which is why I ask if you are perhaps running 'rpi-update' for a reason, despite all its dangers?!)  In any case, for comparison most IIAB installations today use Raspbian's safer kernel 4.19.93, as auto-installed by IIAB's 1-line installer (http://download.iiab.io) via the OS's apt updates.

cc @c-w, @aidan-fitz

holta, 2020-01-31:
@shanti-bhardwa note that the page you created today http://paste.debian.net/1128554/ will disappear in less than 7 days!

(Consider pasting this text &mdash; i.e. your output from the [iiab-diagnostics](https://github.com/iiab/iiab/blob/master/scripts/iiab-diagnostics.README.md) command &mdash; into http://paste.debian.net with a longer expiration date [90 days, or more?] if this issue might not be solved within the week?!)

jvonau, 2020-02-01:
I appears there is an undeclared dependency on libopenjp2.so.7 which is missing from this this roles' install routine. 
On a hunch:
> apt list libopenjp2*
Listing... Done
libopenjp2-7-dev/testing 2.3.1-1 armhf
libopenjp2-7/testing,now 2.3.1-1 armhf [installed,automatic]
libopenjp2-tools/testing 2.3.1-1 armhf`

This file dependency it appears to be supplied by libopenjp2-7. The automatic notation show this is a dependency of another package
>apt-cache rdepends --installed libopenjp2-7
 libopenjp2-7
Reverse Depends:
  libpoppler82
  libmagickcore-6.q16-6
  libgs9
  libpoppler82
  libmagickcore-6.q16-6
  libgs9
 apt-cache rdepends --installed libmagickcore-6.q16-6
libmagickcore-6.q16-6
Reverse Depends:
  php-imagick
  libmagickwand-6.q16-6
  libmagickcore-6.q16-6-extra
  libmagickcore-6.q16-6-extra
  libmagickcore-6.q16-6-extra
  imagemagick-6.q16
  php-imagick
  libmagickwand-6.q16-6
  libmagickcore-6.q16-6-extra
  imagemagick-6.q16

Bingo php-imagick is installed by roles/nextcloud/tasks/install.yml but we have nextcloud_install: False in local_vars.yml, so the libopenjp2-7 deb package needs to be added to lokole's install.yml

holta, 2020-02-01:
Thanks @jvonau:

> Bingo php-imagick is installed by roles/nextcloud/tasks/install.yml but we have nextcloud_install: False in local_vars.yml, so the libopenjp2-7 deb package needs to be added to lokole's install.yml

Coincidentally Nextcloud is installed almost immediately after Lokole (these being largely in alphabetical order in Stage 6) which would explain why the issue was never quite observed!

jvonau, 2020-02-01:
ok that would not prevent the error as the dependency is installed after the lokole install, but cups is installed first also install has false:
>pi@box:/opt/iiab/iiab $ apt-cache rdepends --installed libgs9
libgs9
Reverse Depends:
  ghostscript
  ghostscript
pi@box:/opt/iiab/iiab $ apt-cache rdepends --installed ghostscript
ghostscript
Reverse Depends:
  php-imagick
  netpbm
  libmagickcore-6.q16-6
  imagemagick-6.q16
  cups-filters
  cups-daemon
  cups
  cups-filters
  poppler-data
  poppler-data
  php-imagick
  netpbm
  libmagickcore-6.q16-6
  imagemagick-6.q16
  cups-filters
  cups-daemon
  cups
 
Just to complete the tread started 
>pi@box:/opt/iiab/iiab $ apt-cache rdepends --installed libpoppler82
libpoppler82
Reverse Depends:
  libpoppler-cpp0v5
  poppler-utils
  libpoppler-cpp0v5
  cups-filters-core-drivers
  poppler-utils
  cups-filters-core-drivers

holta, 2020-02-01:
> the libopenjp2-7 deb package needs to be added to lokole's install.yml

I'll implement that in coming hours and ask @shanti-bhardwa to test.

Thanks again @jvonau for spotting this.

Perhaps Lokole's authors (@c-w, @aidan-fitz) could consider publishing a list of its official requirements, so we & all can avoid similar/such bugs in future?

holta, 2020-02-01:
@shanti-bhardwa please re-open this tkt if you discover the applied fix (PR #2224) is somehow not enough, Thanks!

closed 2020-02-01 by holta

jvonau, 2020-02-02:
With the install portion of the role already completed to apply this fix one would need to:
`cd /opt/iiab/iiab`
`sudo git pull` to get the latest code with this fix
`sudo ./runrole --reinstall lokole` to fully rerun the failed role with the fix in place, if this fails make a note of it here or report its success please.  
`sudo iiab` or `sudo ./iiab-install` whichever you are using to continue with install from where the failure occurred.

PR #2224 (2020-02-01): (1) include Lokole dependency libopenjp2-7 (2) always install latest Lokole from pip (3) clean playbook
Fixes #2221, we hope!

For @shanti-bhardwa, thanks to @jvonau.

@c-w & @aidan-fitz please also review!

Smoke-tested on Raspbian Lite.