iiab/iiab #2222: iiab_admin_user_install: False trap

tim-moody, 2020-01-31:
If you set iiab_admin_user_install: False, iiab-install will fail in stage 4. However, it is not enough to go back and set it to True, because the stage in which it is to be set has already been marked as complete. Therefore you must begin again with iiab-install --reinstall.

holta, 2020-02-01:
> Therefore you must begin again with iiab-install --reinstall.

Tangentially Related, is a design question about whether './iiab-install --reinstall' should 'rm /etc/iiab/iiab_state.yml' (or most/all of its contents?)

Background:
- './runrole --reinstall <ROLE NAME>' deletes (1 line) from /etc/iiab/iiab_state.yml
- './iiab-install --reinstall' does not currently delete (any lines) from /etc/iiab/iiab_state.yml

jvonau, 2020-02-01:
Great bug report 'stage 4 failure', recall which sub-role was involved? I'll go out on a limb and suggest it was iiab_lib as #1228 suggest that iiab_admin_user is the owner of the content directory.

holta, 2020-02-01:
> Great bug report

#2222 is easy to remember too ;)

tim-moody, 2020-02-01:
> recall which sub-role was involved?

no. which is why I didn't speculate.

holta, 2020-02-01:
I'm assuming this was closed by accident?

(Seems like something that should be fixed eventually.)

jvonau, 2020-02-02:
> If you set iiab_admin_user_install: False, iiab-install will fail in stage 4. However, it is not enough to go back and set it to True, because the stage in which it is to be set has already been marked as complete. Therefore you must begin again with iiab-install --reinstall.

> Tangentially Related, is a design question about whether './iiab-install --reinstall' should 'rm /etc/iiab/iiab_state.yml' (or most/all of its contents?)
>
> Background:
>
>     * './runrole --reinstall [...]
I feel --reinstall is more targeted at replacing files that iiab provides in stage 1-3 when we have something changing in the base install, like adding a font in httpd or tinkering with iiab_lib.py
I'd suggest --upgrade for any planned programmatic deletions from /etc/iiab/iiab_state.yml where we know in advance when some app's release versions have been bumped.

> 
> Somewhat related to:
> 
> #2210 "'./runrole pbx' fails due to predefined var 'pbx_installed' in pbx/defaults/main.yml -- illustrates weakness of var validation when invoked from ./iiab-install (instead of from ./runrole)"

Tangentially Related, running runrole <role> with the setting <role>_install: False results in the role being installed anyway where the systematic removal of the <role>_install conditional within the roles' main.yml has been applied.

holta, 2020-02-13:
Thanks @jvonau for uncovering the link to #1228 which I'd missed earlier.

I'll make that change to `/library/www/html/local_content` group perm shortly, per @jvonau & @tim-moody's recommendation.

holta, 2020-02-14:
Mitigated by PR #2250.

@tim-moody will also address this within https://github.com/iiab/iiab-admin-console to make things more robust.

closed 2020-02-14 by holta