iiab/iiab #2223: rpi - apt upgrade - os not detected

jvonau, 2020-02-01:
>pi@box:/opt/iiab/iiab $ sudo ./runrole all-vars
        "ansible_python": {
            "executable": "/usr/bin/python3", 
            "type": "cpython", 
        "ansible_python_version": "3.7.6", 
        "ansible_selinux_python_present": false, 
PLAY [all] *********************************************************************
TASK [Gathering Facts] *********************************************************
ok: [127.0.0.1]
ERROR! vars file vars/{{ ansible_local.local_facts.os_ver }}.yml was not found
Could not find file on the Ansible Controller.
If you are using a module and expect the file to exist on the remote, see the remote_src option

> pi@box:/opt/iiab/iiab $ sudo ./iiab-configure
Extracted STAGE=9 (counter) from /etc/iiab/iiab.env
        "ansible_python": {
            "executable": "/usr/bin/python3", 
            "type": "cpython", 
        "ansible_python_version": "3.7.6", 
        "ansible_selinux_python_present": false, 
PLAY [all] *********************************************************************
TASK [Gathering Facts] *********************************************************
ok: [127.0.0.1]
ERROR! vars file vars/{{ ansible_local.local_facts.os_ver }}.yml was not found
Could not find file on the Ansible Controller.
If you are using a module and expect the file to exist on the remote, see the remote_src option

> pi@box:/opt/iiab/iiab $ scripts/local_facts.fact 
{"phplib_dir"             : "/usr/lib/php",
"stage"                   : "9",
"dhcpcd"                  : "enabled",
"network_manager"         : "",
"systemd_networkd"        : "disabled",
"iiab_branch"             : "grouping",
"iiab_commit"             : "70f5b0aba65b3b1fbd4dc1184bb428e9f596392c",
"xo_model"                : "none",
"ansible_version"         : "2.9.4",
"os"                      : "raspbian",
"os_ver"                  : "OS_not_supported"}

>pi@box:/opt/iiab/iiab $ cat /etc/os*
PRETTY_NAME="Raspbian GNU/Linux bullseye/sid"
NAME="Raspbian GNU/Linux"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

>pi@box:/opt/iiab/iiab $ cat /etc/debian_version 
bullseye/sid

holta, 2020-02-01:
> pi@box:/opt/iiab/iiab $ cat /etc/debian_version
bullseye/sid

Was this just pushed by Raspbian?

If so does this mean we need a vars/raspbian-11.yml now? (even 1.5 years before Debian Bullseye's release!)

jvonau, 2020-02-01:
No clue just updated with apt. No VERSION_ID is missing from /etc/os-release no clue why.

holta, 2020-02-01:
> No clue just updated with apt. No VERSION_ID is missing from /etc/os-release no clue why.

I still see see the following line in `/etc/os-release` on Raspbian Lite &mdash; after all apt updates have been applies &mdash; to several RPi 4's:

```
root@box:~# grep VERSION_ID /etc/os-release
VERSION_ID="10"
```

No idea why @jvonau is experiencing this and I'm not.  Anything else that should be tried?

jvonau, 2020-02-02:
Think it was just me, ctl c out of runrole calibre but the testing repo was created then used later during an apt operation.

closed 2020-02-02 by jvonau