iiab/iiab #2232: Should Calibre-Web playbook & vars be renamed with 'calibre_web' internally?

holta, 2020-02-03:
Might be a good idea similar to the recent renaming of roles/usb-lib to roles/usb_lib as @jvonau helped make happen, for cleaner interop with the `./runrole` command, and to reduce confusion/errors across the board.

Specific Question: https://github.com/iiab/iiab/pull/2229#issuecomment-581165835

holta, 2020-04-06:
Tangentially related:

#2311 "Unable to install Calibre-Web on Ubuntu 20.04 pre-release: Python 3.8 / pip3 issue?"

jvonau, 2020-04-08:
7.2 or drop it

jvonau, 2022-04-05:
just close this, unless you are going to do the rename.

holta, 2022-04-05:
This issue might actually be perfect for a new IIAB hacker trying to learn Ansible etc (please consider if that happens to be you!)