iiab/iiab #2241: pxb requires nodejs 10.x, blocks install.

jvonau, 2020-02-06:
> TASK [pbx : FAIL (STOP THE INSTALL) IF nodejs_installed is undefined] **********
skipping: [127.0.0.1]
TASK [pbx : Fail if nodejs_version is incorrect] *******************************
fatal: [127.0.0.1]: FAILED! => {"changed": false, "msg": "PBX: Refusing to install as nodejs_version is not 10.x.  Please fix that (looking into /etc/iiab/local_vars.yml and /opt/iiab/iiab/vars/default_vars.yml) and then rerun."}
PLAY RECAP *********************************************************************
127.0.0.1                  : ok=127  changed=41   unreachable=0    failed=1    skipped=60   rescued=0    ignored=0

holta, 2020-02-07:
@m-anish I noticed this as well, in recent weeks, since IIAB transitioned from Node.js 10.x to 12.x a couple months ago.

Do you think roles/pbx might in fact run under 12.x these days?

holta, 2020-02-13:
Unrelated to #2210 per se...except that both recent issues involve @m-anish's PBX work from 2018 especially...and we could definitely use your guidance Anish on how to navigate and prioritize these together!

holta, 2020-02-13:
@m-anish I'm assigning this to you just so we can together come to a decision here...tolerating Node 12.x even if it's a risk?

jvonau, 2020-05-14:
php-gettext removed from ubuntu-20 https://answers.launchpad.net/ubuntu/+source/php-gettext/+question/690678

holta, 2020-09-03:
@jvonau should we close this now?

jvonau, 2020-09-07:
addressed with #2489

closed 2020-09-07 by holta