iiab/iiab #2244: install opens ports for uninstalled services

tim-moody, 2020-02-07:
if services_externally_visible: True (or 5?) all service ports are opened, even ports for non-installed services

holta, 2020-02-07:
If more granularity in needed, `services_externally_visible: 5` (e.g. main 20 ports open, or some such) and `services_externally_visible: 6` (most all ports open) could be implemented.

All implementers will *never* be satisfied of course.

But as always let's do our best to establish/support recurringly requested common cases.

tim-moody, 2020-02-07:
not the point. why open ports for services that are not installed? who would want that?

tim-moody, 2020-02-07:
I can see none or all installed and enabled or only web server or only web server and non-proxied. But I can't see ports that aren't used.

holta, 2020-02-07:
That is the point (the question).

Some folk despise firewalls getting in their way, and express the strong desire to be able to add their own stuff unimpeded.

We may disagree with their security posture.  But it's a request by a certain class of the DIY crowd to "get the firewall out if my way..."

tim-moody, 2020-02-07:
'getting the firewall out of the way' would be opening all ports. that I could understand, though I personally wouldn't usually do it except maybe when developing. But that is not my point or my complaint. option 5 opens an arbitrary list of ports that makes no sense.

jvonau, 2020-02-13:
The only difference would be a request to a open port with no service running would be the issuance a 'Connection refused' messages in place of silently discarding the request causing the client to wait for a connection timed out message when the port is blocked. The alternative would be to make the port list conditional on the service being 'enabled' but iiab-gen-iptables would need to be rerun every time the state of any service changes, this is do-able but not worth the plumbing required.

closed 2020-02-13 by holta