iiab/iiab #2247: Internetarchive --reinstall not working

mitra42, 2020-02-11:
### Expected behavior and actual behavior.
```
cd /opt/iiab
sudo git pull
sudo ./runrole --reinstall internetarchive
```
I'd expect that it does a `yarn upgrade` at some point as it used to. 

But ... the upgrade code seems to have been pulled out during that previous MR - so it doesn't work

### Specifications like the version of the IIAB, operating system version, or hardware details.
Git pull including up to the MR 2245 from @holta on 10th Feb

holta, 2020-02-11:
> `cd /opt/iiab`

You mean `cd /opt/iiab/iiab` I'm pretty sure?  I'll look into it!

holta, 2020-02-11:
> upgrade code seems to have been pulled out

Upgrade/update code was preserved here: https://github.com/iiab/iiab/blob/master/roles/internetarchive/tasks/main.yml#L27-L42

But let's not get losts in the reeds.  More important I want to call @mitra42, so we all get on the same page first, helping everyone understand what's actually expected/useful/best here going forward.

Whether or not following outlines (like!) https://github.com/iiab/iiab/pull/2238#discussion_r376168178

mitra42, 2020-02-11:
Yes - in https://github.com/iiab/iiab/pull/2238#issuecomment-582940866 it was said that:
```
That was the past, indeed. But let's get rid of special cases now. So that all (or most every) IIAB role is reinstalled using this same syntax:
./runrole --reisntall internetarchive
```
(note there is an obvious typo in that line).

but I took that to mean that update is done only when I give the arg `--reinstall` rather than only when I DO NOT give the `--reinstall` arg.

I also cant find anywhere that `interetarchive_upgrade` is set, since the `set_fact` for that has been commented out. 

I don't really care what the outcome is, except that I *do* need a way to tell it to do an upgrade !

holta, 2020-02-11:
Thanks @mitra42 for calling me just now.  Conclusions:

1) I'll implement your suggestion now with a quick PR, such that `--reinstall` (and any invocation of [internetarchive/tasks/install.yml](https://github.com/iiab/iiab/blob/master/roles/internetarchive/tasks/install.yml)) essentially does a `rm -rf /opt/iiab/internetarchive/node_modules` before running things like `yarn config set child-concurrency 1 && yarn add @internetarchive/dweb-mirror` etc.

2) Later when there is broader IIAB community consensus on the need for `--upgrade` or `--update` with apps in general (and what this generally/should mean) the `when: internetarchive_installed is defined and internet_available   # END BLOCK` on Line 42 of main.yml (wrapped around block https://github.com/iiab/iiab/blob/master/roles/internetarchive/tasks/main.yml#L27-L42) will likely be changed to take that into account those new expectations, likely sometime later this year.

holta, 2020-02-11:
PR #2248 (now merged) makes this much more clear, within both:
- [internetarchive/roles/main.yml#L22-L45](https://github.com/iiab/iiab/blob/master/roles/internetarchive/tasks/main.yml#L22-L45)
- [internetarchive/roles/install.yml#L32-L54](https://github.com/iiab/iiab/blob/master/roles/internetarchive/tasks/install.yml#L32-L54)

TO DO:

- Resolve general expectations as to when wiping/upgrading of IIAB apps is desired, and how this can be cleanly communicated to implementers/operators (via documentation initially, and later via a GUI like Admin Console or OLIP etc...)  Initial design discussions @ https://github.com/iiab/iiab/pull/2238#discussion_r376168178

closed 2020-02-11 by holta