iiab/iiab #2261: Change Nextcloud d/l site from download.nextcloud.com to d.iiab.io

holta, 2020-02-18:
_Does anybody know another download/mirror site for Nextcloud?_

Too many people appear to be affected by the severe throttling that download.nextcloud.com imposes, even those who have not recently downloaded a copy of Nextcloud, are experiencing 100X slowdowns as a result of Nextcloud's download site.

Folks that downloaded several recent copies of Nextcloud can experience a 1000X slowdown in downloading things like the ~82MB https://download.nextcloud.com/server/releases/nextcloud-18.0.1.tar.bz2 (i.e. it feels the 1990s all over again, with such downloads taking more than an hour!)

Very strange that Nextcloud effectively blocks people from downloading their own product...

Nevertheless it's continuing for about 6 months now...so IIAB's default download site for Nextcloud likely needs to change in coming days, in places like the following:

https://github.com/iiab/iiab/blob/master/vars/default_vars.yml#L412-L415

jvonau, 2020-04-08:
close

holta, 2020-09-03:
Good enough: there's now an option to do this within /etc/iiab/local_vars.yml

closed 2020-09-03 by holta