iiab/iiab #2265: Raspbian blocks hostapd when IIAB microSD is built in RPi 4 and then moved to RPi 3 or 3 B+ (or Zero W?)

holta, 2020-02-26:
The workaround (thanks to @jvonau) is to run `rfkill unblock wifi` and then `systemctl restart hostapd`

But first, run `rfkill list` which to confirm the problem appears as follows:

```
root@box:~# rfkill list
0: phy0: Wireless LAN
        Soft blocked: yes
        Hard blocked: no
1: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
```

Background error messages:

```
root@box:~# systemctl status hostapd
● hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
   Loaded: loaded (/etc/systemd/system/hostapd.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Fri 2020-02-21 23:33:05 GMT; 2min 20s ago
  Process: 311 ExecStart=/usr/sbin/hostapd -P /run/hostapd.pid /etc/hostapd/hostapd.conf (code=exited, status=0/SUCCESS)
 Main PID: 311 (code=exited, status=0/SUCCESS)

Feb 21 23:33:05 box.lan hostapd[311]: wlan0: Could not connect to kernel driver
Feb 21 23:33:05 box.lan hostapd[311]: Interface initialization failed
Feb 21 23:33:05 box.lan hostapd[311]: wlan0: interface state COUNTRY_UPDATE->DISABLED
Feb 21 23:33:05 box.lan hostapd[311]: wlan0: AP-DISABLED
Feb 21 23:33:05 box.lan hostapd[311]: wlan0: interface state DISABLED->DISABLED
Feb 21 23:33:05 box.lan hostapd[311]: wlan0: AP-DISABLED
Feb 21 23:33:05 box.lan hostapd[311]: wlan0: CTRL-EVENT-TERMINATING
Feb 21 23:33:05 box.lan hostapd[311]: hostapd_free_hapd_data: Interface wlan0 wasn't started
Feb 21 23:33:05 box.lan hostapd[311]: nl80211: deinit ifname=wlan0 disabled_11b_rates=0
Feb 21 23:33:05 box.lan systemd[1]: hostapd.service: Succeeded.
```
```
root@box:~# more /etc/hostapd/hostapd.conf
# Basic configuration

interface=wlan0

ssid=Internet in a Box
channel=6
bridge=br0

# Hardware configuration
driver=nl80211
country_code=US
# limit emissions to what is legal in country
ieee80211d=1
hw_mode=g

# UNCOMMENT the following 2 lines, so 'hostapd_cli all_sta' and 'hostapd_cli list_sta' work
#ctrl_interface=/var/run/hostapd
#ctrl_interface_group=0
```
```
root@box:~# uname -a
Linux box.lan 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linu [snip]
```

holta, 2020-02-26:
@jvonau is it possible this Raspbian glitch is triggered (exacerbated?) by our country codes contradicting themselves in these 2 files?

```
root@box:~# grep country /etc/hostapd/hostapd.conf
country_code=US
# limit emissions to what is legal in country
```
```
root@box:~# grep country /etc/wpa_supplicant/wpa_supplicant.conf
country=MX
```

Context: `wpa_supplicant.conf` was populated by /usr/bin/raspi-config when the brand new OS was installed a week ago...whereas `hostapd.conf` was auto-populated from https://github.com/iiab/iiab/blob/master/vars/local_vars_medium.yml#L40 (that I should have changed to MX!) via  https://github.com/iiab/iiab/blob/master/roles/network/templates/hostapd/iiab-hostapd.conf.j2

holta, 2020-02-26:
@jvonau can you look this over...to see if the 2 warnings below should still be addressed?

- wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
- wlan0: Could not connect to kernel driver

As excerpted from:

```
root@box:~# systemctl status hostapd
● hostapd.service - Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
   Loaded: loaded (/etc/systemd/system/hostapd.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2020-02-25 23:02:13 CST; 11s ago
 Main PID: 9326 (hostapd)
    Tasks: 1 (limit: 1599)
   Memory: 416.0K
   CGroup: /system.slice/hostapd.service
           └─9326 /usr/sbin/hostapd -P /run/hostapd.pid /etc/hostapd/hostapd.conf

feb 25 23:02:13 box.lan systemd[1]: Started Hostapd IEEE 802.11 AP, IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator.
feb 25 23:02:13 box.lan hostapd[9326]: Configuration file: /etc/hostapd/hostapd.conf
feb 25 23:02:13 box.lan hostapd[9326]: wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
feb 25 23:02:13 box.lan hostapd[9326]: wlan0: Could not connect to kernel driver
feb 25 23:02:13 box.lan hostapd[9326]: Using interface wlan0 with hwaddr dc:a6:32:0d:3c:7a and ssid "Internet in a Box"
feb 25 23:02:13 box.lan hostapd[9326]: wlan0: interface state COUNTRY_UPDATE->ENABLED
feb 25 23:02:13 box.lan hostapd[9326]: wlan0: AP-ENABLED
```

holta, 2020-03-01:
Before we resolve this tkt, @georgejhunt can you also keep an eye on these 2 hostapd warnings (posted in full on #2265 above) so we can uncover whether they are important or not?

- wlan0: interface state UNINITIALIZED->COUNTRY_UPDATE
- wlan0: Could not connect to kernel driver

holta, 2020-03-09:
I've added an `rfkill unblock wifi` explanation to https://github.com/iiab/iiab/wiki/IIAB-7.1-Release-Notes

Question:

Should we save everyone the pain[*] by adding "rfkill unblock wifi" into `/usr/libexec/iiab-startup.sh` as explained in http://faq.iiab.io -> "42 Is there a file like AUTOEXEC.BAT to run jobs on boot?"

[*] IIAB microSD cards are often inserted into headless RPi's in the field...where screen, keyboard and ssh are just not viable.

holta, 2020-05-07:
@georgejhunt can you jot down your conclusions on ticket here[*] as to whether something like "rfkill unblock wifi" into [/usr/libexec/iiab-startup.sh](https://github.com/iiab/iiab/blob/master/roles/2-common/templates/iiab-startup.sh) ?

[*] arising from today's discussion during IIAB weekly call @ http://minutes.iiab.io

holta, 2020-05-07:
@georgejhunt if you don't want to work on this, please unassign yourself, Thanks!

georgejhunt, 2020-05-08:
I unassigned myself

holta, 2020-05-08:
I'm moving this to IIAB 7.2 milestone.

Anybody who can help...please contact @jvonau (e.g. on ticket here) who has good ideas on how to make this happen-

holta, 2020-05-12:
Huge progress thanks to @jvonau's PR #2403 !

closed 2020-05-12 by holta