iiab/iiab #2268: Can we make piclone work on Raspbian Lite and/or Ubuntu 19.10.1+ ?

holta, 2020-03-01:
Non-technical operators & implementers constantly ask for a zero-hassle backups/duplication of their IIAB microSD cards.

The desktop (graphical) version of the Raspbian OS With has solved this in an extremely elegant way: Raspberry start icon (top-left of screen) -> Accessories -> SD Card Copier.

This (and its underlying `piclone` routine) self-clones a running Raspberry Pi — including the auto-shrink (auto-truncate) step so that any brand of microSD card can be used — even if the destination microSD is slightly smaller than the source (running) microSD.

The question: can `piclone` be made to work on Raspbian Lite and/or Ubuntu 19.10.1+ on Raspberry Pi?

To this end, @jvonau outlines the current requirements of `piclone` Raspbian Lite:

```
pi@raspberrypi:~ $ sudo apt install piclone
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  adwaita-icon-theme aspell aspell-en at-spi2-core bubblewrap dbus-user-session dbus-x11
  dconf-gsettings-backend dconf-service dictionaries-common emacsen-common enchant fontconfig
  fontconfig-config fonts-dejavu-core glib-networking glib-networking-common glib-networking-services
  gsettings-desktop-schemas gstreamer1.0-gl gstreamer1.0-libav gstreamer1.0-plugins-base
  gstreamer1.0-plugins-good gstreamer1.0-pulseaudio gstreamer1.0-x gtk-update-icon-cache
  hicolor-icon-theme hunspell-en-us libaa1 libaacs0 libaom0 libaspell15 libass9 libasyncns0
  libatk-bridge2.0-0 libatk1.0-0 libatk1.0-data libatspi2.0-0 libavahi-client3 libavc1394-0
  libavcodec58 libavfilter7 libavformat58 libavutil56 libbdplus0 libbluray2 libbrotli1 libbs2b0
  libcaca0 libcairo-gobject2 libcairo2 libcdparanoia0 libchromaprint1 libcodec2-0.8.1 libcolord2
  libcups2 libdatrie1 libdconf1 libdrm-amdgpu1 libdrm-common libdrm-nouveau2 libdrm-radeon1 libdrm2
  libdv4 libegl-mesa0 libegl1 libenchant1c2a libepoxy0 libfftw3-double3 libflac8 libflite1
  libfontconfig1 libgail-common libgail18 libgbm1 libgdk-pixbuf2.0-0 libgdk-pixbuf2.0-bin
  libgdk-pixbuf2.0-common libgl1 libgl1-mesa-dri libglapi-mesa libglvnd0 libglx-mesa0 libglx0 libgme0
  libgraphene-1.0-0 libgraphite2-3 libgsm1 libgstreamer-gl1.0-0 libgstreamer-plugins-base1.0-0
  libgstreamer1.0-0 libgtk-3-0 libgtk-3-bin libgtk-3-common libgtk2.0-0 libgtk2.0-bin
  libgtk2.0-common libgudev-1.0-0 libharfbuzz-icu0 libharfbuzz0b libhunspell-1.7-0 libhyphen0 libice6
  libiec61883-0 libjack-jackd2-0 libjavascriptcoregtk-4.0-18 libjbig0 libjson-glib-1.0-0
  libjson-glib-1.0-common liblcms2-2 liblilv-0-0 libllvm9 libmp3lame0 libmpg123-0 libmysofa0 libnorm1
  libnotify4 libogg0 libopenjp2-7 libopenmpt0 libopus0 liborc-0.4-0 libossp-uuid16 libpango-1.0-0
  libpangocairo-1.0-0 libpangoft2-1.0-0 libpgm-5.2-0 libpixman-1-0 libpostproc55 libproxy1v5
  libpulse0 libraw1394-11 librest-0.7-0 librsvg2-2 librsvg2-common librubberband2 libsecret-1-0
  libsecret-common libsensors-config libsensors5 libserd-0-0 libshine3 libshout3 libsm6 libsnappy1v5
  libsndfile1 libsodium23 libsord-0-0 libsoup-gnome2.4-1 libsoup2.4-1 libsoxr0 libspeex1
  libsratom-0-0 libssh-gcrypt-4 libswresample3 libswscale5 libtag1v5 libtag1v5-vanilla libthai-data
  libthai0 libtheora0 libtiff5 libtwolame0 libva-drm2 libva-x11-2 libva2 libvdpau-va-gl1 libvdpau1
  libvidstab1.1 libvisual-0.4-0 libvorbis0a libvorbisenc2 libvorbisfile3 libvpx5 libwavpack1
  libwayland-client0 libwayland-cursor0 libwayland-egl1 libwayland-server0 libwebkit2gtk-4.0-37
  libwebp6 libwebpdemux2 libwebpmux3 libwoff1 libx11-xcb1 libx264-155 libx265-165 libxcb-dri2-0
  libxcb-dri3-0 libxcb-glx0 libxcb-present0 libxcb-render0 libxcb-shm0 libxcb-sync1 libxcb-xfixes0
  libxcomposite1 libxcursor1 libxdamage1 libxfixes3 libxi6 libxinerama1 libxkbcommon0 libxrandr2
  libxrender1 libxshmfence1 libxslt1.1 libxtst6 libxv1 libxvidcore4 libxxf86vm1 libzmq5
  libzvbi-common libzvbi0 mesa-va-drivers mesa-vdpau-drivers notification-daemon uuid va-driver-all
  vdpau-driver-all x11-common xdg-dbus-proxy zenity zenity-common
Suggested packages:
  aspell-doc spellutils wordlist gvfs hunspell openoffice.org-hunspell | openoffice.org-core
  libbluray-bdj colord cups-common libdv-bin oss-compat libenchant-voikko libfftw3-bin libfftw3-dev
  libvisual-0.4-plugins gstreamer1.0-tools jackd2 liblcms2-utils opus-tools pulseaudio libraw1394-doc
  librsvg2-bin lm-sensors serdi sordi speex
The following NEW packages will be installed:
  adwaita-icon-theme aspell aspell-en at-spi2-core bubblewrap dbus-user-session dbus-x11
  dconf-gsettings-backend dconf-service dictionaries-common emacsen-common enchant fontconfig
  fontconfig-config fonts-dejavu-core glib-networking glib-networking-common glib-networking-services
  gsettings-desktop-schemas gstreamer1.0-gl gstreamer1.0-libav gstreamer1.0-plugins-base
  gstreamer1.0-plugins-good gstreamer1.0-pulseaudio gstreamer1.0-x gtk-update-icon-cache
  hicolor-icon-theme hunspell-en-us libaa1 libaacs0 libaom0 libaspell15 libass9 libasyncns0
  libatk-bridge2.0-0 libatk1.0-0 libatk1.0-data libatspi2.0-0 libavahi-client3 libavc1394-0
  libavcodec58 libavfilter7 libavformat58 libavutil56 libbdplus0 libbluray2 libbrotli1 libbs2b0
  libcaca0 libcairo-gobject2 libcairo2 libcdparanoia0 libchromaprint1 libcodec2-0.8.1 libcolord2
  libcups2 libdatrie1 libdconf1 libdrm-amdgpu1 libdrm-common libdrm-nouveau2 libdrm-radeon1 libdrm2
  libdv4 libegl-mesa0 libegl1 libenchant1c2a libepoxy0 libfftw3-double3 libflac8 libflite1
  libfontconfig1 libgail-common libgail18 libgbm1 libgdk-pixbuf2.0-0 libgdk-pixbuf2.0-bin
  libgdk-pixbuf2.0-common libgl1 libgl1-mesa-dri libglapi-mesa libglvnd0 libglx-mesa0 libglx0 libgme0
  libgraphene-1.0-0 libgraphite2-3 libgsm1 libgstreamer-gl1.0-0 libgstreamer-plugins-base1.0-0
  libgstreamer1.0-0 libgtk-3-0 libgtk-3-bin libgtk-3-common libgtk2.0-0 libgtk2.0-bin
  libgtk2.0-common libgudev-1.0-0 libharfbuzz-icu0 libharfbuzz0b libhunspell-1.7-0 libhyphen0 libice6
  libiec61883-0 libjack-jackd2-0 libjavascriptcoregtk-4.0-18 libjbig0 libjson-glib-1.0-0
  libjson-glib-1.0-common liblcms2-2 liblilv-0-0 libllvm9 libmp3lame0 libmpg123-0 libmysofa0 libnorm1
  libnotify4 libogg0 libopenjp2-7 libopenmpt0 libopus0 liborc-0.4-0 libossp-uuid16 libpango-1.0-0
  libpangocairo-1.0-0 libpangoft2-1.0-0 libpgm-5.2-0 libpixman-1-0 libpostproc55 libproxy1v5
  libpulse0 libraw1394-11 librest-0.7-0 librsvg2-2 librsvg2-common librubberband2 libsecret-1-0
  libsecret-common libsensors-config libsensors5 libserd-0-0 libshine3 libshout3 libsm6 libsnappy1v5
  libsndfile1 libsodium23 libsord-0-0 libsoup-gnome2.4-1 libsoup2.4-1 libsoxr0 libspeex1
  libsratom-0-0 libssh-gcrypt-4 libswresample3 libswscale5 libtag1v5 libtag1v5-vanilla libthai-data
  libthai0 libtheora0 libtiff5 libtwolame0 libva-drm2 libva-x11-2 libva2 libvdpau-va-gl1 libvdpau1
  libvidstab1.1 libvisual-0.4-0 libvorbis0a libvorbisenc2 libvorbisfile3 libvpx5 libwavpack1
  libwayland-client0 libwayland-cursor0 libwayland-egl1 libwayland-server0 libwebkit2gtk-4.0-37
  libwebp6 libwebpdemux2 libwebpmux3 libwoff1 libx11-xcb1 libx264-155 libx265-165 libxcb-dri2-0
  libxcb-dri3-0 libxcb-glx0 libxcb-present0 libxcb-render0 libxcb-shm0 libxcb-sync1 libxcb-xfixes0
  libxcomposite1 libxcursor1 libxdamage1 libxfixes3 libxi6 libxinerama1 libxkbcommon0 libxrandr2
  libxrender1 libxshmfence1 libxslt1.1 libxtst6 libxv1 libxvidcore4 libxxf86vm1 libzmq5
  libzvbi-common libzvbi0 mesa-va-drivers mesa-vdpau-drivers notification-daemon piclone uuid
  va-driver-all vdpau-driver-all x11-common xdg-dbus-proxy zenity zenity-common
0 upgraded, 224 newly installed, 0 to remove and 7 not upgraded.
Need to get 125 MB of archives.
After this operation, 901 MB of additional disk space will be used.
Do you want to continue? [Y/n] n
Abort.
```

Ref: #823

holta, 2020-03-01:
For now let's try to get a command-line interface (CLI) working.  Something roughly like this:

```
piclone-cli <source card> <dest card>
```

When that essential 1st step is accomplished, the friendly wrappers (that folks demand!) can quickly then follow:
- a CLI wrapper that takes no parameters at all, auto-populating the source mount point (i.e. the running OS, on the OLD microSD card) and the destination mount point (i.e. the NEW microSD card, typically sitting in an SD card reader)
- make it invokable from Admin Console
- make it invokable from Ubuntu 19.10.1+ & 20.04+ graphical desktops (most all new IIAB operators/implementers strongly prefer a graphical desktop, as they & their communities are learning Linux at the same time that they are learning IIAB)

georgejhunt, 2020-03-01:
It looks like this has already been done:
https://github.com/nwright-mcc/piclone_cmd

On Ubuntu 19.01, lxde can be added via apt. It boosts the server installed
footprint from 2.3 GB to 3.9GB.

But this means that we might forever be in the same business that
raspian.org is in -- building images. But on the other hand, a rpi-ready
image, with all IIAB pre-installed might be easier for newbies compared
with our outline installer.  A single image, with an admin console function
that toggles the systemd default.target between graphical and multiuser,
could make Tim's "content-ready" image the sweet spot.

georgejhunt, 2020-03-01:
outline=>oneline

georgejhunt, 2020-03-01:
The cmd line piclone_cmd successfully transferred an image (required
recompile of object file -- perhaps because my OS was 64bit).

holta, 2020-03-01:
Amazing!

Any idea how hard does it'll likely be to try to port this from Raspbian Lite...to [Ubuntu 19.10.1+ on Raspberry Pi](https://ubuntu.com/download/raspberry-pi) ?

(Which apparently has a single partition, containing the root filesystem and nothing else.)

georgejhunt, 2020-03-01:
piclone was successful on Ubuntu_19.01

holta, 2020-03-01:
> piclone was successful on [ Ubuntu_19.10 ]

Incredible!  Was that 32-bit or 64-bit Ubuntu 19.10?

In context with this comment @tim-moody found at [SD Card Copier for debian/Ubuntu ARM64 or ARMHF (64 bit)](https://github.com/raspberrypi-ui/piclone/issues/22#issuecomment-548535935):

> It just creates the same layout of partitions on the new SD card as on the old one, and does straight file system copies of every file in every partition on the source to the equivalent partition on the target. The procedure can easily be deduced by reading the piclone.c file in this repo, as it does everything by making system calls to standard UNIX file system tools.

holta, 2020-03-01:
> piclone_cmd successfully transferred an image (required
recompile of object file -- perhaps because my OS was 64bit).

64-bit Ubuntu 19.10.1+ progress!  Publish basic recipe if poss @georgejhunt ?

holta, 2020-03-01:
Far from the final word, but a 1st pass update of our IIAB FAQ is now published:

http://FAQ.IIAB.IO -> "44 How do I back up, shrink & copy IIAB microSD cards?"

holta, 2020-03-02:
Tangentially Related:

#2269 "enable aarch64 in kiwix, disable stem.php"

holta, 2020-03-22:
@tim-moody please also take look at https://github.com/georgejhunt/piclone_cmd

@georgejhunt is the above ready for Raspbian Lite and Ubuntu 20.04 LTS pre-releases on RPi 4 to your knowledge?

holta, 2020-03-28:
SEE:

PR iiab/iiab-admin-console#283 "0.4.0.10 Add PiClone functionality to Admin Console"

holta, 2020-03-30:
Documented here:

https://github.com/iiab/iiab-admin-console/blob/master/roles/console/files/help/InstContent.rst#clone-iiab-server

closed 2020-03-30 by holta