iiab/iiab #2270: ubuntu 20.04 aarch64 early testing

georgejhunt, 2020-03-03:
I've had fun setting up a development branch ubuntu 20.04 64bit ubuntu on rpi4 from the available install iso. 
Our oneline install script stalled on python3-pip not available yet.  
We may find that they do not meet the release target -- April 23 (current date Mar 2)

georgejhunt, 2020-03-10:
## Recipe for using daily build iso on rpi:
1. I started with ubuntu 19.01 image. But it might have been faster to partition and copy /boot (deleting all of partition2 took a long time)
2. Copy kernel and initrd to /boot.
3. Mount the squashfs from the daily build iso, and copy it to userland, to /dev/mmcblk0p2.
4. Power on, and add "restricted" and "universe" to /etc/sources.list.
5. I always apt-get install vim git tree
5. Needs a /etc/netplan/config.yaml to bring up eth0, set a /etc/resolv.conf that points to a good dns.
5. Expand the /dev/mmcblk0p2 beyond 2GB using resize2fs, after making larger partition with fdisk.
5. Install openssh-server
6. Without a functioning /etc/fstab, /dev/mmcblk0p2 needs to be remounted rw after every boot.
7. We need a real is_rpi. I just put it in ./vars/ubuntu-20 for temporary.
7. Get wget, and get the one line installer.
8. Dnsmasq does not want to start if br0 is missing -- comment it out in /etc/dnsmasq.d/iiab
9. Apparently netplan only works on wifi with NetworkManager, but we use systemd.networkd. So getting br0 going via hostapd needs work.

holta, 2020-03-10:
> I started with ubuntu 19.01 image

There is no 19.01 per se.  (There are 19.10, 19.10.1 and 20.04 daily builds of different kinds.)

To avoid all confusion, @georgejhunt  can you please provide a complete/example URL to an ISO you'd use ?

jvonau, 2020-03-10:
Think it would be easier just to alter apt to use ubuntu-20 sources on a booted 19.10.1 image and then run 'apt dist-upgrade'. This should work on a stock out of the box image or one that has iiab applied to the os already.
13 - hostapd/br0 creation works with ubuntu-19 running systemd-networkd or NetworkManager as is, I would expect 20 to be no different and should work without any mods.
14 - think you are barking up the wrong tree by adding is_rpi to the new vars/ubuntu-20.yml that is a OS flag not a HARDWARE flag.

holta, 2020-03-10:
> 14 - think you are barking up the wrong tree by adding is_rpi to the new vars/ubuntu-20.yml that is a OS flag not a HARDWARE flag.

That may be true (similar discussion around #2275) but let's use a more empathetic tone, whenever possible.

So all ideas are welcomed and considered.

georgejhunt, 2020-03-12:
## Create a Ubuntu 20.04 by upgrading from 19.10
1. apt-get update
1. apt-get upgrade
1. apt --purge autoremove
1. do-release-upgrade -d

georgejhunt, 2020-03-12:
## Changes I made to get this snapshot:
1. add ubuntu-20 to scripts/local_facts.fact
1. commented out the /usr/sbin/iiab invocation of scripts/ansible (L 211 -- installed ansible via apt by hand). The scriipt version had unresolved dependencies.
1. Paste is https://pastebin.com/3SABrZYE

georgejhunt, 2020-03-12:
## Ubuntu 20.04 rpi image Uploaded to archive.org
* https://archive.org/download/iiab-7.1-200312-ubuntu_20.04_rpi-g1b4f928c.img/iiab-7.1-200312-ubuntu_20.04_rpi-g1b4f928c.img.zip
* Five minutes to download, as opposed to 90 minutes to upgrade from 19.10
* Downloaded, and md5sum verified, md5sum=24b7cae431153062b520da81722f53f3
* Installed on an SD card, and functions as expected
* Passwords: user/passwd: root/ubuntu, ubuntu/ubuntu

holta, 2020-03-12:
@georgejhunt can you clarify if https://github.com/iiab/iiab/blob/master/vars/ubuntu-20.yml is the way it should be?

(Even if it's not yet being used by your IIAB installs onto these daily pre-releases of Ubuntu Server 20.04 LTS?)

ASIDE: Has anybody yet tried an IIAB install onto a recent pre-release of Ubuntu Desktop 20.04 LTS?

georgejhunt, 2020-03-13:
10 days ago, I did my first install of IIAB on ubuntu 20.04. The changes I
made to get through the playbook are documented at
https://github.com/iiab/iiab/compare/master...georgejhunt:u2004.
I don't think my quick and dirty edits are PR material. I ran into problems
getting br0 up, -- but then on my first run, I didn't have any of the
kernel modules properly installed. The next run should be more valid
I don't know what you want clarified, Adam

holta, 2020-03-13:
@georgejhunt are the values within https://github.com/iiab/iiab/blob/master/vars/ubuntu-20.yml what you think they should (likely) be for Ubuntu 20.04 LTS?

georgejhunt, 2020-03-13:
re: vars/ubuntu-20.yml
It looks to me like there is a lot of dead wood that we have just
migrated from one OS to the next. Until I get a clean install I will
not propose any changes. I don't see the utility of removing the dead
wood. Or perhaps better said: I'm not motivated.

holta, 2020-03-13:
@georgejhunt have you tried out https://ubuntu.com/blog/how-to-install-ubuntu-with-the-new-raspberry-pi-imager ("How to install Ubuntu with the new Raspberry Pi Imager" from 2020-03-13) that @jvonau recommends?

jvonau, 2020-03-14:
> @georgejhunt have you tried out https://ubuntu.com/blog/how-to-install-ubuntu-with-the-new-raspberry-pi-imager ("How to install Ubuntu with the new Raspberry Pi Imager" from 2020-03-13) that @jvonau recommends?

I didn't recommend using it, I found a write up online and shared it.

jvonau, 2020-03-14:
>     2. commented out the /usr/sbin/iiab invocation of scripts/ansible (L 211 -- installed ansible via apt by hand). The scriipt version had unresolved dependencies.

I suspect ansible's ppa whose dependencies drag in python2 would be the culprit. There is an outstanding issue on ansible's github that I'm tracking about this upstream issue. Without more information than 'unresolved dependencies' I have nothing concrete to report about the urgency to make the switch to python3 in the ppa to upstream ansible.

georgejhunt, 2020-03-14:
Trace of failing ansible install via scripts/ansible: (I think it bears out
your hypothesis about python version)

ANSIBLE NOT FOUND ON THIS COMPUTER -- LET'S TRY TO INSTALL IT!

(Internet-in-a-Box requests Ansible 2.9.6 or higher)

apt update; install dirmngr; PPA to
/etc/apt/sources.list.d/iiab-ansible.list

Hit:1 http://ppa.launchpad.net/ansible/ansible/ubuntu bionic InRelease

Hit:2 http://ports.ubuntu.com/ubuntu-ports focal InRelease

Hit:3 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease

Hit:4 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease

Hit:5 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease

Reading package lists... Done

Building dependency tree

Reading state information... Done

All packages are up to date.

Reading package lists... Done

Building dependency tree

Reading state information... Done

dirmngr is already the newest version (2.2.19-3ubuntu2).

The following packages were automatically installed and are no longer
required:

  ieee-data libestr0 libfastjson4 python3-argcomplete python3-dnspython
python3-jmespath python3-kerberos python3-libcloud

  python3-lockfile python3-netaddr python3-ntlm-auth
python3-requests-kerberos python3-requests-ntlm python3-selinux
python3-winrm

  python3-xmltodict

Use 'apt autoremove' to remove them.

0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

IF YOU FACE ERROR "signatures couldn't be verified because the public key
is not available" THEN REPEATEDLY RE-RUN "sudo apt-key adv --keyserver
hkp://keyserver.ubuntu.com:80 --recv-keys 93C4A3FD7BB9C367"

Executing: /tmp/apt-key-gpghome.YJeldcBktG/gpg.1.sh --keyserver hkp://
keyserver.ubuntu.com:80 --recv-keys 93C4A3FD7BB9C367

gpg: key 93C4A3FD7BB9C367: "Launchpad PPA for Ansible, Inc." not changed

gpg: Total number processed: 1

gpg:              unchanged: 1

apt update; apt install ansible and python3 dependencies explained at:

https://github.com/iiab/iiab/tree/master/scripts/ansible.md

Hit:1 http://ports.ubuntu.com/ubuntu-ports focal InRelease

Hit:2 http://ppa.launchpad.net/ansible/ansible/ubuntu bionic InRelease

Hit:3 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease

Hit:4 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease

Hit:5 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease

Reading package lists... Done

Building dependency tree

Reading state information... Done

All packages are up to date.

Reading package lists... Done

Building dependency tree

Reading state information... Done

python3-setuptools is already the newest version (45.2.0-1).

python3-setuptools set to manually installed.

python3-pip is already the newest version (18.1-5ubuntu1).

virtualenv is already the newest version (15.1.0+ds-2ubuntu1).

Some packages could not be installed. This may mean that you have

requested an impossible situation or if you are using the unstable

distribution that some required packages have not yet been created

or been moved out of Incoming.

The following information may help to resolve the situation:

The following packages have unmet dependencies:

 ansible : Depends: python-paramiko but it is not installable

           Depends: python:any (< 2.8) but it is not installable

           Depends: python:any (>= 2.7.5-5~) but it is not installable

E: Unable to correct problems, you have held broken packages.

root@box:/opt/iiab/iiab#

jvonau, 2020-03-15:
Thanks, updated upstream issue.

holta, 2020-03-15:
> Thanks, updated upstream issue.

https://github.com/ansible/ansible/issues/57342

jvonau, 2020-03-16:
### How to browse/edit an image
Configure a partitioned loop device for the image:
losetup -P -f /path/to/image.img
`sudo losetup -P -f /media/usb1/opt/iiab/ubuntu-19.10.1-preinstalled-server-arm64+raspi3.img`

`ls /dev/loop*`
Look for devices containing p repesenting each partition on the image: ie loopXp1 loopXp2 loopXpX:
Mine ended up with:
/dev/loop0p1
/dev/loop0p2

Create mount points for mounting the partitions:
`sudo mkdir /mnt/part1` 
`sudo mkdir /mnt/part2`

Mount the partitions:
`sudo mount /dev/loop0p1 /mnt/part1` 
`sudo mount /dev/loop0p2 /mnt/part2`

Access via mountpoint
`ls /mnt/part1/`
`ls /mnt/part2/`

jvonau, 2020-03-16:
Using the above to "do some stuff" like [set wifi ssid/passwords](https://discourse.ubuntu.com/t/how-to-install-ubuntu-on-your-raspberry-pi/14660) or alter syscfg.txt for bluetooth support (read /mnt/part1/nobtcfg.txt) before booting the image.

holta, 2020-03-30:
@georgejhunt & All:

Has anyone yet tried to install IIAB onto one of the "arm64" daily pre-releases of Ubuntu 20.04 LTS at http://cdimage.ubuntu.com/ubuntu-server/daily-preinstalled/current/ ?

holta, 2020-03-31:
FYI IIAB install instructions for Ubuntu 20.04 are posted in preliminary form (thanks to @shanti-bhardwa & @jvonau) here:

- https://github.com/iiab/iiab/wiki/IIAB-Platforms
  - Section 2.
    - Install Tips...for Raspberry Pi...arm64 pre-releases

holta, 2020-04-09:
@georgejhunt thinks it's time to close this.

"Mission Accomplished" and this is still very useful documentation as [IIAB 7.1](https://github.com/iiab/iiab/wiki/IIAB-7.1-Release-Notes) approaches final release.

holta, 2020-04-20:
Install tips for anybody trying to install IIAB over WiFi onto the latest Ubuntu 20.04 "daily" builds from http://cdimage.ubuntu.com/ubuntu-server/daily-preinstalled/current/ which still requires...

- WiFi configuration has some remaining netplan bugs (https://bugs.launchpad.net/ubuntu/+source/cloud-init/+bug/1870346) and so requires...
  - Multiple reboots may be required!
    - So that netplan starts and wlan0 appears in the output of 'ip a'
  - Either edit `/boot/firmware/network-config` prior to booting, so it looks something like...
  ```
  version: 2
  ethernets:
    eth0:
      dhcp4: true
      optional: true
  wifis:
    wlan0:
      dhcp4: true
      optional: true
      access-points:
        ACTUAL-SSID:
          password: "XXXXXX"
  ```
  - Or...edit `/etc/netplan/50-cloud-init.yaml` after booting, so it looks something like...
  ```
  network:
      ethernets:
          eth0:
              dhcp4: true
              optional: true
      version: 2
      wifis:
          wlan0:
              access-points:
                  ACTUAL-SSID:
                      password: xxxxxxx
              dhcp4: true
              optional: true
  ```

holta, 2020-04-21:
Just a warning that after IIAB is WiFi-installed onto the [latest daily build Ubuntu 20 (arm64) for RPi 4](http://cdimage.ubuntu.com/ubuntu-server/daily-preinstalled/current/), you will lose your WiFi connection.

I restored it with @jvonau by modifying `/etc/netplan/01-iiab-config.yaml` and creating `/etc/netplan/02-iiab-config.yaml` as follows:

```
root@box:/etc/netplan# more 01-iiab-config.yaml
network:
  version: 2
  renderer: networkd
  ethernets:
    eth0:
      dhcp4: yes
      optional: true

##    bridges:
#      # the key name is the name for virtual (created) interfaces
#        br0:
#        # IDs of the components; switchports expands into multiple interfaces
#        # interfaces: [switchports]
#            # should really compute the netmask like the wan side
#            addresses: [ 172.18.96.1/19]
#            # there should really not be a gateway present testing to prove
#            gateway4: 172.18.96.1
#            nameservers:
#                addresses: [172.18.96.1]
#            parameters:
#                forward-delay: 0
#                stp: false
```

```
root@box:/etc/netplan# more 02-iiab-config.yaml
network:
  wifis:
    wlan0:
      dhcp4: true
      optional: true
      access-points:
        [ACTUAL-SSID]:
          password: [REDACTED]
```

jvonau, 2020-04-21:
The above is not quite true, that test was against master.. don't edit /etc/netplan/50-cloud-init.yaml just create /etc/netplan/02-iiab-config.yaml if you did not use the /boot/firmware/network-config trick.

holta, 2020-04-21:
> use the /boot/firmware/network-config trick.

Is that some kind of template file that should be edited prior to 1st boot of the microSD?

(Is this written up somewhere?)

holta, 2020-04-22:
@jvonau recommends:

https://discourse.ubuntu.com/t/how-to-install-ubuntu-on-your-raspberry-pi/14660

jvonau, 2020-04-22:
small issue with network-config trick https://bugs.launchpad.net/ubuntu/+source/cloud-init/+bug/1870346

holta, 2020-04-22:
1) fyi 'netplan apply' (or 'netplan try') is not quite enough to activate an upstream WiFi connection, after installing /etc/iiab/02-iiab-config.yaml

2) One also needs to run 'systemctl restart netplan-wpa-wlan0.service' (or reboot).

3) ASIDE...the following 2 lines appear to be unnecessary within /etc/iiab/*.yaml files:

   >   version: 2
   >   renderer: networkd

jvonau, 2020-04-23:
netplan bridge problem https://bugs.launchpad.net/ubuntu/+source/systemd/+bug/1860926

holta, 2020-04-23:
fyi Ubuntu 20.04 LTS is now officially released for Raspberry Pi:
https://github.com/iiab/iiab/wiki/IIAB-Platforms

jvonau, 2020-04-23:
cloud-init wifi bug https://bugs.launchpad.net/ubuntu/+source/cloud-init/+bug/1870346

holta, 2020-04-23:
I'm re-opening this issue as an umbrella bug-tracking ticket to help everyone monitor some of the highest priority Ubuntu 20.04 bugs/issues related to IIAB 7.1 and 7.2 on arm64 / Raspberry Pi.

jvonau, 2020-04-23:
Think the valid test for the noted cloud-init bug would be to use network-confg to setup the network before booting the first time and see if the network is available when booted. Should the network be unavailable does `sudo netplan apply` bring the network up is the next question.

jvonau, 2020-04-23:
not really a bug but noted behavior of netplan https://bugs.launchpad.net/ubuntu/+source/netplan.io/+bug/1874494

jvonau, 2020-04-23:
uart uboot related https://bugs.launchpad.net/ubuntu/+source/u-boot/+bug/1873520

holta, 2020-05-07:
fyi "Dual WiFi" (PR #2393) is now part of [IIAB 7.1](https://github.com/iiab/iiab/wiki/IIAB-7.1-Release-Notes) [Release Candidate 1](https://github.com/iiab/iiab/releases/tag/7.1-RC1).

Related:

#2379 "Explain how to connect IIAB's internal Wi-Fi to the Internet: how to enter SSID & password"
#2382 "Merged 2380 [how to configure upstream & downstream WiFi?]"