iiab/iiab #2271: What's the best way to monitor RPi's CPU temp when running Ubuntu 19.10.1+ ?

holta, 2020-03-03:
@tim-moody mentions:

- "vcgencmd measure_temp" does not work on Ubuntu 19.10.1 on Raspberry Pi
- paste <(cat /sys/class/thermal/thermal_zone*/type) <(cat /sys/class/thermal/thermal_zone*/temp) | column -s $'\t' -t | sed 's/\(.\)..$/.\1°C/'
- https://askubuntu.com/questions/15832/how-do-i-get-the-cpu-temperature

georgejhunt, 2020-03-11:
## This recipe makes lxde desktop optional on 64bit Ubuntu server:
1. Install Ubuntu 19.10 server edition -- about 2.3 GB.
1. The newly booted server will do automatic upgrades/maintenance. So wait a bit to run any apt-get installs!
1. Do an "apt-get install lxde" -- This adds about 1.8 GB to the SD card and takes less than 30 minutes on an ISP with 7mB download speed.
1.  "Apt install xinit".
1. As root, issue the command "systemctl set-default multi-user.target" -- this makes the console default to text mode.
1. When user wants cpu temperature, start up lxde desktop via "startx".
1. Right click on lxde menu bar (the following 3 instructions need only be done once).
1. Select "Add / Remove Panel Items"
1. Click green plus with the 'Add" label.
1. Scroll down to Temperature Monitor, Select, and click "Add".
1. Thereafter, the CPU temperature will be continuously displayed on the LXDE desktop.
1. "startx" and "logout" from the lxde menu bar, to toggle GUI.
1. The "systemctl set-default (multi-user/graphical).target changes the power-on default.

georgejhunt, 2020-03-16:
This video suggests that firmware updates have been very effective at reducing power consumption. And it compares active cooling (fans) with passive (flirc).
https://www.youtube.com/watch?v=VJC6OpGpq0Y

holta, 2020-09-10:
FYI `cat /sys/class/thermal/thermal_zone0/temp` works on Ubuntu and also on Raspberry Pi OS.

closed 2020-09-10 by holta