iiab/iiab #2274: WordPress fails to publish anything: "Publishing failed. Error message: The response is not a valid JSON response"

holta, 2020-03-04:
This IIAB was built in RPi 4 and used within RPi 3, after running "rfkill unblock wifi" per #2263.

Trying to upload an image triggered the message.

Trying to post an blog entry (1) from WordPress's library of image or (2) even without any images...both then failed, with the very same error message.

[*] Earlier, uploading an image (as part of a blog post) failed because the "image was too large" &mdash; as WordPress defaults are way too low &mdash; blocking most all modern photos _(instead of offering to compress them as it should!)_

[**] The original WordPress error messages were in Spanish.

holta, 2020-03-04:
FYI uploading of many different (very small) images into WordPress was attempted, but these all failed, with the same WordPress error message every time.

(Surprising that this WordPress error message is not more specific and/or actionable...)

holta, 2020-03-04:
@tim-moody & others who know more about NGINX:

Can you help explain why http://box/wordpress/wp-admin/site-health.php gives the following error?

> The REST API encountered an unexpected result
>
> Performance
The REST API is one way WordPress, and other applications, communicate with the server. One example is the block editor 
>
> screen, which relies on this to display, and save, your posts and pages.
>
> The REST API call gave the following unexpected result: (404)
404 Not Found
nginx/1.14.2

This would seem to a result of converting from Apache to NGINX :/

Specifically, please install a MEDIUM-sized IIAB (or run `cd /opt/iiab/iiab` then `./runrole wordpress` with /etc/iiab/local_vars.yml set to install/enable WordPress) and then click on http://box/wordpress/wp-admin/tools.php -> Site Health to see the above error &mdash; which is likely the underlying cause to WordPress being unable to publish anything (anymore!)

tim-moody, 2020-03-04:
did fresh install on 19.10 VM
site-health displays a page (which says all tests passed)

tim-moody, 2020-03-04:
how did you try to upload the images?

tim-moody, 2020-03-04:
in /etc/nginx/conf.d/wordpress-nginx.conf after 
`root /library; `
add 
`try_files $uri $uri/ /wordpress/index.php$is_args$args;`

holta, 2020-03-06:
@ericnitschke @kananigit

1. This works.  I added (and indented) the line @tim-moody suggests above (`try_files $uri $uri/ /wordpress/index.php$is_args$args;`) to /etc/nginx/conf.d/wordpress-nginx.conf (after line `root /library;`) and then ran `systemctl restart nginx` and I can now publish basic blog posts to http://box.lan/wordpress

2. The error/warning `Publishing failed. Error message: The response is not a valid JSON response.` no longer appears within http://box/wordpress/wp-admin/site-health.php (certainly after reboot, but I did not try that prior to reboot...)

3. For now I changed /etc/php/7.3/fpm/php.ini (on this one particular offline RPi 3 B+) to permit the school to post normal/modern photos:

   upload_max_filesize = 2M -> 20M
   post_max_size = 8M -> 80M

   REASON: the RPi 3 B+ does not have enough RAM/resources to handle the high php limits as specified in https://github.com/iiab/iiab/blob/master/roles/www_front_end/tasks/main.yml#L40-L52 when `nginx_high_php_limits` is set to True.

   RECOMMENDATION: IIAB should set interim/reasonable upload_max_filesize & post_max_size (similar to the above) so that WordPress far more usable by all e.g. everyday teachers posting photos from everyday phones.  Presumably this should be implemented alongside `nginx_high_php_limits` in roles/www_front_end/tasks/main.yml ?

holta, 2020-03-07:
This all worked like a charm at a new Chiapas, Mexico school today where I installed with a half-day-long training including all teachers, their smartphones and some ancient/underpowered Windows laptops (Samsung N150 Plus, running Windows 7 Starter) from ~2010.  Let's cook up some PR soon, ideally with these 2 lines somehow baked into `/etc/php/{{ php_version }}/fpm/php.ini` :

```
upload_max_filesize = 20M    ; 2M (default) is not nearly enough in 2020
post_max_size = 80M    ; 8M (default) not nearly enough in 2020
```

i.e. let's find some way to set these 2 variable across all IIAB installations by default.  A good place to make this happen is likely [www_front_end/tasks/main.yml#L40-L52](https://github.com/iiab/iiab/blob/master/roles/www_front_end/tasks/main.yml#L40-L52) of nearby?

(Are any of the other 3 PHP variables therein also critical??)

holta, 2020-03-07:
As soon as this tweak (fix) is implemented to override bogus (archaic) PHP defaults, let's also fix http://FAQ.IIAB.IO -> "36 WordPress & Moodle Administration: What tips & tricks exist?" -> "How do I optimize WordPress and Moodle for high-traffic usage?"

holta, 2020-03-09:
Solved by PR #2280.

closed 2020-03-09 by holta